openpnp/openpnp #1036: Neoden4 connect instruction

mateusz-spychala, 2020-08-12:
#747 Feature Request

Hello,

does anybody know what is status of support of Neoden4 in OpenPnP project. 
I have searched internet and there is very little information about this topic. 
In OpenPnP repo there is driver for Neoden4. It seems fully operational, but there is
no instruction how to connect PC to Neoden's control board. 

It seems that using OpenPnP to control Neoden will solve lot of software related
problems. It would be great to have even brief instruction how to get start with
OpenPnP on Neoden4. 

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.

Have a brief instruction how to connect Neoden4, how to get started with OpenPnP.

## Why Do We Need It?
1. What problem does it solve?
- There is driver written, but it is hard to use it. 95% of effor is made, but it is hard to use.
2. Is it useful for everyone, or does it only solve a problem for your machine?
- It is useful for people that uses Neoden4 machine, since vendor software lack a lot of features. 

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

Simmilar documentation is present for Zeva 460
[https://github.com/openpnp/openpnp/wiki/Retrofitting-a-Zeva-460---Part-1](url)

Regards,
Mateusz

vonnieda, 2020-08-12:
@tlan Would you be able to assist here?

tlan, 2020-08-12:
Yes, I'll answer when I'm back from vacation next week  :-)

Mvh / Best regards,
Thomas Langås
(+47) 95731546

vonnieda, 2020-08-12:
Thanks Thomas!

mateusz-spychala, 2020-08-13:
Thanks Thomas! Have a nice vacation!

felix-dc, 2020-08-19:
hello Everyone, 

Is there already more information about this topic? 

Kind regards, 
Felix

mateusz-spychala, 2020-09-13:
Hello @tlan
Can you help with this topic?

Best regards,
Mateusz

MatSpy, 2021-02-26:
Hello, 

This is almost 100% duplicate of #1063. One should be left open. This should be closed since #1063 is much mover
informative.

closed 2021-02-28 by mateusz-spychala