openpnp/openpnp #1051: Can't change speed/feedate on Liteplacer/TinyG

Sabia123, 2020-08-26:
# Problem Report
ISSUE
I can set the Speed (%) anywhere from 100% to 0% - the liteplacer moves at the same, fast speed no matter what. 

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Jog the machine
4. Set speed lower
5. Jog machine 
6. Observe the speed
7. I have tried setting the speed to 0.1 during nozzle changes. It still moves at full speed. (So I have tried both the sliding adjustment and text-based adjustment, neither have an effect. 

## Expected Result
I expected the speed adjustment to have an effect. 

## Actual Result
The head moves at full speed. 

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue

I have my MOVE_TO command like this: G0 {X:X%.4f} {Y:Y%.4f} {Z:Z%.4f} {Rotation:A%.4f} {FeedRate:F%.0f}
(Initially this was set to G0 {X:X%.4f} {Y:Y%.4f} {Z:Z%.4f} {Rotation:A%.4f} F{FeedRate:%.0f} but that seemed wrong when looking at the other parameters. Either way the response/speed is the same. 

I have the log file of 2 moves, each 10mm in X. One at 100%, the other at less than 10%. Both moves went equally quickly. 

2020-08-26 10:58:39.231 ReferenceNozzle DEBUG: Nozzle1.moveTo((68.000000, 142.300000, 0.000000, 0.000000 mm), 0.96)
2020-08-26 10:58:39.231 GcodeDriver DEBUG: sendCommand(G0 X150.4000    F960, 5000)...
2020-08-26 10:58:39.241 GcodeDriver DEBUG: sendCommand(serial://COM3 G0 X150.4000    F960, 5000) => [tinyg [mm] ok>]
2020-08-26 10:58:39.241 GcodeDriver DEBUG: sendCommand(null, 250)...
2020-08-26 10:58:39.491 GcodeDriver DEBUG: sendCommand(serial://COM3 null, 250) => [qr:31, qi:1, qo:0, posx:147.197,posy:167.700,posz:0.000,posa:0.000,feed:960.00,vel:3553.27,unit:1,coor:1,dist:0,frmo:1,stat:5]
2020-08-26 10:58:39.491 GcodeDriver DEBUG: sendCommand(null, 250)...
2020-08-26 10:58:39.744 GcodeDriver DEBUG: sendCommand(serial://COM3 null, 250) => [posx:150.400,posy:167.700,posz:0.000,posa:0.000,feed:960.00,vel:0.00,unit:1,coor:1,dist:0,frmo:1,stat:3, qr:32, qi:0, qo:1]
2020-08-26 10:58:48.809 ReferenceNozzle DEBUG: Nozzle1.moveTo((78.000000, 142.300000, 0.000000, 0.000000 mm), 0.09)
2020-08-26 10:58:48.809 GcodeDriver DEBUG: sendCommand(G0 X160.4000    F90, 5000)...
2020-08-26 10:58:48.826 GcodeDriver DEBUG: sendCommand(serial://COM3 G0 X160.4000    F90, 5000) => [tinyg [mm] ok>]
2020-08-26 10:58:48.826 GcodeDriver DEBUG: sendCommand(null, 250)...
2020-08-26 10:58:49.078 GcodeDriver DEBUG: sendCommand(serial://COM3 null, 250) => [qr:31, qi:1, qo:0, posx:157.761,posy:167.700,posz:0.000,posa:0.000,feed:90.00,vel:3385.59,unit:1,coor:1,dist:0,frmo:1,stat:5]
2020-08-26 10:58:49.078 GcodeDriver DEBUG: sendCommand(null, 250)...
2020-08-26 10:58:49.329 GcodeDriver DEBUG: sendCommand(serial://COM3 null, 250) => [posx:160.400,posy:167.700,posz:0.000,posa:0.000,feed:90.00,vel:0.00,unit:1,coor:1,dist:0,frmo:1,stat:3, qr:32, qi:0, qo:1]

I suspect the issue is that the VELOCITY is what should be changed, not feedrate....? Because that remains constant. But how can I make the speed setting change that?

BTW I also used the Menu -> Help -> Submit Diagnostics feature ...that failed too, with an error message "Error posting to https://pastebin.com/api/api_post.php using the params: api_paste_code=OpenPnP+Diagnistics.......etc etc That message was like 10 screens wide or more so I won't type all that, sorry.

markmaker, 2020-08-26:
Hi Sabia

The Speed [%] will only affect the maximum **allowable** velocity. When you are doing a short move (10mm) this maximum speed will never be reached, the machine needs to decelerate long before that point can be reached, otherwise it would overshoot the target. 

[image]
https://makr.zone/thinking-machine/82/

However, you see in the log, that OpenPnP correctly sends the F word in G-code: 

`G0 X150.4000 F960`
vs.
`G0 X160.4000 F90`

If you try a larger move (100mm) you may start to see a difference at low % settings. 

OpenPnP currently only controls the feed-rate, not the acceleration. Therefore, the speed factor is often absolutely ineffective. 

This is one of the many reasons for the large Motion Control overhaul I'm working on. In this Version of OpenPnP the acceleration is now also controlled. It is scaled back even by the square of the Speed [%], so if you reduce the speed to 10% it will accelerate at only 1%. This will result in a move that act as if in slow-motion, i.e. it will take exactly 1/10% = 10 times as long. In case you're wondering why 1% is correct for acceleration to achieve that, it's because the acceleration affects the position of the machine multiplied by **time squared**, so if we want it to take 10 times longer, we need to divide acceleration by the square of 10 i.e /100 = 1%.
[image]

https://en.wikipedia.org/wiki/Acceleration#Uniform_acceleration

[image]

If you want to check it out, there is a testing version available. 

https://openpnp.org/test-downloads/

But be mindful that the documentation is not there yet and bugs have to be expected. 

Follow the links here:
https://makr.zone/openpnp-global-axes-and-enhanced-motion-control/490/

_Mark

frasier2k, 2020-11-12:
You have to use the G1 command instead of G0.

Arccording to TinyG Wiki:

https://github.com/synthetos/TinyG/wiki/Gcode-Support

Gcode | Parameters | Command | Description
G0 | axes | Straight traverse | Traverse at maximum velocity. At least one axis must be present
G1 | axes, F | Straight feed | Feed at feed rate F. At least one axis must be present

markmaker, 2021-02-21:
Assuming this is solved.

closed 2021-02-21 by markmaker