openpnp/openpnp #1055: The program loads the processor by 100%

bacrua, 2020-09-07:
My HP laptop with a processor and 3 and 6GB of RAM when starting the openPnP program reacts for a very long time to any actions in the program. Opened the Windows 7 task manager, the openPnP program loads the processor by 100%

vonnieda, 2020-09-07:
Try lowering the FPS setting on the cameras. In most cases there is no
reason to go above 2-3 FPS and that is usually the cause of significant CPU
load.

bacrua, 2020-09-08:
Thanks, but didn't help. Having tried the modes up to 320x288, the load is reduced to 48%, but it is still not possible to work and perform tasks. It seems to me that it should not be so. In fact, if you do not include image processing, this is a normal video stream.

markmaker, 2020-09-08:
Can you try with [ProcessExplorer](https://docs.microsoft.com/en-us/sysinternals/downloads/process-explorer) and look at the individual threads?
_Mark

bacrua, 2020-09-08:
Ok

bacrua, 2020-09-08:
screen
[image]

markmaker, 2020-09-08:
Yes, now double-click on `OpenPnP.exe` and look at the threads.

Should look similar to this (one ELP camera):
[image]

bacrua, 2020-09-08:
it `s Magic. I connected the camera and control via the usb hub, made a screen of the processes. I connected both cameras and control directly to the port. Surprisingly, in both cases, the load is about 50%. I don't understand what happened, but I was already able to use the auto setup feeder

markmaker, 2020-09-08:
Has the lighting changed? Are the cameras now in a darker environment? 

One thing to consider is this:
https://makr.zone/camera-fps-cpu-load-and-lighting-exposure/519/

In auto-exposure, the more light the camera has, the more frames per second it delivers and the more CPU it uses. 

If you set the exposure manually (a **must** in computer vision!) then it depends on the exposure setting. The higher the exposure, the lower the frame rate, the lower the CPU load.

_Mark

bacrua, 2020-09-08:
new screen
[image]

bacrua, 2020-09-08:
usb hub
[image]

bacrua, 2020-09-08:
Only usb port
[image]

bacrua, 2020-09-08:
> Has the lighting changed? Are the cameras now in a darker environment?
>
> One thing to consider is this:
> https://makr.zone/camera-fps-cpu-load-and-lighting-exposure/519/
>
> In auto-exposure, the more [...]

Indeed, the lighting has changed. When the processor was loaded it was light from the window. It's darker now, you can see it on the camera. I did not change the exposure. I have it installed manually. The camera looks down at -10. The camera looks up -8

bacrua, 2020-09-08:
The joy did not last long.

bacrua, 2020-09-09:
> Has the lighting changed? Are the cameras now in a darker environment?
>
> One thing to consider is this:
> https://makr.zone/camera-fps-cpu-load-and-lighting-exposure/519/
>
> In auto-exposure, the more [...]

Thanks, Mark fr your answer. If camera bottom and camera top working continiously it is not correctly. Bottom camera on when capture nozzle and component. But CPU loading all time. Top camera works when auto setup feeder and vizualizations. Why both camera must to work all time.

markmaker, 2020-09-09:
>But CPU loading all time. Top camera works when auto setup feeder and vizualizations. Why both camera must to work all time.

That is not the fault of OpenPnP, it's just how these cameras work. The CPU load happens (mostly) inside the DirectShow `qcap.dll` and this is a part of Windows, not OpenPnP. 

Remember, we are using affordable webcams for Machine Vision. The reason this works, is that the camera is continually streaming images to the application, and we can take **any** of these images **immediately**, as we like. If we were to open the camera stream only when an image it is needed, it would add a long delay, because this takes considerable time. 

Also remember that unlike most Webcam applications we use **two** cameras at the same time. 

Furthermore, if you compare your ELP camera (assuming you have one) with a "normal" webcam (e.g. built into a notebook) you will see a big difference in image detail quality. This detail quality comes at a price in terms of image decompression CPU load. 

So it is quite expected that when you use two high quality cameras on a 5-year-old low price processor such as the i3-5005, it takes its toll. This may also be related to some GPU acceleration feature employed for image decompression that is severely limited on these low level processor models.

These are probably part of the reasons, why cheap chinese PnP machines still use two switched analog cameras on one USB capture device, i.e. only one image stream is used. 

Having said all this, there is a small part of the CPU load that could still be optimized in OpenPnP one day.  

_Mark

bacrua, 2020-09-09:
@markmaker 
It's not OpenPnP's fault, it's just that these cameras are working. The CPU load happens (mostly) inside DirectShow qcap.dll and it is part of Windows, not OpenPnP.

But functions are called from openpnp
they can be called at the right time, or streams can be switched. There is no case in machine operation when cameras analyze both simultaneously.
But this is so, thinking out loud. Thanks for the time.

markmaker, 2020-09-09:
@bacrua there is a SwitcherCamera in OpenPnP, so you can do it if you like. 

https://github.com/openpnp/openpnp/wiki/SwitcherCamera

_Mark

aneox, 2020-09-09:
On my machine(i5 3570 + RX580, macos) I see a high CPU load(50%+) even in idle machine state, maybe camera capture should activate before capture place. I cant work on a computer with openpnp running even in idle state.

vonnieda, 2020-09-09:
This seems abnormal. I have no problem running OpenPnP with two cameras
along with FireFox, Eclipse, Spotify, Slack, Discord, SourceTree, etc. all
running and working at the same time. And that's while developing code in
Eclipse.

Can you post your machine.xml, and manufacturer / model of your cameras?

Thanks,
Jason

aneox, 2020-09-09:
@vonnieda 
usb ELP cameras from aliexpress
[machine.xml](https://yadi.sk/d/rejhFrQndQwI_A)

vonnieda, 2020-09-09:
Can you post the exact model or link? Curious if they are the 720p ones I
use, or something much higher.

Jason

aneox, 2020-09-09:
@vonnieda i purchased them about 2 years ago, its simple 720p 30fps cameras
but looks like for some reason fps is much lower in openpnp

bacrua, 2020-09-09:
usb100w05mt
[image]
[image]

[machine_xml.zip](https://github.com/openpnp/openpnp/files/5196889/machine_xml.zip)

darenschwenke, 2020-09-09:
As the random newbie attempting to solve this issue 2 years ago in
something that may or may not be related and then moving on to something
else... video capture mode?
Some introduce extreme load on the bus, others introduce extreme load on
the processor. Where you end up is vastly different.

bacrua, 2020-09-12:
@vonnieda 
I sent the file and model of the camera.

bacrua, 2020-09-12:
@markmaker 
@vonnieda 
[image]
[image]
[image]

Tested on ather notebook CPU pentium N3700 windows 8.1 
the picture is the same. But unlike the last test, I can at least work.

markmaker, 2021-02-21:
@bacrua what is the state of this?

bacrua, 2021-02-21:
@markmaker 
Hello. Now I am using a different computer and one of the latest versions. I can check on the old one a little later if there were changes in the program. Have there been changes to address this issue?

closed 2021-02-21 by bacrua

markmaker, 2021-02-21:
There have been changes (bugs fixed) but I'm not sure if your experience is related. If you can test the newest version on the old computer, it would help. Thank you.

Jnesselr, 2021-02-21:
I actually experienced this issue between when this issue was opened and a couple of weeks ago. It had nothing to do with the cameras themselves, either virtual or physical. I remember I did a git bisect and found that [this commit](https://github.com/openpnp/openpnp/commit/2fa440ca97dc8a7b6f4ea9e6f4a269e954279ef0) was the culprit. I did not bisect the branch itself that was merged in. I also have not bisected to find when the issue was fixed, either. I just know the difference was 50% of my CPU vs 100-200% of my CPU.

markmaker, 2021-02-21:
Hi @Jnesselr 

As the author of these commits I'm very interested.

> I did a git bisect 

Intriguing. Can you elaborate? How exactly did you do that? What about the configuration? Did you just compare the basic auto-migrated machine.xml? Or were some new configurations (perhaps by Issues & Solutions) involved? 

>  I also have not bisected to find when the issue was fixed, either. 

Just to be clear: are you saying the issue is now fixed?

_Mark

Jnesselr, 2021-02-21:
I did a git bisect across the develop branch, yeah. Each time I nuked the openpnp configuration and told it to make and run openpnp. If it hit >100% CPU for 5 seconds, it marked it as bad. Rinse, repeat. Unfortunately, I don't have the program I used with git bisect run anymore. I can try to regenerate it if you want, but it was pretty simple overall.

vonnieda, 2021-02-21:
I think this is two separate issues. The issue @Jnesselr is referring to was introduced in the initial merge of AMC in December, if I remember correctly. They mentioned it on Discord a few weeks back and I looked at it, and found it was present in December, but not in the commits at the time we were discussing it, which would have been January. I didn't bother to track the fix commit down since it seemed fixed.

The issue @bacrua is referring to was prior to the AMC merge, so I think it must not be related. I suspect that the significant refactoring of all the camera stuff over the past two months has probably changed, if not addressed, the issue. So yes, I'd say if we want to be thorough @bacrua should test on the older computer. Or we can just leave this closed since at the moment everyone concerned seems happy :)

bacrua, 2021-02-21:
Thanks for your hard work. Yes, I'll do the tests. After all, I owe you in some way. Without your labor, I would not have a car. I cannot help financially, but I am always ready to do useful work.

bacrua, 2021-02-22:
Good day to all. Just tested the latest version of OpenPnP. All EXCELLENT I checked the rendering, the cameras are working fine. The processor is not loaded. I specially launched video from the network at the same time. Everything worked, nothing slowed down. There were some questions in the illumination of the lower chamber. But most likely I did not configure something. But my goal was to check the cameras and the total load. EVERYTHING IS VERY EXCELLENT

vonnieda, 2021-02-22:
That great to hear @bacrua - thank you for testing!