openpnp/openpnp #1056: ACTUATOR_READ_REGEX does not properly parse numbers

CraftyMyner, 2020-09-14:
# Problem Report
## Describe the Issue
I am trying to use the heated bed MOSFET outputs to turn on and off the vacuum pump based on a pressure sensor. This is all working well, however, I would also like to use the same sensor for part sensing. I use the vacuum sensor as if it is a temperature sensor so the pressure can be anywhere from 0-255. For part sensing I setup an ACTUATOR_READ_COMMAND as "M105" to get the temperature sensor values, this returns the pressure as bed temperature.

```nc
ok T:-15.00 /0.00 B:113.33 /0.00 T0:-15.00 /0.00 T1:-15.00 /0.00 @:0 B@:0 @0:0 @1:0
```
I set up an ACTUATOR_READ_REGEX as: ```^ok.*B:(?<Value>\d+.\d+)``` which should return a named group with the pressure. I click read value under the actuator tab and it returns "No matching responses found." Setting the regex to ```^ok.*B:(?<Value>.*)``` returns the temperature and everything after. Setting the regex to ```^ok.*B:(?<Value>\d)``` returns the same "No matching responses found."

There seems to be an issue with the regex parsing as "\d", "[0-9]" and other ways of matching numbers does not work.

## Setup
- BTT SKR 1.4 Turbo running Marlin
- Latest OpenPNP 2.0 download
- MacBook Pro running macOS 10.14.6

markmaker, 2020-09-14:
Hi @CraftyMyner 

Look at the example at the end here:
https://makr.zone/vacuum-sensor/192/

I'm no Regex expert, but I think you are missing the trailing

`.*`

allowing for the rest of the characters after the group to match. 

Btw, you might consult the user group first, before opening an Issue here.
https://groups.google.com/g/openpnp

Your next stop:
https://github.com/openpnp/openpnp/wiki/Setup-and-Calibration:-Vacuum-Sensing

_Mark

CraftyMyner, 2020-09-14:
Ahh, yes you are exactly right. I am used to regex that requires a "$" to also match the end of the string otherwise it will just not care.

closed 2020-09-14 by CraftyMyner