openpnp/openpnp #1058: G21; set millimeter mode error

cx2323, 2020-09-25:
I haven't been able to get open pnp to connect to my machine 
[image]
[image]

bilsef, 2020-09-26:
What does the log show?

cx2323, 2020-09-28:
set millimeter mode error

bilsef, 2020-09-28:
Most likely it is a communication problem.  G21 is usually the first
command sent to the printer.  This is what I would like to see in the log:

[image: image.png]

In this case OpenPnP is not connected to the machine (there is no COM13),
so the write fails, the command times out, and gives the error.

If you can show your complete TRACE log there should be some clues as to
why it is not communicating.

Savvy97, 2020-10-02:
[image] will this help

bilsef, 2020-10-02:
It looks like there is a COM3, but there is no G-code aware controller
connected to it.  What controller are you using?  Are you connected by
USB?  Are you sure COM3 is the correct port?

Savvy97, 2020-10-02:
Thank you for responding,

We are using TinyG. We are connected by USB, and COM3 is the correct port.

vonnieda, 2020-10-02:
TinyG may not support Gcode comments - I can't remember offhand. Try
removing the comment after the command.

Jason

tshiao4695, 2020-10-25:
can you working for your machine now? my openpnp is also note TimeoutG21 for me.

frasier2k, 2020-11-11:
You have to configure the driver section as described in the link below:

https://github.com/openpnp/openpnp/wiki/TinyG

markmaker, 2021-02-21:
I believe we can consider TinyG communications issues as being solved by proper XON/XOFF and Confirmation Flow Control configuration. Both are now suggested/warned by [Issues & Solutions](https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions). Merged from test into develop.

closed 2021-02-21 by markmaker