openpnp/openpnp #1059: Openpnp won't jog or move in Z direction after homing

Sabia123, 2020-10-02:
I had my liteplacer with tinyg working with openpnp 2.0 and tried to get the contact probe working but gave up. Then I added a new ReferenceNozzle to replace the ContactProbeNozzle, and copied the values for the nozzle tips, etc. 

Saved configuration and restarted openpnp. 

Now when I press home the homing is done just fine like always, but I cannot jog in Z direction anymore. X and Y jog just fine, and the homing sequence does move Z as it should. 

When I try to change nozzles the X and Y movements go as expected, but without Z movement. 

## Steps to Reproduce
1. Start OpenPnP
2. Click the Power On button.
3. Add new nozze
4. Copy offsets from existing/old nozzle
5. Delete old nozzle
6. Restart openpnp and press home
7. Try to jog Z down 10mm, or try to change nozzle

## Expected Result
Z to move -10mm (or down to change nozzle if that was the operation asked for)

## Actual Result
No Z movement. 

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
TRACE output says:
(date, time) ReferenceNozzle DEBUG: RefNox.moveTo((-32.400000, -19.500000, -1.00000, 0.000000 mm), 0.78)

The gcode driver doesn't give the sendCommand like it does for X and Y jogs....?

vonnieda, 2020-10-02:
You probably need to add the head mountable ID for the new nozzle to your
axis mapping:
https://github.com/openpnp/openpnp/wiki/GcodeDriver%3A-Axis-Mapping#mapping-axes-to-headmountables

Jason

Sabia123, 2020-10-03:
Thanks, Jason, 

I renamed the nozzle to id="N1" and changed the z axis mapping to <string>N1</string> and now it works. Thanks!

closed 2020-10-03 by Sabia123