openpnp/openpnp #1063: Status of Neoden4 Driver?

maxkunes, 2020-10-16:
Just wondering about the status of the Neoden4 driver and how one could even use it?

markmaker, 2020-10-16:
Yes, I'm also interested, because it needs testing/attention with the new Advanced Motion Control:

#1061 

https://github.com/openpnp/openpnp/wiki/Advanced-Motion-Control

Things that need attention are marked with TODO: in the code.

_Mark

MitiCHE, 2020-12-28:
Hi,
I have a Neoden 4 machine. I try to change the Chinese software with openpnp. I can control the machine movement but can't make camera working. It's seems that they use old camera, without standard driver.
The test I made is on linux, but without camera I cannot use it with openpnp. The linux detect there it's a device attached over USB but not as a camera.
Product H0001 Camera Shibz
If someone has a driver for this cameras it will be helpful.
[image]

mateusz-spychala, 2021-01-08:
Hello MitiCHE, 

Can you describe how to connect Neoden board to Linux. I want also to play with it.

mateusz-spychala, 2021-01-12:
Hello MitiCHE, 

Have you read this thread?
https://charliex2.wordpress.com/2019/01/31/neoden-4-pick-and-place/
They suggest to use Cypress driver with small modification.

Regards

mateusz-spychala, 2021-01-31:
Hello all,

It seems that there is currently driver for Neoden4 cammera in OpenPnp. It connect via TCP link to server app running on host machine. I have searched for this server app, but it seems that it is not publish anywhere in repo. 

Regards,

MatSpy, 2021-02-04:
Hello all, 

To run camera it seems you have to:

1. Compile and run this app on neoden4. This will exposes Neoden4 camera over REST.
https://github.com/ExploratoryEngineering/GoNeoCam
It require probably windows 32bit version to run. 

2. Use Neoden4 camera in OpenPnp. 

It seems that all is implemented apart from feeders control. 
Connection wizard also seems to crash, but it can be easily fixed. I will provide path, but it seems that you have to add only few lines to Neoden4Driver.java

`    public double getHomeCoordinateX() {
        return this.homeCoordinateX;
    }

    public void setHomeCoordinateX(double homeCoordinateX) {
        this.homeCoordinateX = homeCoordinateX;
    }

    public double getHomeCoordinateY() {
        return this.homeCoordinateY;
    }

    public void setHomeCoordinateY(double homeCoordinateY) {
        this.homeCoordinateY = homeCoordinateY;
    }    
    `

Regards,
Mateusz

MatSpy, 2021-02-08:
Hello @MitiCHE,

I have run OpenPnp on my Neoden4, but I cannot control movement. The only thin that is working is reading pressure sensors. Can you tell production date of your machine? Maybe communication protocol has changed over time?

Regards,
Mateusz

markmaker, 2021-02-08:
Hi Mateusz

When I introduced the Advanced Motion Control into OpenPnP I had to port the Neoden driver too. I made the best effort to do that without breaking things, but ultimately I could never be sure. The same applies to the camera.

So the machine not working could be related to that. It could also be completely unrelated (e.g. with a change in protocol).

For instance: unlike for the GcodeDriver the axes were not defined, and I just left them at the standard X Y Z C axes. So you have to model the additional axes like they are on the machine, so OpenPnP knows how to handle them. 

https://github.com/openpnp/openpnp/wiki/Machine-Axes

They must also be mapped to the Nozzles, Camera etc.

https://github.com/openpnp/openpnp/wiki/Mapping-Axes

Another thing is speed control. There seems to no notion of millimeters per time unit, just a factor of 0..1. So I related this to the default 250mm/s i.e 15000mm/min that is set in OpenPnP. This could be made more realistic if you can tell me the max speed of that machine.

Most of the new motion control features, axis handling etc. will not work in the Neoden. The motion commands seem very basic, I've seen no acceleration control for instance. But maybe that Advanced Motion Control stuff is handled inside the controller instead.

If the specs are available (or can be measured) you can enter them into the axes. This way OpenPnP can accurately predict the motion. Accurate motion prediction is currently only useful for diagnostics and simulation. But in the future it might become more important e.g. to provide cost estimates to the job planner.

Once you have done all this, please send me the machine.xml so I can automatically create the model in the future, as soon as the user chooses the Neoden driver.

_Mark

MitiCHE, 2021-02-08:
Hi @MatSpy ,
Just checked in my machine and the version (serial number) is: 47965.
My test was done by changing the ssd disk and install a new Windows 10. After that I installed OpenPNP and test directly with the Nooden4 driver from OpenPNP. For the moment the problem is the camera. I will try to make more test in the future.
My issues is that I use the Neoden machine regularly and it's a little bit complicate to change the SSD etach time when I want to test. This was the reaseon to try with Linux which can be booted from external SSD, that I didn't try the OpenPNP because I focus on camera. But unfortunately there is no driver in Linux for this type of camera.
However the china's soft is not good, there are many bugs and the interface is not friendly. I hope find a way and use this machine with OpenPNP.

MitiCHE, 2021-02-08:
Hello @MatSpy 

There is a while when I tested it, but I remember that I select the Neoden driver form OpenPNP configuration and serial COM but not remember exactly the baudrate, I think 115200 or 9600. Do not forget to enable the machine in soft before sending commands.
[image]

MatSpy, 2021-02-26:
Hello all,

Good news,
I have managed to bring neoden driver up and working. Currently it is in very alpha version. 
All subsystems works. 
- Cameras (up down)
- Vacuum + blowof
- Movements xyzc
- All for nozzles
- Feeders
- Homing

It works only on windows 32 bit (Neoden driver for cameras is not signed so best option is Windows 7). 
Linux version do not work (Serial driver and Neoden dll). Serial driver sends character to slowly and triggers timeouts in main controller. 

- Cameras need to be set to 0FPS (512 x 512 resolution is much faster than 1024x1024). 

You can find very alpha version on my fork. 
I try to isolate all changes inside Neoden specific classes so merge should be easy. 

Early tests are very promising. It seems that openpnp solves biggest problem of neoden. 
- Visual homing solves inconsistent endstop homing of neoden (fine pitch placement + 0201).
- Visual template matching during feed (ReferenceLeverFeeder) solves low precision feeders. 
- Way better bottom pipeline
- Way better fiducial pipeline
- Correct nozzle calibration (original soft assumes perfectly parallel all 4 Z axis) only first nozzle is correctly
calibrated. Nozzles 2,3,4 always have slight offset. 

Feel free to test / ask questions. 

Regards,
Mateusz

MitiCHE, 2021-02-26:
Hi @MatSpy,

Very good news and thank you for sharing all your work with us.

I also connect wire to sniff the protocol over usart and move the SSD form inside to outside of the machine in order to keep the original SSD and keep working with china soft until have openpnp fully functionally.

How you get image from cameras ? What driver did you choose when you installed the camera driver on Windows 7, because I try on Windows 10 and when I selected cyusb3.sys there are a lists of driver that can be use.

I attache also few picture of the changes that I made on my machine.

Regards,
Mihail

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

MatSpy, 2021-02-27:
Hello MitiCHE,

Nice solution. I also need fast switching between original / openpnp soft. 
According cameras. 

1. Download this repo (https://github.com/charlie-x/neoden4). You will find there drivers + neoden DLL for cameras.
2. Download https://github.com/ExploratoryEngineering/GoNeoCam
3. Download golang for x86 (32bit)
4. Download github.com/gorilla/mux
5. Put mux to C:\Users\<USER>\Go\src\github.com\gorilla\mux
6. Install golang
7. Compile GoNeoCam with command go build. It will create GoNeoCam.exe
8. Install cameras using driver. Copy dll to GoNeoCam directory. 
9. Run GoNeoCam.exe in terminal
10. You are ready to connect to cameras from openpnp.

Regards.
Mateusz

izimihail, 2021-02-27:
Hello @MatSpy,

Thanks for info, I will try and share more info after doing few tests.

Regards,
Mihail

izimihail, 2021-03-03:
Hi @MatSpy ,

Which version of openpnp did you used when to test NeoDen machine ? Did you use openpnp from your github repo branch mat ?
When I try openpnp form official repo (branch develop), I cannot configure the NeoDen driver. I can add it on openpnp but I can't select the serial port because there is no menu displayed. I remember that when I try last year, when I clicked on NeoDen driver, a menu was displayed to configure the communication port.

Thanks,
Mihail

MatSpy, 2021-03-04:
Hello Mihail,

You have to compile mat branch from my repo. On openpnp repo there is bug
that crashes configuration.
Regards,
Mateusz

markmaker, 2021-03-04:
> You have to compile mat branch from my repo. 

Will you eventually create a pull request?

_Mark

izimihail, 2021-03-04:
I confirm it's working with mat branch.
Thank you Mateusz

[image]

MatSpy, 2021-03-04:
Hello Mark,

Yes I will create pull request.
Now I testing it extensivelly and making small changes at daily basis. 
You should consider this code unstable/testing.

One of last major thing that makes me troubles is following.

Neoden camera driver works at 0FPS. Frame on 
demand.
Neoden camera driver have problem when you want
to gather top and bottom camera concurrency. 

Every time when frame demand is generated by Openpnp,
it gathers 2 frames ob bottom camera and one from top. 

Is it possible to configure Openpnp to gather only frames
that it needs:

- When job is running it is simple. Gather only bottom, when fiducial recognition top.
- When free move gather only one that camera view sees. 

I have considered hacks like:
- Check light and gather only requests that have light.

Is there clear way of doing it?

Regards,
Mateusz

MatSpy, 2021-03-04:
Hello all.

Connection is COM2, 115200. 

Regards

markmaker, 2021-03-04:
@MatSpy thanks for the reply.

> Yes I will create pull request.

Cool. 

> Is it possible to configure Openpnp to gather only frames
that it needs

Yes it should work like that, if you set the cameras **Preview FPS** to 0 fps.  But this only truly works in the newer versions i.e. you would have to merge the newest **develop** branch into your branch!

See also #1095 and follow-up PRs.

If it does not work like that, please report back. 

_Mark

MatSpy, 2021-03-04:
Hello Mark,
Yes I do have merged develop branch. 

I have played with camera settings. That are my observations. 
When Openpnp window has focus. Every movement (even during job) causes 1 or 2 frames from top and bottom.
When Openpnp windows do not have focus. Everything is fine. 3 down-looking photos for loosepart feeder + 1 photo for up
looking for placement. 

I have tried different settings in camera setup but nothing is changing. 
Do you see what I am doing wrong?

Mateusz

markmaker, 2021-03-04:
> When Openpnp window has focus. Every movement (even during job) causes 1 or 2 frames from top and bottom.

When Jogging, yes. The assumption was that you want a new camera view when finished.  

But when in the Job or during other multi-move machine tasks, it should not happen, if you set the "suspend during tasks" checkbox:

[image]

Please report back, if it does not work this way. I will fix it, if it is not ok.

_Mark

MatSpy, 2021-03-04:
Hello Mark,

It seems that this option requires restart of openpnp. After restart it
works. Thank you.

If you dont mind two more questions.

1 Is there any mechanism for landling bigger parts that field of view? In
orginal neoden software it recognized 2 corners. With that it managed to
determin center and rotation.

2 When nozzles are not used, neoden releases nozzle motors for rotation.
There is no cooling and afeter some time they get hot. I have written short
script that is executed on job compleated that activate actuator
responsible of releasing drivers. It seems that it is not always called.
E.g. when job is stopped. Is it required behavior? Is there better way to
handle disabling C motors?

Regards
Mateusz

markmaker, 2021-03-04:
> It seems that this option requires restart of openpnp. After restart it
works. Thank you.

You're half right, it requires not a restart but the running one more machine task (a jog should be enough), before it becomes effective.

> 1 Is there any mechanism for landling bigger parts that field of view? In
orginal neoden software it recognized 2 corners. With that it managed to
determin center and rotation.

No. :-(

> 2 When nozzles are not used, neoden releases nozzle motors for rotation.

There is no direct solution for that. But it is on my list and no big deal. 

There is already the `org.openpnp.spi.MachineListener.machineBusy(Machine, boolean)` listener method, you could use. 

But if I were to implement it, I would like to add a timeout, so it would only engage on longer idle times.  

_Mark

izimihail, 2021-03-05:
Hi Mateusz,
Did you made a specific configuration to make working the camera on OpenPNP ?
I compile OpenPNP mat branch. The axe, vacuum, ligth camera are working but not the image. What is strange is that if I run api in firefox I can see the camera. 
It's look like OpenePNP do not trigger the api to get images.
Do you have any idea ?

[image]
[image]
[image]

Thanks,
Mihail

MatSpy, 2021-03-05:
Hello Mihail,

Please pull branch Neoden4. It should work. Other branches was created 
for experiments,
and they are deleted now. You will find there also actuators for feeders.

Btw. Can you write reference numbers of parts to make SSD hack?
I am very impressed.

Regards,
Mateusz

izimihail, 2021-03-05:
Hello Mateusz,

Thanks for replay, I will try with Neoden4 branch.

Concerning the hack for SSD, I buy the following references:
* One FFC Câble 34 pins - 0.5mm length 400 mm
[Ebay FFC Cable](https://www.ebay.fr/itm/AWM-20624-80C-60V-VW-1-4-60pin-0-5-1-0mm-FFC-C%C3%A2ble-plat-Flexib-C%C3%A2ble-Flex-2-50cm/323873851786?ssPageName=STRK%3AMEBIDX%3AIT&var=512976979015&_trksid=p2060353.m2749.l2649)

[image]

* Two mSATA SSD SATA Mini PCIE SSD Extension Cable
[Amazon SSD Extension Cable](https://www.amazon.fr/gp/product/B088HBHRH9/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1)

[image]

I though that the length of 400mm FFC cable will be a problem because of electronic perturbations but for the moment it's working without any problems. I use daily the NeoDen machine to assembly boards and the switch between the SSD is very quick if I need to test OpenPNP software.

Seeing the code improvement for NoeDen in openpnp, I hope to stop switching between the SSD and use only OpenPNP :)

Regards,
Mihail

MatSpy, 2021-03-06:
Hello all,

Interesting topic to investigate.
According to Charliex2 blog our Neoden4 has in fact color cameras. BGR to
gray conversion happens inside dll driver. If we rewrite it we can get
access to color information, that would be very usefull in openpnp.

Mateusz

P.S. Thank you Mihail.

izimihail, 2021-03-08:
Hi Mateusz,

I have a little question. Do you have this error when you build? it seems to be related to checkstyle-rules ?

`[ERROR] Failed to execute goal org.apache.maven.plugins:maven-checkstyle-plugin:3.1.0:check (validate) on project openpnp-gui: Failed during checkstyle execution: There are 6 errors reported by Checkstyle 8.29 with d:\MyWORK\01__Proiecte\01__IZITRON\NeoDen\openpnp\target\checkstyle-rules.xml ruleset. -> [Help 1]`

In order to build I need change the format of vriable declaration in NeoDenFeederActuatorConfgurationWizard.java, like this:
[image]

Do you have the same error ? Or the checkstyle is dependent on java versions ?

Thanks,
Mihail

MatSpy, 2021-03-08:
Hello,

No I do not have this error. I simply import project to eclipse. Run maven
project update. Then build.

How you build? I will check what is correct style and I will correct that.

izimihail, 2021-03-08:
I use Visual Studio Code maybe is a configuration problem. However is not a the main problem :) but I was curios if you had the same error.

[image]

MatSpy, 2021-03-08:
It is now corrected on my branch.

izimihail, 2021-03-11:
Hello,

Just for info, I tested the new Mat branch and seems the NeoDen machine it's working with almost all the features.
Remarks:
- I cannot move machine using mouse and drag the corner on image
[image]

- When I move only on X direction, there it's a little movement on Y direction

https://user-images.githubusercontent.com/79627682/110834167-7fc49c80-829d-11eb-9601-7f88492de0a9.mp4

- Cameras images not very fluent, but it's explained because of workaround go server

- Leds on camera can have multiple light mode, but only on it's use
- The pipeline to recognize holes, circle, etc doesn't work very well

However the NeoDen porting on openpnp it's very well advanced

MatSpy, 2021-03-11:
Hello, 

I have played with openpnp for over a week and my conclusion is that,
you have to give you some time to make you familiar with all features. 
Compared to orginal neoden software here we have massives number of option. 

If you want to drag camera you have to configure "Unit per pixel" (read wiki)
This is also needed for fiducial recognition. 
Here there is very intelligent system for fiducials. 
It takes model of marking. Creates how it should look under
camera. Then it search for it. Without unit per pixel
it does not know how big fiducial is. 
Apart from that you have influence on pipeline that
do all image processing. 

I have few things left on my todo list. My priority list is
1. Accuracy
2. Stability 
3. Speed
 
This little move on Y is Backlash Compensation. 
I have played with my own implementation, which I belive is better that used in Openpnp. 
Neoden also works that way. 

If you does not like it just comment out 
    	double comp = backlashCompensation;
        moveXy(x+comp, y+comp);
        Thread.sleep(10);
        setMoveSpeed(0.05);
        Thread.sleep(100);
in moveXySafe

Mateusz

izimihail, 2021-03-12:
Hello Mateusz,

It's true, there are lot of options and need to read the documentation carefully.
However now because I can play with openpnp and I will take time and read the doc and try to configure step by step each part.

Thank for info and good job !

MatSpy, 2021-06-23:
Hello, 
After some time of improving it I want to share our results, which are quite impressive. 

We have added a lot of features that speed up setting job and make it work much smoother. 

From the accuracy view biggest problems with original Neoden4:
1. Software - Incorrect calculation of placement for nozzles 2-4, full accuracy can be acquired only on nozzle 1. 
Accuracy on other nozzles depends of how parallel nozzles are assembled relative to nozzle 1. 
2. Hardware - After some measurements it occurred that stepper motor drives has poor micro-step accuracy. 
(about 50-70us additional inaccuracy or 7-8 microsteps). Similar to described here
[hackaday article](https://hackaday.com/2016/08/29/how-accurate-is-microstepping-really/)
 We have replaced them with other one (be very carefull since they are powered from 110V, even marking suggest 24V).

We can reliably place 0201 and even 01005 components. 

[Youtube movie](https://www.youtube.com/watch?v=W2gCo6_mSY4)

Regards
Mateusz

izimihail, 2021-06-23:
Hi Mateusz,

Thank you for sharing with us your work and well done !

Did you still use <go server> to get images from cameras ? When I done my tests I notice that is limiting a lot the speed assembling.

Regards,
Mihail

MatSpy, 2021-06-24:
> Did you still use to get images from cameras ? When I done my tests I

Hello Mihail,

We have improved that. Now openpnp connects directly to cameras dll and
acquiring photo takes 0.1s. I will push machine xml.

Regards.

izimihail, 2021-06-24:
Hi Mateusz,

Very very interesting, I will test also.

But it's true the Neoden have many problems related the China software. The calibration is not good, the distance between heads are not correctly, etc.

However, nice work and congratulation !

izimihail, 2021-06-24:
Hi Mateusz,

One more question. I don't know if you have the same problem as I have with the feeders. 
For small component like 0402 or 020, because of the vibration of the feeder, the components are blow up from the feeder. Let's say that for 10 rotations 1 component is lost (blow up from feeder).

Dis you see the same problem ? If yes, did you make something to improve ?

I try changing setting, to decrease strength feeder in software but not to much improvement.

Thanks,
Mihail

alexgoldstone, 2021-06-24:
This is very interesting... thank you for all sharing the collaboration effort... I had considered (and decided against) the Neoden 4 but this inspires me to reconsider it again.

@mateusz-spychala Did I understand correctly that you had to change all the stepper motors? If so could you share a link to what you used? You mentioned they are powered from 110V... presumably different stepper motors would be required for countries that use 240V mains power?

mateusz-spychala, 2021-06-24:
Hello all, 

------------- Cause parts not to vibrate out of feeder ----------------------
1. Make sure that tape moves smoothly in feeder in term of it width. Feeders are cast iron so tolerances are bit and if it is to narrow for tape it will cause it to vibrate (get stuck). Use sand paper & file to bring feeder width to 0.2mm bigger than cut tape width. 
2. Adjust peel strength and feedstrength to be as little as possible
3. Put rod to reels so they will not jump in reels holder
4. Put another rod so tape will be pulled back, not back and up (image 1)
Here it should work well for 0402 & 0201 on paper tape, but can still be problem with plastic tape
5. Remove compleatly or remove "legs" in those elements (image 1.1). Those legs causes to raise whole spring, leaving 0.5mm gap in which plastic tape will vibrate. Without it plastic tape is well pressured by spring. 
6. If it is not enough 3d print spacer and add it to spring so tape will be additionally pressured. 

[image]
[image]

------------- Microstep accuracy ----------------------
We have replaced step motor driver, not stepper motors itself. On casing orginal ones have marking 24V but it is not
true. They have beed modified to run from 110V generated in power board. It is almost imposible to find one powered
from that supply so we have replaced them with standard one powered from 24V and powered them from external supply. 

We have bought 2DM542. It works well. It also do generates much less vibrations during movements. By default
it run another direction so one must swap one coil terminators. 

[image]

Regards

MatSpy, 2021-06-24:
Hello all, 

------------- Cause parts not to vibrate out of feeder ----------------------
1. Make sure that tape moves smoothly in feeder in term of it width. Feeders are cast iron so tolerances are bit and if it is to narrow for tape it will cause it to vibrate (get stuck). Use sand paper & file to bring feeder width to 0.2mm bigger than cut tape width. 
2. Adjust peel strength and feedstrength to be as little as possible
3. Put rod to reels so they will not jump in reels holder
4. Put another rod so tape will be pulled back, not back and up (image 1)
Here it should work well for 0402 & 0201 on paper tape, but can still be problem with plastic tape
5. Remove compleatly or remove "legs" in those elements (image 1.1). Those legs causes to raise whole spring, leaving 0.5mm gap in which plastic tape will vibrate. Without it plastic tape is well pressured by spring. 
6. If it is not enough 3d print spacer and add it to spring so tape will be additionally pressured. 

[image]
[image]

------------- Microstep accuracy ----------------------
We have replaced step motor driver, not stepper motors itself. On casing orginal ones have marking 24V but it is not
true. They have beed modified to run from 110V generated in power board. It is almost imposible to find one powered
from that supply so we have replaced them with standard one powered from 24V and powered them from external supply. 

We have bought 2DM542. It works well. It also do generates much less vibrations during movements. By default
it run another direction so one must swap one coil terminators. 

[image]

Regards

Iangitpers, 2021-06-26:
By changing the bios settings I was able to boot to a flash drive on my 2015ish Neoden4. This might be a good route to test OpenPnP without opening it to swap the ssd. I plan to give it a shot soon. 

Is it just the camera driver and neoden.dll that prevent it from working with Linux? One of the neoden ceos gave me this pnp gratis and installed it in my office personally. I have them on wechat and can do my best to push for something specific.

MatSpy, 2021-06-27:
Hello Iangitpers, 

Yes, exactly. Neoden dll camera driver is biggest problem. Protocol is reverse engineered, so it is mater of implementation, but it is probably quite time consuming. In linux there is another problem. It seems that there are big gaps while transmitting RS232 data in openpnp (~10ms) which triggers timeout on controller board side. 

I have spoken with Neoden support and they wont help with openpnp. There are still some messages in protocol that we do not know what they are doing. On the other hand they are not necessary for machine to work, since we have all features up and running. 

Regards, 
Mateusz

maxkunes, 2021-06-27:
> By changing the bios settings I was able to boot to a flash drive on my 2015ish Neoden4. This might be a good route to test OpenPnP without opening it to swap the ssd. I plan to give it a shot soon.
> 
> Is it just the camera driver and neoden.dll that prevent it from working with Linux? One of the neoden ceos gave me this pnp gratis and installed it in my office personally. I have them on wechat and can ask for something specific.

The USB stick idea is a big deal. I can't really afford to take apart/installing an SSD in my neoden right now because we are using it. If I could just plug in a USB stick, that would be awesome?

Iangitpers, 2021-06-27:
In my experience the serial delay in Java is a Java issue. A c shim or something might work, I think there are some out there. I'm just getting up to speed though. 

Today I read through the eevblog thread and Charlie Xs reverse post. (Charlie X updated my eagle neoden script! Open source is awesome!)

I grabbed a Windows 7 iso from Microsoft. I made a Windows To Go install on a 64gb USB stick. Got it setup for the neoden (failed on first boot then did a long setup process and worked after that, didn't even know windows To Go was a thing, maybe this is normal). Installed the video drivers for Intel D525 to make things pretty. There is a driver for the Ethernet on a mirror site, but Intel sadly doesn't have them anymore. 

Unfortunately my camera is the older CGImage something. There's a seller on taobao with an sdk, we'll try to get it tomorrow. I will also ask neoden for the driver but your post doesn't give me much hope. Maybe neoden will sell me the more recent camera and I can retrofit. 

Using Windows 7 To Go took an hour or something to setup so I thought it would be unbearably slow, but once it gets past the setup stage it really works fine.

I'm really sad to hear they don't want to work with openPnP. When they brought the machine we had a long discussion about the market they've been successful with and how the community would fill in gaps if given the chance :(

I also think a Linux ISO with OpenPnP ready to be loaded on a flash drive would be a game changer for them. It solves a lot of issues with their software, and dissuades people from mucking about in the internals. 

Did you get the replacement stepper drivers on Ali? I see the same ones from your photo for 22usd or so. I'm going to get two. Have you pushed the machine.xml anywhere? I haven't been able to find a recent push in your repo or openpnp.

Iangitpers, 2021-06-28:
CGImagetech on Taobao gave me the drivers for the camera as well as the SDK. The neoden camera vid/pid isn't in the driver inf so I added it and both cameras install without error. 

[image]

None of the SDK demos find the camera though. Not sure if it's the USB id or some other issue. 

Just for fun I started the neoden software from the main driver under windows 7. It complained about a missing image-something.dll and reported that the cameras were not found. 

I'm going to read through the SDK document and see if I can get it going by selecting the USB id directly.

Iangitpers, 2021-06-28:
I spoke too soon. The answer was in the comments on Charlie's post. I installed the x86 package of Microsoft Visual C++ 2008 redistributable. Now I can run imgdll_test in the neoden4 folder on the data drive. It alternates captures from the top and bottom camera. 

[image]
I tried running the neoden software again and top and bottom cameras both work. The CGImagetech demos do not work though. 

[image]

izimihail, 2021-06-28:
Hi @Iangitpers,

Can you share the driver for the cameras and the SDK.

Thank you,
Mihail

Iangitpers, 2021-06-28:
https://easyupload.io/m/96ybt7

These are the drivers. x86 and 64 bit. Please note I did NOT update the . inf in the x64 drivers because I didn't use them, just including them too.

https://easyupload.io/d3vsjg

Here is the full sdk.

Please note that these links will expire in 30 days. I have no idea what the license is, but it's a Chinese SDK so I assume it's meant to be used and includes right to redistributeables with your own implementation.

izimihail, 2021-06-28:
Hi @Iangitpers,
Thank you for your share. I suppose that, because the cameras are now detected by windows, the openpnp should also be able to get image directly from camera. I will try...

maxkunes, 2021-06-28:
So it sounds like there is no reason why a basic openpnp setup on this machine can't just be a bootable os via USB right? I've got a newer machine so maybe its less of a hassle...

Iangitpers, 2021-06-28:
If you have a machine with the newer camera then the drivers are already implemented and the current code should support it (I haven't found the latest yet, will look over it tomorrow). If you have the CGImagetech camera then the driver still needs to be implemented in openpnp as I understand. 

Yes, it's definitely possible to boot from a USB drive and use the machine. I can even use the original software from Windows 7 on a USB flash drive with the older camera.

maxkunes, 2021-06-28:
> I can even use the original software from Windows 7 on a USB flash drive with the older camera.

Could you explain what you mean by this?

Iangitpers, 2021-06-28:
I boot off a USB flash drive. I installed all the drivers for Windows 7 (just the camera driver and the visual c redistributable). The internal neoden drive is visible as d: and e: drives.  I can navigate to the original neoden software on the internal drive and run it normally. So it's totally possible to run original software and openPnP from the same setup (no disk swapping).  It's really nice to have the unlocked windows 7 desktop in wide screen format. I plan to install an FTP server so I can copy jobs without juggling flash drives. I also have a hex editor and some other tools for poking around. 

I'm sure there's a variety of main boards, but the atom D525 in mine is plenty fast and usable.

Give it a shot! It doesn't change the original setup, and you can still have access to original and openpnp with a nicer workspace :)

maxkunes, 2021-06-28:
> I boot off a USB flash drive. I installed all the drivers for Windows 7 (just the camera driver and the visual c redistributable). The internal neoden drive is visible as d: and e: drives. I can [...]

Oh, that's awesome! I'm going to try that out. I think I'll give windows 10 a shot and just disable driver signature enforcement.

Iangitpers, 2021-06-28:
Depending on your main board (graphics, Ethernet) drivers may not be available for x64 or Windows 10. I'm also not sure how the original software and DLLs behave under windows 10/x64, please let us know.

maxkunes, 2021-06-28:
> Depending on your main board (graphics, Ethernet) drivers may not be available for x64 or Windows 10. I'm also not sure how the original software and DLLs behave under windows 10/x64, please let us know.

I've had a decent experience with windows 10 compatibility over the years, so I'm hoping it won't be too much of an issue. I think it will be worth it if I can get it to work.

[135 more replies]