openpnp/openpnp #1084: Error when saving feeder config

maisonsmd, 2020-12-08:
# Problem Report
## Describe the Issue
After configurating feeder, click apply, nothing happens, exception error appears on log

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Move to feeder section
3. Choose a feeder
4. Make some changes, eg: hole coordinates
5. Click apply

## Expected Result
Changes are saved, apply and reset button gray out

## Actual Result
Changes are not saved, error appears on log

## Log file
```
2020-12-08 20:16:24.696 SystemLogger ERROR: Exception in thread "AWT-EventQueue-0" org.jdesktop.beansbinding.PropertyResolutionException: Exception invoking method public void org.openpnp.spi.base.AbstractFeeder.setPart(org.openpnp.model.Part) on S01 - 0805 1K
2020-12-08 20:16:24.696 SystemLogger ERROR: 	at org.jdesktop.beansbinding.BeanProperty.invokeMethod(BeanProperty.java:791)
2020-12-08 20:16:24.696 SystemLogger ERROR: 	at org.jdesktop.beansbinding.BeanProperty.write(BeanProperty.java:891)
2020-12-08 20:16:24.696 SystemLogger ERROR: 	at org.jdesktop.beansbinding.BeanProperty.setValue(BeanProperty.java:571)
2020-12-08 20:16:24.696 SystemLogger ERROR: 	at org.openpnp.gui.support.JBindings$WrappedBinding.save(JBindings.java:111)
2020-12-08 20:16:24.696 SystemLogger ERROR: 	at org.openpnp.gui.support.AbstractConfigurationWizard.saveToModel(AbstractConfigurationWizard.java:117)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at org.openpnp.gui.support.AbstractConfigurationWizard$1.actionPerformed(AbstractConfigurationWizard.java:176)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.Component.processMouseEvent(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at javax.swing.JComponent.processMouseEvent(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.Component.processEvent(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.Container.processEvent(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.Component.dispatchEventImpl(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.Container.dispatchEventImpl(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.Component.dispatchEvent(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.Container.dispatchEventImpl(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.Window.dispatchEventImpl(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.Component.dispatchEvent(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.EventQueue.access$500(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(Unknown Source)
2020-12-08 20:16:24.697 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.security.AccessController.doPrivileged(Native Method)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.awt.EventQueue$4.run(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.awt.EventQueue$4.run(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.security.AccessController.doPrivileged(Native Method)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEvent(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at org.openpnp.gui.MainFrame$2.dispatchEvent(MainFrame.java:464)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at java.awt.EventDispatchThread.run(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: Caused by: java.lang.reflect.InvocationTargetException
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
2020-12-08 20:16:24.698 SystemLogger ERROR: 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
2020-12-08 20:16:24.699 SystemLogger ERROR: 	at java.lang.reflect.Method.invoke(Unknown Source)
2020-12-08 20:16:24.699 SystemLogger ERROR: 	at org.jdesktop.beansbinding.BeanProperty.invokeMethod(BeanProperty.java:782)
2020-12-08 20:16:24.699 SystemLogger ERROR: 	... 42 more
2020-12-08 20:16:24.699 SystemLogger ERROR: Caused by: java.lang.NullPointerException
2020-12-08 20:16:24.699 SystemLogger ERROR: 	at org.openpnp.spi.base.AbstractFeeder.setPart(AbstractFeeder.java:73)
2020-12-08 20:16:24.699 SystemLogger ERROR: 	... 47 more
```

markmaker, 2020-12-08:
Hi
I guess this happens if you deleted a part that was formerly assigned to that feeder.  You need to set a new part. 
_Mark

maisonsmd, 2020-12-08:
Found the culprit, if the feeder is not set to a specific part, NullPointerException will occur.
I have some feeders that empty, I created them as placeholder.
A hack in the meantime maybe point the feeder to some random part and disable it.

closed 2020-12-08 by maisonsmd