openpnp/openpnp #1085: openpnp on remote device with x11 forwarding

nerdyscout, 2020-12-11:
# Problem Report
## I installed OpenPnP on a remote device and tried to use X11 forwarding

## Steps to Reproduce
installed OpenPnP on a raspberrypi following this procedure
1. wget https://s3-us-west-2.amazonaws.com/openpnp/OpenPnP-unix-develop.sh
2. chmod +x OpenPnP-unix-develop.sh
3. ./OpenPnP-unix-develop.sh
4. add to /etc/ssh/ssh_config
```
    ForwardX11 yes
    X11UseLocalhost no
```
5. restart sshd
6. run from different computer:
```
ssh user@ip -X /home/user/openpnp/OpenPnP
```

## Expected Result
running OpenPnP remotely

## Actual Result
1. error dialog pops up
[image]
2. on the terminal I can see
```
2020-12-11 20:48:14.489 Configuration INFO: No packages.xml found in configuration directory, loading defaults.
2020-12-11 20:48:15.517 Configuration INFO: No parts.xml found in configuration directory, loading defaults.
2020-12-11 20:48:15.585 Configuration INFO: No machine.xml found in configuration directory, loading defaults.
```
a folder .openpnp2 is created in the home folder on the remote device, but the three files are missing

.openpnp2/log/OpenPnP.log contains

```
2020-12-11 20:50:22.700 Configuration INFO: No packages.xml found in configuration directory, loading defaults.
2020-12-11 20:50:23.756 Configuration INFO: No parts.xml found in configuration directory, loading defaults.
2020-12-11 20:50:23.823 Configuration INFO: No machine.xml found in configuration directory, loading defaults.
2020-12-11 20:50:24.029 SystemLogger ERROR: java.lang.Exception: Error while reading machine.xml (null)
2020-12-11 20:50:24.032 SystemLogger ERROR: 	at org.openpnp.model.Configuration.load(Configuration.java:287)
2020-12-11 20:50:24.034 SystemLogger ERROR: 	at org.openpnp.gui.MainFrame.<init>(MainFrame.java:656)
2020-12-11 20:50:24.036 SystemLogger ERROR: 	at org.openpnp.Main$1.run(Main.java:152)
2020-12-11 20:50:24.038 SystemLogger ERROR: 	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
2020-12-11 20:50:24.039 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
2020-12-11 20:50:24.041 SystemLogger ERROR: 	at java.awt.EventQueue.access$500(EventQueue.java:97)
2020-12-11 20:50:24.043 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(EventQueue.java:709)
2020-12-11 20:50:24.045 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(EventQueue.java:703)
2020-12-11 20:50:24.047 SystemLogger ERROR: 	at java.security.AccessController.doPrivileged(Native Method)
2020-12-11 20:50:24.048 SystemLogger ERROR: 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
2020-12-11 20:50:24.051 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
2020-12-11 20:50:24.053 SystemLogger ERROR: 	at org.GNOME.Accessibility.AtkWrapper$6.dispatchEvent(AtkWrapper.java:705)
2020-12-11 20:50:24.055 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
2020-12-11 20:50:24.057 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
2020-12-11 20:50:24.060 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
2020-12-11 20:50:24.062 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
2020-12-11 20:50:24.064 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
2020-12-11 20:50:24.066 SystemLogger ERROR: 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
2020-12-11 20:50:24.069 SystemLogger ERROR: Caused by: java.lang.reflect.InvocationTargetException
2020-12-11 20:50:24.071 SystemLogger ERROR: 	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
2020-12-11 20:50:24.073 SystemLogger ERROR: 	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
2020-12-11 20:50:24.075 SystemLogger ERROR: 	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
2020-12-11 20:50:24.077 SystemLogger ERROR: 	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
2020-12-11 20:50:24.079 SystemLogger ERROR: 	at org.simpleframework.xml.core.InstanceFactory.getObject(InstanceFactory.java:100)
2020-12-11 20:50:24.081 SystemLogger ERROR: 	at org.simpleframework.xml.core.InstanceFactory$ClassInstance.getInstance(InstanceFactory.java:233)
2020-12-11 20:50:24.084 SystemLogger ERROR: 	at org.simpleframework.xml.core.ObjectInstance.getInstance(ObjectInstance.java:95)
2020-12-11 20:50:24.086 SystemLogger ERROR: 	at org.simpleframework.xml.core.ObjectInstance.getInstance(ObjectInstance.java:76)
2020-12-11 20:50:24.088 SystemLogger ERROR: 	at org.simpleframework.xml.core.Composite$Builder.read(Composite.java:1376)
2020-12-11 20:50:24.091 SystemLogger ERROR: 	at org.simpleframework.xml.core.Composite.read(Composite.java:201)
2020-12-11 20:50:24.093 SystemLogger ERROR: 	at org.simpleframework.xml.core.Composite.read(Composite.java:148)
[... 13 lines cut ...]
2020-12-11 20:50:24.124 SystemLogger ERROR: 	at org.simpleframework.xml.core.Persister.read(Persister.java:543)
2020-12-11 20:50:24.126 SystemLogger ERROR: 	at org.simpleframework.xml.core.Persister.read(Persister.java:521)
2020-12-11 20:50:24.127 SystemLogger ERROR: 	at org.simpleframework.xml.core.Persister.read(Persister.java:426)
2020-12-11 20:50:24.129 SystemLogger ERROR: 	at org.openpnp.model.Configuration.loadMachine(Configuration.java:416)
2020-12-11 20:50:24.131 SystemLogger ERROR: 	at org.openpnp.model.Configuration.load(Configuration.java:280)
2020-12-11 20:50:24.133 SystemLogger ERROR: 	... 17 more
2020-12-11 20:50:24.136 SystemLogger ERROR: Caused by: java.lang.ExceptionInInitializerError
2020-12-11 20:50:24.138 SystemLogger ERROR: 	at nu.pattern.OpenCV$SharedLoader.getInstance(OpenCV.java:232)
2020-12-11 20:50:24.140 SystemLogger ERROR: 	at nu.pattern.OpenCV.loadShared(OpenCV.java:181)
2020-12-11 20:50:24.141 SystemLogger ERROR: 	at org.openpnp.vision.pipeline.CvPipeline.<clinit>(CvPipeline.java:49)
2020-12-11 20:50:24.143 SystemLogger ERROR: 	at org.openpnp.machine.reference.vision.ReferenceFiducialLocator.createDefaultPipeline(ReferenceFiducialLocator.java:496)
2020-12-11 20:50:24.146 SystemLogger ERROR: 	at org.openpnp.machine.reference.vision.ReferenceFiducialLocator.<init>(ReferenceFiducialLocator.java:64)
2020-12-11 20:50:24.148 SystemLogger ERROR: 	at org.openpnp.machine.reference.ReferenceMachine.<init>(ReferenceMachine.java:92)
2020-12-11 20:50:24.151 SystemLogger ERROR: 	... 46 more
2020-12-11 20:50:24.154 SystemLogger ERROR: Caused by: java.lang.NullPointerException
2020-12-11 20:50:24.156 SystemLogger ERROR: 	at java.util.Objects.requireNonNull(Objects.java:203)
2020-12-11 20:50:24.157 SystemLogger ERROR: 	at java.nio.file.Files.copy(Files.java:2984)
2020-12-11 20:50:24.159 SystemLogger ERROR: 	at nu.pattern.OpenCV.extractNativeBinary(OpenCV.java:382)
2020-12-11 20:50:24.160 SystemLogger ERROR: 	at nu.pattern.OpenCV.extractNativeBinary(OpenCV.java:317)
2020-12-11 20:50:24.161 SystemLogger ERROR: 	at nu.pattern.OpenCV.access$100(OpenCV.java:22)
2020-12-11 20:50:24.163 SystemLogger ERROR: 	at nu.pattern.OpenCV$SharedLoader.<init>(OpenCV.java:204)
2020-12-11 20:50:24.164 SystemLogger ERROR: 	at nu.pattern.OpenCV$SharedLoader.<init>(OpenCV.java:187)
2020-12-11 20:50:24.166 SystemLogger ERROR: 	at nu.pattern.OpenCV$SharedLoader$Holder.<clinit>(OpenCV.java:228)
2020-12-11 20:50:24.167 SystemLogger ERROR: 	... 52 more
```

vonnieda, 2020-12-11:
This error is due to opencv not loading correctly on Raspberry Pi. Raspberry Pi is not currently a supported platform. There is work ongoing to support it in https://github.com/openpnp/opencv/pull/59

nerdyscout, 2020-12-11:
wow this was a quick reply! thank you :)

I assume the fact I am actually using an orangePi (did say raspberry because it is more known) does not help much...?

ian-arkver, 2020-12-11:
FYI, once you get OpenPnP running the X forwarding over ssh should work fine. I use it myself with a remote Jetson Nano. Getting it to run on an OrangePi might be tricky, and I've no helpful experience with armbian.

G-Pereira, 2021-02-10:
@ian-arkver Hey! I am trying to do the same (run on a Jetson Nano) and I am getting the above error. Can you share your procedure?

G-Pereira, 2021-02-10:
@ian-arkver actuallly I am getting errors loading the CvPipeline library:
[image]

nerdyscout, 2021-02-10:
I solved this issue for me by exchanging hardware.

closed 2021-02-10 by nerdyscout