openpnp/openpnp #1091: MOVE_TO_COMPLETED_COMMAND sent at wrong place

maisonsmd, 2020-12-26:
# Problem Report
## Describe the Issue
I just updated the program to the latest build (2020-12-12), and noticed a weird problem:
In the previous build, MOVE_TO_COMPLETE_COMMAND (M400) is sent after MOVE_TO_COMMAND is confirmed by the machine, but now it doesn't, instead, it's sent after HOME_COMMAND (in the previous build it's not).

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Connect to machine
3. Home
3. Jog the machine 10mm to the right.
4. MOVE_TO_COMPLETE_COMMAND not being sent

## Expected Result
HOME_COMMAND only expect confirmation once, software won't send MOVE_TO_COMPLETED_COMMAND after this
MOVE_TO_COMPLETE_COMMAND is sent after MOVE_TO_COMMAND

## Actual Result
MOVE_TO_COMPLETE_COMMAND is sent after HOME_COMMAND 
MOVE_TO_COMPLETE_COMMAND is not sent after MOVE_TO_COMMAND

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
communication log:
```
22:27:32.462 A->PC G28 
22:27:33.750 PC->A ok
22:27:33.757 A->PC M400
22:27:34.649 PC->A ok
22:27:43.540 A->PC G0 X10.00    F1000
22:27:45.289 PC->A ok
22:28:01.889 A->PC G0 X20.00    F1000 ; << where is M400 now?
22:28:03.478 PC->A ok
```
Is there any change in the latest build that I'm missing? 
I've already filled MOVE_TO_COMPLETE_COMMAND and checked "Allow Pre-Move Commands".

markmaker, 2020-12-26:
Hi @maisonsmd 

All this is intended and for good reasons.

1. The whole Advanced Motion Control update is about optimizing motion in OpenPnP i.e. make it more fluid, faster, more controlled.
  https://github.com/openpnp/openpnp/wiki/Advanced-Motion-Control
2. MOVE_TO_COMPLETE_COMMAND is now only sent, when functionally required, i.e. when OpenPnP really needs to wait. Otherwise, it sends multiple move commands and lets the controller work as fast as permitted, and in parallel to OpenPnP going on planning the next moves. Example: a wait is not needed until you really want to do Computer Vision. The full list is here: https://github.com/openpnp/openpnp/wiki/Motion-Planner#motion-path-planning 
3. Sending it after HOME_COMMAND is needed in case you home across multiple drivers/controllers. Strictly speaking, this was missing before. 
4. If you Jog, it does not send the M400, to allow for continuous jogging, If you press the Arrows quickly, the machine will keep on going without slowing down in between moves (like in other programs). No more stuttering 1mm moves. 
5. However, if the move is commanded in a job or for a GUI action, then MOVE_TO_COMPLETE_COMMANDs will be sent as needed and explained above.

_Mark

maisonsmd, 2020-12-26:
Thank you I see it now. Looking forward to the new motion blending.

closed 2020-12-26 by maisonsmd