openpnp/openpnp #1094: Submit Diagnostics Broken

vonnieda, 2020-12-30:
Submit Diagnostics is failing to post to pastebin. Not clear why - just a post error:

```
2020-12-30 00:22:45.003 SystemLogger ERROR: java.lang.RuntimeException: Error posting to https://pastebin.com/api/api_post.php using the params: api_paste_code=OpenPnP+Diagnostics%0A%0AImages%3A+http%3A%2F%2Fimgur.com%2Fa%2F9OAiExN%0A%0ADescription%3A+test%0A%0AFiles%3A%0A%0A****+machine.xml+****%0A%0A%3Copenpnp-machine%3E%0A+++%3Cmachine+class%3D%22org.openpnp.machine.reference.ReferenceMachine%22+speed%3D%221.0%22%3E%0A++++++%3Caxes%3E%0A+++++++++%3Caxis+class%3D%22org.openpnp.machine.reference.axis.ReferenceControllerAxis%22+id%3D%22AXS16556a53f42e088c%22+name%3D%22x%22+type%3D%22X%22+driver-id%3D%22DRV16556a53f40b9e52%22+backlash-compensation-method%3D%22None%22+backlash-speed-factor%3D%220.1%22+limit-rotation%3D%22false%22+wrap-around-rotation%3D%22false%22+invert-linear-rotational%3D%22false%22+soft-limit-low-enabled%3D%22false%22+soft-limit-high-enabled%3D%22false%22+safe-zone-low-enabled%3D%22false%22+safe-zone-high-enabled%3D%22false%22%3E%0A++++++++++++%3Chome-coordinate+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cbacklash-offset+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csoft-limit-low+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csoft-limit-high+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csafe-zone-low+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csafe-zone-high+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cfeedrate-per-second+value%3D%22333.3333333333333%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cacceleration-per-second-2+value%3D%22666.6666666666666%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cjerk-per-second-3+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cresolution%3E1.0E-4%3C%2Fresolution%3E%0A+++++++++%3C%2Faxis%3E%0A+++++++++%3Caxis+class%3D%22org.openpnp.machine.reference.axis.ReferenceControllerAxis%22+id%3D%22AXS16556a53f4375607%22+name%3D%22y%22+type%3D%22Y%22+driver-id%3D%22DRV16556a53f40b9e52%22+backlash-compensation-method%3D%22None%22+backlash-speed-factor%3D%220.1%22+limit-rotation%3D%22false%22+wrap-around-rotation%3D%22false%22+invert-linear-rotational%3D%22false%22+soft-limit-low-enabled%3D%22false%22+soft-limit-high-enabled%3D%22false%22+safe-zone-low-enabled%3D%22false%22+safe-zone-high-enabled%3D%22false%22%3E%0A++++++++++++%3Chome-coordinate+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cbacklash-offset+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csoft-limit-low+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csoft-limit-high+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csafe-zone-low+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csafe-zone-high+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cfeedrate-per-second+value%3D%22333.3333333333333%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cacceleration-per-second-2+value%3D%22666.6666666666666%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cjerk-per-second-3+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cresolution%3E1.0E-4%3C%2Fresolution%3E%0A+++++++++%3C%2Faxis%3E%0A+++++++++%3Caxis+class%3D%22org.openpnp.machine.reference.axis.ReferenceVirtualAxis%22+id%3D%22AXS16556a53f43921a0%22+name%3D%22zTop%22+type%3D%22Z%22%3E%0A++++++++++++%3Chome-coordinate+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A+++++++++%3C%2Faxis%3E%0A+++++++++%3Caxis+class%3D%22org.openpnp.machine.reference.axis.ReferenceVirtualAxis%22+id%3D%22AXS16556a53f439b22c%22+name%3D%22rotationTop%22+type%3D%22Rotation%22%3E%0A++++++++++++%3Chome-coordinate+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A+++++++++%3C%2Faxis%3E%0A+++++++++%3Caxis+class%3D%22org.openpnp.machine.reference.axis.ReferenceControllerAxis%22+id%3D%22AXS16556a53f43a302a%22+name%3D%22zN1%22+type%3D%22Z%22+driver-id%3D%22DRV16556a53f40b9e52%22+backlash-compensation-method%3D%22None%22+backlash-speed-factor%3D%220.1%22+limit-rotation%3D%22false%22+wrap-around-rotation%3D%22false%22+invert-linear-rotational%3D%22false%22+soft-limit-low-enabled%3D%22false%22+soft-limit-high-enabled%3D%22false%22+safe-zone-low-enabled%3D%22true%22+safe-zone-high-enabled%3D%22true%22%3E%0A++++++++++++%3Chome-coordinate+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cbacklash-offset+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csoft-limit-low+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csoft-limit-high+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csafe-zone-low+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csafe-zone-high+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cfeedrate-per-second+value%3D%22333.3333333333333%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cacceleration-per-second-2+value%3D%22666.6666666666666%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cjerk-per-second-3+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cresolution%3E1.0E-4%3C%2Fresolution%3E%0A+++++++++%3C%2Faxis%3E%0A+++++++++%3Caxis+class%3D%22org.openpnp.machine.reference.axis.ReferenceControllerAxis%22+id%3D%22AXS16556a53f43a8e1d%22+name%3D%22rotationN1%22+type%3D%22Rotation%22+driver-id%3D%22DRV16556a53f40b9e52%22+backlash-compensation-method%3D%22None%22+backlash-speed-factor%3D%220.1%22+limit-rotation%3D%22true%22+wrap-around-rotation%3D%22false%22+invert-linear-rotational%3D%22false%22+soft-limit-low-enabled%3D%22false%22+soft-limit-high-enabled%3D%22false%22+safe-zone-low-enabled%3D%22false%22+safe-zone-high-enabled%3D%22false%22%3E%0A++++++++++++%3Chome-coordinate+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cbacklash-offset+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csoft-limit-low+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csoft-limit-high+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csafe-zone-low+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Csafe-zone-high+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cfeedrate-per-second+value%3D%223333.3333333333335%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cacceleration-per-second-2+value%3D%226666.666666666667%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cjerk-per-second-3+value%3D%220.0%22+units%3D%22Millimeters%22%2F%3E%0A++++++++++++%3Cresolution%3E1.0E-4%3C%2Fresolution%3E%0A+++++++++%3C%2Faxis%3E%0A++++++%3C%2Faxes%3E%0A++++++%3Cheads%3E%0A+++++++++%3Chead+class%3D%22org.openpnp.machine.reference.ReferenceHead%22+id%3D%22H1%22+name%3D%22H1%22+visual-homing-method%3D%22None%22%3E%0A++++++++++++%3Cnozzles%3E%0A+++++++++++++++%3Cnozzle+class%3D%22org.openpnp.machine.reference.ReferenceNozzle%22+axis-X-id%3D%22AXS16556a53f42e088c%22+axis-Y-id%3D%22AXS16556a53f4375607%22+axis-Z-id%3D%22AXS16556a53f43a302a%22+axis-rotation-id%3D%22AXS16556a53f43a8e1d%22+id%3D%22N1%22+name%3D%22N1%22+pick-dwell-milliseconds%3D%220%22+place-dwell-milliseconds%3D%220%22+current-nozzle-tip-id%3D%22NT1%22+changer-enabled%3D%22false%22+nozzle-tip-changed-on-manual-feed%3D%22false%22+enable-dynamic-safe-z%3D%22false%22+limit-rotation%3D%22true%22%3E%0A++++++++++++++++++%3Ccompatible-nozzle-tip-ids+class%3D%22java.util.ArrayList%22%3E%0A+++++++++++++++++++++%3Cstring%3ENT1%3C%2Fstring%3E%0A++++++++++++++++++%3C%2Fcompatible-nozzle-tip-ids%3E%0A++++++++++++++++++%3Chead-offsets+units%3D%22Millimeters%22+x%3D%220.0%22+y%3D%220.0%22+z%3D%220.0%22+rotation%3D%220.0%22%2F%3E%0A++++++++++++++++++%3Cvacuum-sense-actuator-name%3EA1%3C%2Fvacuum-sense-actuator-name%3E%0A++++++++++++++++++%3Cvacuum-actuator-name%3EA1%3C%2Fvacuum-actuator-name%3E%0A+++++++++++++++%3C%2Fnozzle%3E%0A++++++++++++%3C%2Fnozzles%3E%0A++++++++++++%3Cactuators%3E%0A+++++++++++++++%3Cactuator+class%3D%22org.openpnp.machine.reference.ReferenceActuator%22+id%3D%22A1%22+name%3D%22A1%22+interlock-actuator%3D%22false%22+coordinated-before-actuate%3D%22true%22+coordinated-after-actuate%3D%22false%22+coordinated-before-read%3D%22true%22+index%3D%220%22%3E%0A++++++++++++++++++%3Chead-offsets+units%3D%22Millimeters%22+x%3D%220.0%22+y%3D%220.0%22+z%3D%220.0%22+rotation%3D%220.0%22%2F%3E%0A+++++++++++++++%3C%2Factuator%3E%0A++++++++++++%3C%2Factuators%3E%0A++++++++++++%3Ccameras%3E%0A++++++++ [... cut ...]
2020-12-30 00:22:45.004 SystemLogger ERROR: 	at com.github.kennedyoliveira.pastebin4j.api.WebUtils.post(WebUtils.java:86)
2020-12-30 00:22:45.004 SystemLogger ERROR: 	at com.github.kennedyoliveira.pastebin4j.api.PasteBinApiImpl.createPaste(PasteBinApiImpl.java:137)
2020-12-30 00:22:45.005 SystemLogger ERROR: 	at com.github.kennedyoliveira.pastebin4j.PasteBin.createPaste(PasteBin.java:87)
2020-12-30 00:22:45.005 SystemLogger ERROR: 	at org.openpnp.gui.SubmitDiagnosticsDialog$1.lambda$0(SubmitDiagnosticsDialog.java:346)
2020-12-30 00:22:45.005 SystemLogger ERROR: 	at java.lang.Thread.run(Thread.java:748)
```

vonnieda, 2020-12-30:
It's probably time to spin up an S3 bucket and Heroku app or something for storing these instead of relying on services that don't want them.

brainstorm, 2021-05-14:
I found and used this one the other day:

http://sprunge.us/

if you cannot be bothered to host it yourself, this could work well for a while I reckon?: 

https://stackoverflow.com/questions/4013947/how-to-use-pastebin-from-shell-script/24130329#24130329

vonnieda, 2021-05-14:
That looks interesting, and gets us part way there, but I'd really love to find something that can handle image uploads as well as text. Right now we push the images to imgur and the text to pastebin, and it would be nice if those could be combined into one small script.