openpnp/openpnp #1101: Altium import heading decode

DavidG-Aus, 2021-01-06:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
The board import function does not recognise the current Altium AD20 CSV headings

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
This is a sample CSV file from AD20 (AD21 has just been released but I don't have it yet).

Altium Designer Pick and Place Locations
D:\PCB\AAAA\Aligner\.\Pick Place\Pick Place for Aligner.csv

========================================================================================================================
File Design Information:

Date:       07/01/21
Time:       10:17
Revision:   Not in VersionControl
Variant:    No variations
Units used: mm

"Designator","Comment","Layer","Footprint","Center-X(mm)","Center-Y(mm)","Rotation","Description","Height(mm)"
"R21","33k","TopLayer","1206","54.4576","64.4428","270","Resistor 1206 1/4W","0.0000"
"J2","LINK6","TopLayer","IDC6","50.0126","17.3990","180","6 pin link","0.0000"
"R20","4k7","TopLayer","1206","52.9336","59.4136","90","Resistor 1206 1/4W","0.0000"
"R19","22k","TopLayer","1206","52.8858","78.4352","360","Resistor 1206 1/4W","0.0000"

## Expected Result
The X & Y headings are not being decoded.

## Actual Result
Import fails.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
From OpenPNP log:

2021-01-07 10:20:52.225 NamedCSVImporter TRACE: checkLine: "Designator","Comment","Layer","Footprint","Center-X(mm)","Center-Y(mm)","Rotation","Description","Height(mm)"
2021-01-07 10:20:52.237 NamedCSVImporter TRACE: checkCSV: DESIGNATOR = 0
2021-01-07 10:20:52.238 NamedCSVImporter TRACE: checkCSV: COMMENT = 1
2021-01-07 10:20:52.238 NamedCSVImporter TRACE: checkCSV: FOOTPRINT = 3
2021-01-07 10:20:52.238 NamedCSVImporter TRACE: checkCSV: Ref = 0
2021-01-07 10:20:52.238 NamedCSVImporter TRACE: checkCSV: Val = 1
2021-01-07 10:20:52.238 NamedCSVImporter TRACE: checkCSV: Pack = 3
2021-01-07 10:20:52.238 NamedCSVImporter TRACE: checkCSV: X = -1
2021-01-07 10:20:52.238 NamedCSVImporter TRACE: checkCSV: Y = -1
2021-01-07 10:20:52.238 NamedCSVImporter TRACE: checkCSV: Rot = -1
2021-01-07 10:20:52.239 NamedCSVImporter TRACE: checkCSV: TB = -1
2021-01-07 10:20:52.239 NamedCSVImporter TRACE: checkCSV: HT = -1
Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
Enhance the decode algorithm.

## How Will It Look?
None

## Why Do We Need It?
1. Altium Designer import failure.
2. All users needing to import from Altium designer.

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

richard-sim, 2021-01-08:
Looks like the issue is that Ref-X(mm) and Ref-Y(mm) changed to Center-X(mm) and Center-Y(mm) respectively:
https://github.com/openpnp/openpnp/blob/8f290c2ced797bf3c65d585c49e0eacef3c56057/src/main/java/org/openpnp/gui/importer/NamedCSVImporter.java#L118

DavidG-Aus, 2021-01-08:
Altium lets you select several items you can output in the P&P file.  Ref X is different to the Centre X. Ref is a Reference point on the component that is often pin 1 and not the centre of the component.

richard-sim, 2021-01-08:
I just caught myself up on the deal here, as it's been a while.

The [geometric] Center-X/Y isn't what you always want to be using as some components, usually connectors, the pick location on the part is not the same due to weight distribution or just odd geometry. Ref-X/Y is just the origin of the footprint, so is user-defined and thus can take into account odd footprints. Of course, as you point out, some people use pin 1 as the reference point so that the footprint is aligned to the grid when moved.

Personally all my footprints are built using the pick-location as the origin. Looking around for what others do, the Celestial Altium Library (https://github.com/issus/altium-library) does the same, for the same reason.
> Every part's center position is where a pick and place machine's head should collect the component. For companies running an in house Pick and Place machine, this is very convenient and allows faster machine setup. Your pick and place export list for products built with the Celestial Altium Library contains centers in the correct location.

Not sure what to do here, as I think the current behaviour of OpenPnP is correct, yet it's a PITA for people such as yourself. But likewise, if it was changed then it's a PITA for others (and breaks those 'odd' off-center components).

I think the simplest solution would be to support both methods in Altium: prefer Ref-X/Y (current behaviour), and fall back to Center-X/Y (proposed). It should be as simple as adding Center-X/Y in the list(s) after Ref-X/Y. Then document the behaviour somewhere. :)

DavidG-Aus, 2021-01-08:
Yes I agree - I think OpenPNP should parse whatever is in the CSV file and the user chooses which value to export as they see fit.
On a related topic:
It would also be nice if the doco mentions that it is necessary to set the origin in Altium to the reference point of the board (Edit/Origin/Set Origin)- took me while to figure that out, as the default origin is the bottom left of the designer area not the board. I was expecting OpenPNP to have an origin offset setting somewhere - which may not be a bad idea so people can deal with CSV files that other people have created incorrectly?

markmaker, 2023-05-02:
Should be solved through  #1549, thank you @janm012012!

closed 2023-05-02 by markmaker