openpnp/openpnp #1102: Unable to visit Feeders tab

maisonsmd, 2021-01-14:
# Problem Report
## Describe the Issue
OpenPnP randomly fails to open Feeders tab, it just hangs and does nothing.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Switch to Feeders tab
3. Fail to open, errors found in log.

## Expected Result
Switch to Feeders tab, show user's feeders

## Actual Result
Software hang
Sometimes it fails immediately right after software starts, sometimes it fails after a period of using time

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
I tried delete all feeders and add one by one again, that didn't help
Here is the log

2021-01-14 11:18:29.312 SystemLogger ERROR: Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at org.openpnp.gui.FeedersPanel$2.getTableCellRendererComponent(FeedersPanel.java:201)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.JTable.prepareRenderer(Unknown Source)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.plaf.basic.BasicTableUI.paintCell(Unknown Source)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.plaf.basic.BasicTableUI.paintCells(Unknown Source)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.plaf.basic.BasicTableUI.paint(Unknown Source)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.plaf.ComponentUI.update(Unknown Source)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.JComponent.paintComponent(Unknown Source)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.JComponent.paintChildren(Unknown Source)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.JViewport.paint(Unknown Source)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.JComponent.paintChildren(Unknown Source)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.JComponent.paintChildren(Unknown Source)
2021-01-14 11:18:29.312 SystemLogger ERROR: 	at javax.swing.JSplitPane.paintChildren(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.JComponent.paintChildren(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.JComponent.paintChildren(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.JComponent.paintChildren(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.JSplitPane.paintChildren(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.JComponent.paintToOffscreen(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.RepaintManager$PaintManager.paint(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.RepaintManager.paint(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.JComponent._paintImmediately(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.JComponent.paintImmediately(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.RepaintManager$4.run(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.RepaintManager$4.run(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at java.security.AccessController.doPrivileged(Native Method)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.RepaintManager.prePaintDirtyRegions(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.RepaintManager.access$1200(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at javax.swing.RepaintManager$ProcessingRunnable.run(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at java.awt.EventQueue.access$500(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at java.security.AccessController.doPrivileged(Native Method)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEvent(Unknown Source)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at org.openpnp.gui.MainFrame$2.dispatchEvent(MainFrame.java:467)
2021-01-14 11:18:29.313 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
[... 34 lines cut ...]
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at javax.swing.JComponent.paintImmediately(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at javax.swing.RepaintManager$4.run(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at javax.swing.RepaintManager$4.run(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at java.security.AccessController.doPrivileged(Native Method)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at javax.swing.RepaintManager.prePaintDirtyRegions(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at javax.swing.RepaintManager.access$1200(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at javax.swing.RepaintManager$ProcessingRunnable.run(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at java.awt.EventQueue.access$500(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(Unknown Source)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at java.security.AccessController.doPrivileged(Native Method)
2021-01-14 11:18:31.596 SystemLogger ERROR: 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
2021-01-14 11:18:31.597 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEvent(Unknown Source)
2021-01-14 11:18:31.597 SystemLogger ERROR: 	at org.openpnp.gui.MainFrame$2.dispatchEvent(MainFrame.java:467)
2021-01-14 11:18:31.597 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
2021-01-14 11:18:31.597 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
2021-01-14 11:18:31.597 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
2021-01-14 11:18:31.597 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
2021-01-14 11:18:31.597 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
2021-01-14 11:18:31.597 SystemLogger ERROR: 	at java.awt.EventDispatchThread.run(Unknown Source)

markmaker, 2021-01-14:
Hi @maisonsmd 

It seems that this error happens, if you have a feeder that does not have a part assigned. This may be the case after you delete the part that was assigned. 

You can access feeders both from the Machine Setup tab and from the Feeders tab, so this may be a way to assign a new part to the feeder that may cause the crash. 

That's a bug obviously **and this issue should stay open**, but the info might help you to set it right for now. 

_Mark

RRudi92, 2021-01-20:
Experienced the Same Issue. 
All Feeders have a Part assigned.
The Feeders-Tab can be accessed after OpenPnP is started, but after a Job is loaded, the feeders tab is no longer accessable.

The Problem is even worse, because if a Feeder runs out in the middle of a Job, the feeder is set to dissabled -> the programm has to be restarted to reenable the feeder.

markmaker, 2021-01-20:
Could you post the log?
_Mark

RRudi92, 2021-01-20:
It may take a few days, before I will have access to the machine again. 
Do you need some other Informations for debugging?

RRudi92, 2021-02-17:
As of Version 2021-02-04_06-49-15.d539596 this Issue persists (in a different manner). 
### Steps to reproduce:
A Board file must be loaded (*.board.xml) that contains at least **one missing part that is set to "placement"** (No part assigned / Part field empty). If this is the case, the "feeders" tab can't be accessed.
### Actual Result
The feeders Tab will not be accessible. Error Log shown below. No Crashes.
### Problem solving
The board file must not have empty Parts fields. Either delete this line (part), or set to fiducial, or assign a part. After that the feeders tab can be accessed without Issues. 

I Hope this information helps to debug this issue.

###Error Log:

`2021-02-17 16:22:01.000 SystemLogger ERROR: Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException

2021-02-17 16:22:01.001 SystemLogger ERROR: 	at org.openpnp.gui.FeedersPanel$2.getTableCellRendererComponent(FeedersPanel.java:201)

2021-02-17 16:22:01.001 SystemLogger ERROR: 	at javax.swing.JTable.prepareRenderer(Unknown Source)

2021-02-17 16:22:01.001 SystemLogger ERROR: 	at javax.swing.plaf.basic.BasicTableUI.paintCell(Unknown Source)

2021-02-17 16:22:01.001 SystemLogger ERROR: 	at javax.swing.plaf.basic.BasicTableUI.paintCells(Unknown Source)

2021-02-17 16:22:01.001 SystemLogger ERROR: 	at javax.swing.plaf.basic.BasicTableUI.paint(Unknown Source)

2021-02-17 16:22:01.001 SystemLogger ERROR: 	at javax.swing.plaf.ComponentUI.update(Unknown Source)

2021-02-17 16:22:01.001 SystemLogger ERROR: 	at javax.swing.JComponent.paintComponent(Unknown Source)

2021-02-17 16:22:01.001 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)

2021-02-17 16:22:01.001 SystemLogger ERROR: 	at javax.swing.JComponent.paintChildren(Unknown Source)

2021-02-17 16:22:01.001 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)

2021-02-17 16:22:01.001 SystemLogger ERROR: 	at javax.swing.JViewport.paint(Unknown Source)

2021-02-17 16:22:01.001 SystemLogger ERROR: 	at javax.swing.JComponent.paintChildren(Unknown Source)

2021-02-17 16:22:01.001 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.JComponent.paintChildren(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.JSplitPane.paintChildren(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.JComponent.paintChildren(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.JComponent.paintChildren(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.JComponent.paintChildren(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.JSplitPane.paintChildren(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.JComponent.paint(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.JComponent.paintToOffscreen(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.RepaintManager$PaintManager.paint(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.RepaintManager.paint(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at javax.swing.JComponent._paintImmediately(Unknown Source)

[... 26 lines cut ...]
2021-02-17 16:22:01.002 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at java.awt.EventQueue$3.run(Unknown Source)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at java.security.AccessController.doPrivileged(Native Method)

2021-02-17 16:22:01.002 SystemLogger ERROR: 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)

2021-02-17 16:22:01.003 SystemLogger ERROR: 	at java.awt.EventQueue.dispatchEvent(Unknown Source)

2021-02-17 16:22:01.003 SystemLogger ERROR: 	at org.openpnp.gui.MainFrame$2.dispatchEvent(MainFrame.java:467)

2021-02-17 16:22:01.003 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)

2021-02-17 16:22:01.003 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)

2021-02-17 16:22:01.003 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)

2021-02-17 16:22:01.003 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)

2021-02-17 16:22:01.003 SystemLogger ERROR: 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)

2021-02-17 16:22:01.003 SystemLogger ERROR: 	at java.awt.EventDispatchThread.run(Unknown Source)

`

markmaker, 2021-02-17:
Hi @RRudi92 

Thanks for the report.

The bug was so far only fixed in the **testing** version. It will be merged into the regular version soon.

Your version d539596 is from the **regular** version (develop branch):
https://github.com/openpnp/openpnp/actions/runs/536295637

I would of course be very grateful if you could upgrade to the testing version and help us test this!

;-) 

See here:
https://groups.google.com/g/openpnp/c/DnpvszIEA5Q/m/mgmoZx-BBgAJ

_Mark

RRudi92, 2021-02-17:
In Version 2021-02-08_07-41-03.fd3eeb9 this bug is fixed. 
If I will have time, I will use these versio (or subsequent Testversions) an report bugs if I spot any.

markmaker, 2021-02-17:
Great thanks!

markmaker, 2021-02-21:
This is now fixed by the merge into develop.

closed 2021-02-21 by markmaker