openpnp/openpnp #1109: ReferenceTrayFeeder: Selecting a move to the current pick location ignores feed count

BaxEDM, 2021-01-20:
# Problem Report
## Describe the Issue
For a ReferenceTrayFeeder, selecting the "move the camera to the selected feeders pick location" button always moves the camera to the first position of the feeder, regardless of the feedcount. Other feeders take the feedcount into consideration

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1) Add a ReferenceTrayFeeder
2) Pick an initial pick location in the ReferenceTrayFeeder configuration, click apply
3) add some X offset in the ReferenceTrayFeeder configuration, e.g. X 10, click apply
4) Set tray count X =10, click apply
5) press the "move the camera to the selected feeders pick location" button, the camera moves to the pick location
6) Change the feedcount to 1, click apply
7) press the "move the camera to the selected feeders pick location" button, the camera moves to the pick location

## Expected Result
for step #7 of the steps to reproduce, I expected the camera to move to the picklocation + 1*X offset, since the feedcount was set to 1

## Actual Result
The feedcount is ignored and the camera always moves to the initial pick location

doppelgrau, 2021-04-09:
The pick-location is only updated after a feed-operation, so without a feed operation, the pick-position is not changed.

diorcety, 2021-05-30:
> 
> 
> The pick-location is only updated after a feed-operation, so without a feed operation, the pick-position is not changed.

This is not the case for ReferenceStripFeeder for exemple.

ozzysv, 2021-06-25:
> The pick-location is only updated after a feed-operation, so without a feed operation, the pick-position is not changed.

but the ReferenceStripFeeder for some reason simultaneously updates the position and this is convenient. And with ReferenceTrayFeeder  the question has already been raised several times, but the authors pretend that they don't care...)))

doppelgrau, 2021-06-25:
It's OpenSource.
Just create a good Pull-Request after thinking if in some corner-cases the change might cause problems.

markmaker, 2022-02-03:
As @doppelgrau correctly said, that's expected behavior. 
_Mark

closed 2022-02-03 by markmaker