openpnp/openpnp #1111: MOVE_TO_COMMAND Backslash compensations. What to do?

bacrua, 2021-01-26:
[image]

markmaker, 2021-01-26:
Hi @bacrua 

First make sure you have a backup of your machine.xml

If this is the newest [testing version](https://openpnp.org/test-downloads/), you should already have one, best go check it in 

`~/.openpnp2/backups`

If not, please make a manual backup now (or better yet, upgrade to the [testing version](https://openpnp.org/test-downloads/) and again check the auto backup).

Then you have two Options:

**1. Minimal changes, you don't want new features.**

Go to the driver and select all the `MOVE_TO_COMMAND`s (there may be several on the different HeadMountables, if you have multiple nozzles).

Delete the extra backlash move. Usually this means deleting the whole first line and changing the `BacklashFeedRate` to `FeedRate` on the second line.
```
G0 {BacklashOffsetX:X%.4f} {BacklashOffsetY:Y%.4f} {Z:Z%.4f} {Rotation:A%.4f} F{FeedRate:%.0f}
G1 {X:X%.4f} {Y:Y%.4f} {Z:Z%.4f} {Rotation:A%.4f} F{BacklashFeedRate:%.0f}
```
https://github.com/openpnp/openpnp/wiki/GcodeDriver#backlash-compensation

A pristine command is here:
https://github.com/openpnp/openpnp/wiki/GcodeDriver%3A-Command-Reference#move_to_command

**2. You want the new features.**

Just use the Issues & Solutions system to guide you through the optimization process. Note that after each round of accepting Solutions, you need to find new ones (best watch the video).

https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions

**Background**

Why the fuss? The new Backlash Compensation is explained here. The new **DirectionalCompensation** can really significantly speed up operations (but it must be calibrated):
https://github.com/openpnp/openpnp/wiki/Backlash-Compensation

_Mark

bacrua, 2021-01-26:
Thank you @markmaker . 
The latest version. I deleted the line, then there is no control (nothing is moved). So far I have not understood anything, but I will understand.

closed 2021-01-26 by bacrua

markmaker, 2021-01-27:
Don't be shy to ask more questions, but perhaps better in the [user group](https://groups.google.com/g/openpnp).
_Mark