openpnp/openpnp #1117: smoothie firmware can not be read

cx2323, 2021-02-01:
it will connect to the smoothie board but can not detect the firmware on the smoothie.
is there anyway to update firmware?
im on windows 7 with open pnp 2.

markmaker, 2021-02-02:
Hi cx,

See here:

https://makr.zone/smoothieware-new-firmware-for-pnp/500/ 
<https://makr.zone/smoothieware-new-firmware-for-pnp/500/>

And more generally here:

https://github.com/openpnp/openpnp/wiki/Motion-Controller-Firmwares 
<https://github.com/openpnp/openpnp/wiki/Motion-Controller-Firmwares>

And be sure to use this:

https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions 
<https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions>

_Mark

markmaker, 2021-02-02:
Btw. better use the user group for questions like these.

https://groups.google.com/g/openpnp

_Mark

closed 2021-02-16 by vonnieda