openpnp/openpnp #1120: Gcode Driver Timeout waiting for response to M114 get position

Simon24j, 2021-02-06:
Hello everybody,

have now set up my openpnp machine and start setting up openpnp.
I use a SKR V1.4 board with smothieware in the latest openpnp version.
I can drive all axes and switch the actuators via openpnp, but when I want to home the machine, all axes moves to the home position and after a while the following error "Gcode Driver Timeout waiting for response to M114 get position" occurs and the axes in openpnp do not zero themselves either the house button does not turn green.
what did I set wrong?

Thanks

markmaker, 2021-02-06:
What does Issues & Solutions say?

https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions

_Mark

Simon24j, 2021-02-06:
I think this shows the problem but i don´t know what i have to change!?!?
[image]

markmaker, 2021-02-07:
> image

How are our axes reported? You can see on the driver, the lower part:

_Mark

markmaker, 2021-02-07:
Stupid Github Mail gateway. Once again>>

How are our axes reported? You can see on the driver, the lower part:
[image]

_Mark

Simon24j, 2021-02-07:
[image]

Simon24j, 2021-02-07:
is the firmware (smoothieware) the problem? 
i can drive all axes X / X / Z and A & B for Nozzle rotation

markmaker, 2021-02-07:
No, I think the firmware is ok, though to be 100% sure I don't see all of it, the window was too narrow.

Could you send the config.txt and machine.xml?

_Mark

Simon24j, 2021-02-07:
Hi Mark,

here the files.
[config.txt](https://github.com/openpnp/openpnp/files/5939022/config.txt)

[machine.xml.txt](https://github.com/openpnp/openpnp/files/5939031/machine.xml.txt)
Thanks for your help
Simon

markmaker, 2021-02-07:
I see, you have the A B axes defined as extruders:

```
# Extruder module configuration
# See http://smoothieware.org/extruder
extruder.hotend.enable                          true          # Whether to activate the extruder module at all. All configuration is ignored if false
extruder.hotend.steps_per_mm                    8.8888      # Steps per mm for extruder stepper
```
Yes, Smoothie can drive them with the A and B letters, but they are still treated as if they were switched with the T command, and therefore only the E axis reported in M114. That is no longer compatible with Advanced Motion Control.

My machine is different, but you might use settings in my config.txt it as a guide. See towards the end of that article here:

 https://makr.zone/choosing-a-motion-controller-the-panucatt-azteeg-x5-gt-32bit/455/

Background:
https://smoothieware.org/6axis

_Mark

Simon24j, 2021-02-07:
Thanks Mark, now it works, i included the delta and epsilon axes instead of the Extruder, and now i can home it.
THANKS

closed 2021-02-13 by Simon24j

markmaker, 2021-02-25:
Added a hint in the Wiki:
https://github.com/openpnp/openpnp/wiki/Motion-Controller-Firmwares#smoothieware

[image]

Moonlightxy, 2021-08-29:
Hi, Mark!
I have deleted the extruder part according to the instructions of the wiki, but why am I still reporting an error?
This is my config.txt
[config.txt](https://github.com/openpnp/openpnp/files/7072118/config.txt)
And I use 5-axis
[image]

Thanks for your help
Wang

markmaker, 2021-08-29:
Hi @Moonlightxy,

You need to document what error is reported. Best include a `machine.xml` and a log at TRACE level.

_Mark

Moonlightxy, 2021-08-29:
Hi Mark!
This is my machine.xml
[machine.xml.txt](https://github.com/openpnp/openpnp/files/7072299/machine.xml.txt)

And this is my log
[OpenPnP.1.log](https://github.com/openpnp/openpnp/files/7072300/OpenPnP.1.log)

I am a beginner of Openpnp. 
Thank you very much for your help！

markmaker, 2021-08-29:
Hi @Moonlightxy,

It seems to be a communications problem. According to the log, OpenPnP can neither write nor read to/from the COM9 port.

Be sure to close any other running programs (Pronterface etc.) that have this port open. Windows does _not_ allow sharing of COM-ports. 

With unrelated questions such as these, please go to the discussion group in the future:
http://groups.google.com/group/openpnp

_Mark

Moonlightxy, 2021-08-29:
Hi Mark!
Sorry to trouble you again!
I just tried to change the port but openpnp still reports an error. Why is this?

And this is my log
[OpenPnP.log](https://github.com/openpnp/openpnp/files/7072448/OpenPnP.log)
config.txt
[config.txt](https://github.com/openpnp/openpnp/files/7072451/config.txt)
machine.xml
[machine.xml.txt](https://github.com/openpnp/openpnp/files/7072453/machine.xml.txt)

Thank you again!

markmaker, 2021-08-29:
Please ask in the discussion group. This does not belong in this Issue about M114.