openpnp/openpnp #1122: First Job can not be startet by "error can´t move to x ....."

Simon24j, 2021-02-13:
422 / 5000
Übersetzungsergebnisse
Hi,
I've now set up my machine so far and wanted to start the first job, unfortunately I get the error "can't move x to -27,000, lower than soft limit 0" after looking for some things I found the 27,000 at the nozzle two. Unfortunately I don't know where the problem is, I can place each nozzle on the pcb with the button "position the tool to the placement's location". 
[image]
[image]
I get the same error when I click on unload at "nozzel tips".
[image]

What i setup wrong?
Thanks
Simon

markmaker, 2021-02-13:
Set the log to TRACE level and look at the trace.

If is likely the machine wanting to load a nozzle tip to the second 
nozzle or something and your nozzle tip change location is not set up 
yet (i.e. the location is still at 0, 0, 0 resulting in the typical 
-27mm error).

Note, even if you do not have an automatic tool changer setup (yet) you 
need to define the manual change location.

The way to do that has changed in recent versions, so if you are using 
the latest testing version (which I hope ;-) it is now available on the 
nozzle (no need to define it on /each /nozzle tip):

Manual

https://github.com/openpnp/openpnp/pull/1107 
<https://github.com/openpnp/openpnp/pull/1107>

_Mark

markmaker, 2021-02-13:
@Simon24j,

Btw, it would be better to ask such support questions in the discussion group: 

http://groups.google.com/group/openpnp

The Github Issues here are more useful to track bugs or missing features, i.e. after they have been discussed and confirmed in the discussion group.

_Mark

Simon24j, 2021-02-14:
Thanks Mark it work, i posted a new problem at google group with the bottom vision....

closed 2021-02-14 by Simon24j