openpnp/openpnp #1132: Multiple LitePlacer errors after update

brainstorm, 2021-02-23:
# Problem Report
## Describe the Issue

Pre-COVID pandemic, our Liteplacer PnP machine was homing and jogging fine. Post-pandemic, after upgrading to a developer build (latest version as of today when filing this issue) and switching to the new GCodeAsync driver, it started to spit some errors and stopped jogging, only homing works now.

The "Detect Firmware" button detects no firmware and my LitePlacer is sitting ducks now :_(

But in the GCode/serial console, TinyG's `$$` command dumps all the settings from the board and I can communicate fine with it.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Home button.
4. The machine goes home but a POSITION_REPORT_REGEX shows up (see screenshots below).
4. The machine fails to jog and shows an M115 error (see screenshots below).

## Expected Result

I expected that the upgrade and the "Problem finder" was compatible with TinyG/Liteplacer. Could perfectly be a PEBKAC issue or a combination of both ;)

## Actual Result

[image]
[image]
[image]
[image]
[image]

**I'll use the Menu -> Help -> Submit Diagnostics feature and include the URL here.**

brainstorm, 2021-02-23:
> I'll use the Menu -> Help -> Submit Diagnostics feature and include the URL here.

... I ...  tried? :_/

[image]

brainstorm, 2021-02-23:
I zipped the crucial files from the machine here, hope that helps?:

[openpnp_error_report.zip](https://github.com/openpnp/openpnp/files/6028208/openpnp_error_report.zip)

markmaker, 2021-02-23:
Hi @brainstorm 

When you say "switching to the new GCodeAsync driver", did you use Issues & Solutions? 
https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions

If yes, did the diagnostic work? It should have said:
> Unknown firmware. 
> Check out firmwares known to be well supported. See info link.

And link to: https://github.com/openpnp/openpnp/wiki/Motion-Controller-Firmwares
> This page list Controllers and Firmwares that are known to be compatible with GcodeAsyncDriver and Advanced Motion Control features. ...

It this was not the case, please report so I can fix it.

As for the actual problem, you need to upgrade your firmware, if you want to use the GcodeAsyncDriver and/or Advanced Motion Control features. 
https://github.com/openpnp/openpnp/wiki/Motion-Controller-Firmwares#tinyg

If you want to go for the new firmware (and because TinyG is really super quirky), be sure to read about all the warnings, caveats and preparations here:
https://makr.zone/tinyg-new-g-code-commands-for-openpnp-use/577/

Or go back to the GcodeDriver (without the "Async"). For TinyG the testing feed-back was somewhat inconclusive. The original goal was to allow using the old firmware as long as you don't use any of the new features. That goal is still valid, but I don't know if it is fulfilled at the moment, as all the testers wanted to use the new features and switched anyway. If you don't want to switch to new firmware, please report findings back, so I can try and fix remaining bugs.

Btw. in the future, better ask in the user group first. Github issues should only be opened, once a bug or feature request was confirmed. 

_Mark

brainstorm, 2021-02-23:
Thanks a ton Mark for the speedy response!

Yes I used the Issues and Solutions dialog and IIRC it just hang for a while trying to detect the firmware, I don't recall the "See info link" dialog or message, but that was a few weeks ago so I might be mis-remembering.

In any case, I reckon that "Detect firmware" should always tell the user that the current firmware is not compatible/suitable instead of the message shown in the screenshot above in `(ˆFIRMWARE.*): No matching responses found`? Irrespective of the GCode(sync or async)?

I will update to the newest firmware after reading all that great documentation, diagrams and threads, for sure. I also think that allowing the old firmware is a pity given all the great improvements and effort you have put in the new one. So I'm fine with `avrdude`-ing a new firmware into that board, I would just have liked to have more clarity in the UI that this was needed for GCodeAsync to work properly?

Thanks again!

markmaker, 2021-02-23:
Usually, firmware detection is used by Issues & Solutions. Directly pressing the button in the driver was not intended as user guidance. However, Issues & Solutions should give you the aforementioned feedback. You can still try now! 

Detect Firmware is a bit of a chicken and egg problem. When the controller does not have a M115 command, then OpenPnP has no idea what controller it is. So it can't give you good advice. Hence I added it to TinyG. But of course, before the firmware update that's still chicken and egg ;-)

_Mark

brainstorm, 2021-03-16:
Thanks @markmaker ... I managed to flash the firmware successfully (and it gets detected correctly as .21) but now I'm facing some bizarre issues with the limit switches: the homing procedure that used the work before does not anymore :_(

Is this a known issue?

Here's the log after hitting the (yellow) home button: https://pastebin.com/zG6unqQg

More specifically:

1) I connect to the board (green power button) successfully.
2) The head moves to the "min" limit switch of the head, and stops, leaving that limit switch pressed but without error blinking light on the board. Before it retracted after activating that limit and carried on testing the other limit switches (usual homing procedure).
3) Any subsequent homing (black home button), results on the head banging against the limit switch (which is pressed from point 2).

Any ideas of what might be causing this behaviour? I checked your [TinyG code changes on the firmware](https://github.com/synthetos/TinyG/pull/258/files) and it does not seem to be any change of the [LitePlacer Z-inverting quirks kind of change](https://www.liteplacer.com/phpBB/viewtopic.php?f=13&t=314#p2163).

markmaker, 2021-03-16:
Hi @brainstorm 

Some users had issues with some settings in the EEPROM changing on them, after the upgrade, if the previous firmware was outdated. It seems TinyG's persistence model is not stable across the versions. I have documented that caveat in my post in the "Flashing the TinyG" section...

https://makr.zone/tinyg-new-g-code-commands-for-openpnp-use/577/

...so I hope you wrote those settings down and can now reconfigure them. If your former firmware several versions back, the endstop semantics might also have evolved ... I remember reading something about breaking changes. So you perhaps need to re-read the TinyG documentation and reconfigure according to the new semantics. 

Furthermore, some usage of homing commands (I think G92) must have changed behavior (or introduced bugs) in the master version I used as the basis of my firmware. Much if this is discussed in the threads in the user group:
https://groups.google.com/g/openpnp/c/veyVAwqS0do/m/F_6_YWIIAgAJ

In the group there are some TinyG gurus like @tonyluken that may be able to better help and/or provide example `machine.xml` and `$$` settings.

Hang in there: AFAIK every user upgrading to the new firmware/OpenPnP version has managed to get it going after some work. 

_Mark

brainstorm, 2021-03-18:
Thanks for following up so fast, @markmaker! Nope, not giving up and yes, I did backup the config, so I could diff it and easily spot the issue:

```diff
$ diff -u tinyG_config_pre_makr.zone_update.txt tinyG_config_post_makr.zone_update.txt
--- tinyG_config_pre_makr.zone_update.txt	2021-03-18 19:34:42.000000000 +1100
+++ tinyG_config_post_makr.zone_update.txt	2021-03-18 19:38:50.000000000 +1100
@@ -1,14 +1,8 @@
-[fv]  firmware version            0.97
-
 > $FB
-[fb]  firmware build            440.20
+[fb]  firmware build            440.21
 tinyg [mm] ok>
-
-
-
-
 > $$
-[fb]  firmware build            440.20
+[fb]  firmware build            440.21
 [fv]  firmware version            0.97
 [hp]  hardware platform           1.00
 [hv]  hardware version            8.00
@@ -16,7 +10,7 @@
 [ja]  junction acceleration  100000 mm
 [ct]  chordal tolerance           0.0100 mm
 [sl]  soft limit enable           0
-[st]  switch type                 1 [0=NO,1=NC]
+[st]  switch type                 0 [0=NO,1=NC]
 [mt]  motor idle timeout        300.00 Sec
 [ej]  enable json mode            0 [0=text,1=JSON]
 [jv]  json verbosity              2 [0=silent,1=footer,2=messages,3=configs,4=linenum,5=verbose]
@@ -110,7 +104,7 @@
 [ajh] a jerk homing           11520 deg/min^3 * 1 million
 [ajd] a junction deviation        0.0500 deg (larger is faster)
 [ara] a radius value              0.1989 deg
-[asn] a switch min                0 [0=off,1=homing,2=limit,3=limit+homing]
+[asn] a switch min                1 [0=off,1=homing,2=limit,3=limit+homing]
 [asx] a switch max                0 [0=off,1=homing,2=limit,3=limit+homing]
 [asv] a search velocity         600 deg/min
 [alv] a latch velocity          100 deg/min
```

So I'll now apply those particular changes back ;)

brainstorm, 2021-03-18:
Those two settings were the issue!

closed 2021-03-18 by brainstorm

markmaker, 2021-03-18:
@brainstorm, very nice preparation and follow-up. I guess this **diff method** could also help others. I made a link from my blog. Thanks!

I'm a bit astonished that you had 440.20 before (and not something older) and still you got flipped EEPROM settings?

Cannot see how this can happen, I did not touch the EEPROM stuff (memory layout) _at all_!?  

_Mark

fxframes, 2021-03-30:
@markmaker,

Thanks for this.
I'd like to just contribute to this topic reporting that my TinyG was also at the latest fw version before flashing and it also lost many of its settings, so what happened to @brainstorm wasn't an isolated incident.

[fb]  firmware build            440.20
tinyg [mm] ok>
[fv]  firmware version            0.97
tinyg [mm] ok>

After flashing the LitePlacer started moving in unexpected directions (X was inverted) and ways (jogging Y would also move Z), so I flashed the 440.20 back onto it.

I'm still very interested in this though and want to give it another try.

One question that's not clear to me, though. Will the "regular" OpenPnP build work at all with your firmware? From your description is seems that you recommend installing the 'test' build but it doesn't seem mandatory. I have the latest OpenPnP version on my Mac and none of the "new" items specifically related to the TinyG in "Issues & Solutions" came up. Not sure if I missed a step there?

I still tried following some suggestions that were already there in the regular build and the machine seemed ok with enabling the Advanced Motion Planner but didn't like the Async driver much, it started complaining about the missing M114 command.

Thanks for any insights.

markmaker, 2021-03-30:
Hi @fxframes 

Thanks for the feedback. This is very unexpected.

When you flashed back  440.20, did the settings miraculously "recover"? 

If yes, this probably means that there were multiple "versions 440.20" with changing EEPROM layout. Obviously, the settings persistence design is very simplistic (non-associative binary layout) due to the constraints of the tiny 8bit MCU. This is very hard to _robustly_ migrate to new versions. 

You need to save all your settings (obtained through `$$`) and then `diff` them after the flash update, as @brainstorm did. Then re-enter the settings that changed. 

> One question that's not clear to me, though. Will the "regular" OpenPnP build work at all with your firmware? 

Yes, definitely. 

Be sure to upgrate to the latest regular version and use [Issues & Solutions](https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions). 

> From your description is seems that you recommend installing the 'test' build but it doesn't seem mandatory.

Where did you read this? I thought I had removed all these outdated recommendations. The test version was long ago merged into the regular version i.e. the features are now all included in [regular OpenPnP 2.0](https://openpnp.org/downloads/).  In fact, the test version is currently lagging behind the regular version. :-)

[image]
https://github.com/openpnp/openpnp/tree/test

_Mark

fxframes, 2021-03-30:
Hey @markmaker,

> When you flashed back 440.20, did the settings miraculously "recover"?

I was able to flash 440.20 back with your script only, Syntheto's didn't work, just a side note.
After that the settings were _still_ wrong, but I went into the LitePlacer software and there's a button there to reset all the controller settings and that worked, bringing the TinyG back to "normal".

> You need to save all your settings (obtained through `$$`) and then `diff` them after the flash update, as @brainstorm did. Then re-enter the settings that changed.

I'll do as @brainstorm did next time.

> Be sure to upgrate to the latest regular version and use [Issues & Solutions](https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions).

This is weird because I definitely have the latest regular version. Could the change in settings have prevented the I&S bit to "find" the TinyG? I'll give it another go anyway.

> Where did you read this? I thought I had removed all these outdated recommendations. The test version was long ago merged into the regular version i.e. the features are now all included in [regular [...]

[Here](https://makr.zone/tinyg-new-g-code-commands-for-openpnp-use/577/).
You say "Download the newest OpenPnP 2.0 version or even better the Testing version to use this" which sounds like the test version would be a better option.

Thanks!

markmaker, 2021-03-30:
> After that the settings were still wrong, but I went into the LitePlacer software and there's a button there to reset all the controller settings and that worked, bringing the TinyG back to "normal"

Ok, that should also work with my firmware.

But you do know that the LitePlacer Software machine setup is different from what it should be for OpenPnP, right? 

Most notably the Z axis is inverted, it points **down** for Liteplacer Software, but it must point **up** for OpenPnP which is the "correct" orientation according to Cartesian geometry. 

https://en.wikipedia.org/wiki/Cartesian_coordinate_system#Three_dimensions

See also here:
[image]
https://github.com/openpnp/openpnp/wiki/Machine-Axes#the-openpnp-coordinate-system

This also reverses the meaning of the end-switches i.e. the Z probing switch. 

The issue is solved here:
https://www.liteplacer.com/phpBB/viewtopic.php?f=10&t=308#p2126

@tonyluken (Liteplacer+TinyG guru) may help you with more specifics, AFAIK he has a switch on the end-stop wires that lets him use both OpenPnP and Liteplacer Software. 

_Mark

fxframes, 2021-03-30:
Yes @markmaker, I'm aware, thanks. I switch the cabling at the controller when needed.
I didn't hardwire a switch (yet) because my intention was to settle on OpenPnP, we'll see how that goes.

fxframes, 2021-04-13:
Hello @markmaker,

I gave this another try today and it's looking good.

For the record, there were quite a number of settings that were different between 440.20 and 440.21 after flashing, but I fixed them as you suggested and it seems to be working well at least for the basic functionality.

I have now the following issue. I'd appreciate any help.

When homing I get _GcodeDriver configuration error: missing POSITION_REPORT_REGEX_. I tried to look it up but didn't find much. I found someone who used the same M114 command as in GET_POSITION_COMMAND which suppresses the error but causes a timeout. The machine homes, but the cameras freeze, I can't click on the images to move the nozzle anymore, it's only when I try clicking somewhere else that I get a timeout error.

Thanks,

fxframes, 2021-04-13:
Here's my driver section of machine.xml. Maybe there's something else in there.

`<driver class="org.openpnp.machine.reference.driver.GcodeAsyncDriver" id="DRV166f4908f8ceddc8" name="GcodeDriver" motion-control-type="SimpleSCurve" communications="serial" connection-keep-alive="false" units="Millimeters" max-feed-rate="0" backlash-offset-x="-0.3" backlash-offset-y="-0.4" backlash-offset-z="0.0" backlash-offset-r="0.0" non-squareness-factor="0.0" backlash-feed-rate-factor="0.1" timeout-milliseconds="15000" connect-wait-time-milliseconds="1000" visual-homing-enabled="true" backslash-escaped-characters-enabled="false" remove-comments="true" compress-gcode="true" logging-gcode="false" supporting-pre-move="false" using-letter-variables="false" infinity-timeout-milliseconds="60000" writer-polling-interval="100" writer-queue-timeout="60000" max-commands-queued="1000" confirmation-flow-control="true" reported-location-confirmation="true" interpolation-max-steps="32" interpolation-jerk-steps="4" interpolation-time-step="0.001" interpolation-min-step="16">
            <serial line-ending-type="LF" port-name="tty.usbserial-D308QQ68" baud="115200" flow-control="XonXoff" data-bits="Eight" stop-bits="One" parity="None" set-dtr="false" set-rts="false" name="SerialPortCommunications"/>
            <tcp line-ending-type="LF" ip-address="127.0.0.1" port="23" name="TcpCommunications"/>
            <simulated line-ending-type="LF"/>
            <homing-fiducial-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
            <detected-firmware><![CDATA[FIRMWARE_NAME:TinyG, FIRMWARE_URL:https%3A//github.com/synthetos/TinyG, FIRMWARE_VERSION:0.97, FIRMWARE_BUILD:440.21, HARDWARE_PLATFORM:1.00, HARDWARE_VERSION:8.00]]></detected-firmware>
            <reported-axes><![CDATA[X:15.9730 Y:2.8585 Z:0.0000 A:0.0000 B:0.0000 C:0.0000]]></reported-axes>
            <command type="DISABLE_COMMAND">
               <text><![CDATA[M9 M5]]></text>
            </command>
            <command type="ENABLE_COMMAND">
               <text><![CDATA[G21 M9]]></text>
            </command>
            <command type="POST_VISION_HOME_COMMAND">
               <text><![CDATA[G92 {X:X%.4f} {Y:Y%.4f}]]></text>
            </command>
            <command type="CONNECT_COMMAND">
               <text><![CDATA[$SV=2]]></text>
               <text><![CDATA[$ME]]></text>
               <text><![CDATA[$ex=1]]></text>
               <text><![CDATA[$1pm=2]]></text>
               <text><![CDATA[$2pm=2]]></text>
               <text><![CDATA[$3pm=2]]></text>
               <text><![CDATA[$4pm=2]]></text>
               <text><![CDATA[$mt=300]]></text>
               <text><![CDATA[G21 G90 G92 X0 Y0 Z0 A0 M8 M5]]></text>
               <text><![CDATA[$1sa=0.900]]></text>
               <text><![CDATA[$1tr=39.9190]]></text>
               <text><![CDATA[$1mi=8]]></text>
               <text><![CDATA[$2sa=0.900]]></text>
               <text><![CDATA[$2tr=39.9060]]></text>
               <text><![CDATA[$2mi=8]]></text>
               <text><![CDATA[$3sa=1.800]]></text>
               <text><![CDATA[$3tr=8.0000]]></text>
               <text><![CDATA[$3mi=8]]></text>
               <text><![CDATA[$3po=1]]></text>
               <text><![CDATA[$4sa=0.900]]></text>
               <text><![CDATA[$4tr=160.0000]]></text>
               <text><![CDATA[$4mi=8]]></text>
               <text><![CDATA[$xjm=500]]></text>
               <text><![CDATA[$xjh=2000]]></text>
               <text><![CDATA[$xvm=5000]]></text>
               <text><![CDATA[$xsv=2000]]></text>
               <text><![CDATA[$xfr=5000]]></text>
               <text><![CDATA[$xsn=3]]></text>
               <text><![CDATA[$xsx=2]]></text>
               <text><![CDATA[$yjm=500]]></text>
               <text><![CDATA[$yjh=2000]]></text>
               <text><![CDATA[$yvm=5000]]></text>
               <text><![CDATA[$ysv=2000]]></text>
               <text><![CDATA[$yfr=5000]]></text>
               <text><![CDATA[$ysn=3]]></text>
               <text><![CDATA[$ysx=2]]></text>
               <text><![CDATA[$zjm=500]]></text>
               <text><![CDATA[$zjh=500]]></text>
               <text><![CDATA[$zvm=5000]]></text>
               <text><![CDATA[$zsv=1000]]></text>
               <text><![CDATA[$zfr=2000]]></text>
               <text><![CDATA[$zsx=3]]></text>
               <text><![CDATA[$zsn=2]]></text>
               <text><![CDATA[$zzb=2]]></text>
               <text><![CDATA[$ajm=5000]]></text>
               <text><![CDATA[$avm=50000]]></text>
               <text><![CDATA[$afr=200000]]></text>
            </command>
            <command type="HOME_COMMAND">
               <text><![CDATA[G28.2 X0 Y0 Z0 A0 ]]></text>
               <text><![CDATA[G28.3 X0 Y0 Z0]]></text>
[... 14 lines cut ...]
               <text><![CDATA[M4]]></text>
            </command>
            <command type="PUMP_OFF_COMMAND">
               <text><![CDATA[M5]]></text>
            </command>
            <command type="MOVE_TO_COMMAND">
               <text><![CDATA[G0 {X:X%.4f} {Y:Y%.4f} {Z:Z%.4f} {Rotation:A%.4f} F{FeedRate:%.0f}]]></text>
            </command>
            <command head-mountable-id="ACT1582497401237" type="ACTUATE_BOOLEAN_COMMAND">
               <text><![CDATA[{True:M4}{False:M5}]]></text>
            </command>
            <command head-mountable-id="ACT1582497414326" type="ACTUATE_BOOLEAN_COMMAND">
               <text><![CDATA[{True:M8}{False:M9}]]></text>
            </command>
            <command type="GET_POSITION_COMMAND">
               <text><![CDATA[M114]]></text>
            </command>
            <command type="MOVE_TO_COMPLETE_COMMAND">
               <text><![CDATA[M400]]></text>
            </command>
            <command type="SET_GLOBAL_OFFSETS_COMMAND">
               <text><![CDATA[G28.3 {X:X%.4f} {Y:Y%.4f} {Z:Z%.4f} {A:A%.4f}]]></text>
            </command>
            <junction-deviation value="0.02" units="Millimeters"/>
         </driver>`

markmaker, 2021-04-13:
Use Issues & Solutions to help you. best watch the video linked here:

https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions 
<https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions>

_Mark

fxframes, 2021-04-13:
Thanks @markmaker I've tried that but unless it's got something to do with the axis letter it didn't pick up a problem.

[image]

[image]

fxframes, 2021-04-13:
Thanks I didn't realize at first that it was something to do with the axis configuration, but combining the I&S report with the log file it was able to sort it out. Thanks again!

markmaker, 2021-04-14:
> Screenshot 2021-04-13 at 18 33 46
> Screenshot 2021-04-13 at 18 35 54

Just so you and other know: Whenever you see a*"Fundamental" *suggestion 
in Issues & Solutions, it means that a whole bunch of other Issues & 
Solutions are pending "behind" that one. These need to be resolved 
first, so the system can build on top.

In the case of the *Axis Letters*: these are mandatory for practically 
all the Advanced Motion Control features and all the automatic G-code 
generation.

 From the Wiki:

/*Axis Letter*//gives you the letter of the axis by which it is 
addressed in the controller (usually a G-code letter). The letter must 
obviously be unique for a given driver. If you have multiple drivers, 
you typically have the same letter multiple times (and that is why 
inside OpenPnP you refer to the axis by its conceptual //*Name*//rather 
than its //*Letter*//)./

https://github.com/openpnp/openpnp/wiki/Machine-Axes#controller-settings 
<https://github.com/openpnp/openpnp/wiki/Machine-Axes#controller-settings>

As you can imaging, the system can only build G-code for you, if it 
knows the letters. You're missing out a lot by not using them.

_Mark

fxframes, 2021-04-15:
@markmaker,

Thanks that was exactly the case, I didn't realize there was a lot more behind it. That one is sorted now!

One other thing I haven't quite understood yet is how to set the values below for the TinyG specifically.
I read [here](https://github.com/openpnp/openpnp/wiki/Machine-Axes#kinematic-settings--rate-limits) the definitions and about the unit conversion, that's all ok. But I also read [here](https://groups.google.com/g/openpnp/c/veyVAwqS0do/m/YvSJXovwBQAJ) that some of these settings might not be relevant to the TinyG, for example the acceleration limit. Specially for the jerk setting is there any range that would be recommended? I have it at 0 now which effectively turns it off, correct?

[image]

I ran some tests and was surprised to see that the Y motor performance seems to be limited to approx. half of the X performance even if I set it at the same values. Since it's driving two belts I suppose that would make sense? I'm seriously thinking of following your steps and upgrade the Y motor to a NEMA23.

[image]

Thanks.

fxframes, 2021-04-15:
Sorry I take back the ½ performance on the Y axis, for the test I had the distance traveled by the Y axis about half that of the X axis. Making them the same allowed them also to have similar performance.

[image]

markmaker, 2021-04-15:
Yes the acceleration setting is ineffective on a TinyG, it has only a jerk limit.

You need to set a Jerk limit for speed control. I would start with 100000 mm/s³ and then go down, if that is too drastic. 

Also make sure to set the SimpleSCurve control type:

https://github.com/openpnp/openpnp/wiki/GcodeAsyncDriver#gcodedriver-new-settings

But that should be suggested by Issues & Solutions.

_Mark