openpnp/openpnp #1139: GCode Regex

Protoneer, 2021-03-09:
# Problem Report
## Describe the Issue
Installed latest version on Windows last night. When configuring the gcode driver regex was not populated. I picked the Async GCode driver and later the older Gcode driver.

I have an old version and I am sure that it was populated for you in the past.
[image]

Is this intentional? Was not able to get my controller to talk with openPNP without these settings.

vonnieda, 2021-03-09:
This is intentional. Once you've selected a Driver you can go to Machine Setup -> Reference Machine -> Issues and Solutions and you will have items ready to fill in those Gcodes and Regex. This change is so that we can support many different firmwares and generate the right codes.

Protoneer, 2021-03-10:
Thanks, I had a look at the "Issues and Solutions" section. 

I think the first problem is that it does not recognize my GRBL board to start with. I am looking at using GRBL_ESP32 that offers me 6 axis and plenty of IO to control actuators.

Protoneer, 2021-03-10:
@markmaker , thanks for adding more details.

I am in discussions with the team working on GRBL_ESP32 and I find them very open to suggestions and change.

It would be nice to make their board compatible , but I sort of have to make sure GRBL will work minus the the extra few M codes needed for auto setup.

Protoneer, 2021-03-11:
Like I said the GRBL_ESP32 team is onto it...

https://github.com/bdring/Grbl_Esp32/tree/UserDefinedGcode

From the tests this is what it returns:

> Grbl 1.3a ['$' for help]
> m115
> FIRMWARE_NAME:Grbl_ESP32, FIRMWARE_VERSION:1.3a, FIRMWARE_BUILD:20210308
> ok

Will it be okay with the 'ok' at the end?

markmaker, 2021-03-11:
That was quick! Yes that's good.

Protoneer, 2021-03-11:
@markmaker 

[image]

Protoneer, 2021-03-11:
I am looking at M114 next... is that the machine poistion or the offset position that G92 set?

Protoneer, 2021-03-11:
Wow... I manged to get the M114 command working as well.

> Grbl 1.3a ['$' for help]
> g0x100
> ok
> m114
> X:100.000000 Y:0.000000
> ok

Where do I go from here? This is getting of topic from the main Regex topic.

Protoneer, 2021-03-11:
Noting that , GCodeserver thing but I think I misunderstood.

After the Issues& resolution has been run do I still need to manually enter the Regex? 

My confusion was that I was expecting regex example to be populated after doing I&R

markmaker, 2021-03-12:
> I am looking at M114 next... is that the machine poistion or the offset position that G92 set?

I guess you found out. 
https://duet3d.dozuki.com/Wiki/Gcode#Section_M114_Get_Current_Position

For future questions, I recommend looking at these references:

- https://duet3d.dozuki.com/Wiki/Gcode (best documentation, IMHO)
- https://www.reprap.org/wiki/G-code (has a nice overview of controller compatibility)
- https://smoothieware.org/supported-g-codes
- http://linuxcnc.org/docs/html/gcode/g-code.html and http://linuxcnc.org/docs/2.8/html/gcode/m-code.html

I know, these are sometimes very terse, you need to read multiple to get it. Or play around with the GcodeServer for expected behavior. 

> Noting that , tried the GCodeserver thing, but I think I misunderstood.

Does it work at all? Can you connect and move around? Note, you must enter exactly "GcodeServer" (without the quotes) i.e. it is case-sensitive.

> doing I&R

Just to exclude any misunderstanding: I&S only proposes Solutions. You need to **accept** them, before they become effective.

Best watch the video linked here:
https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions

_Mark

markmaker, 2021-04-25:
@Protoneer, any progress here?

whoim2, 2021-07-29:
> @Protoneer , есть ли здесь прогресс?

Hey! I also decided to try esp32 grbl with openpnp. Soon I will be ready to test and help with this.
At the moment, the movement of the head across the field is working, including the movement of the mouse in the camera window, the movement of two nozzles with a parking in the middle is working. I am still awaiting some details, optical limit switches, cameos, etc. They will be arriving shortly.
If I can help you at this stage, let me know.

Protoneer, 2021-07-30:
> @Protoneer, any progress here?

Still waiting on supplies... I have a custom board going already. Thinking of ways to control a bunch of feeders.

whoim2, 2021-08-04:
> > @Protoneer, any progress here?
> 
> Still waiting on supplies... I have a custom board going already. Thinking of ways to control a bunch of feeders.

esp32 grbl is limited to 4 digital outputs and 4 analog outputs, there is also a spindle and two flood outputs. For now, I decided to use a solenoid to move the tapes, in the future I will probably try autonomous options using photocells. When the head is placed over the feeder, a photocell is triggered and the part is fed.
I will try to do without microcontrollers.
I used a 3-axis board that I had previously designed for a milling machine and built an expansion board with two 4988 drivers and 4 step-down modules with a TTL input. More precisely, with five modules, but two work through an inverter on a transistor. When the lower chamber is turned on, its illumination is turned on, the upper illumination is automatically turned off, and this requires only one control output.
Here's a short video.
https://youtu.be/kW1HNVAuCJw