openpnp/openpnp #1163: [FEATURE REQUEST] License for the documentation

darigovresearch, 2021-04-02:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
Have a license for the documentation such as the quick start guide, manual etc for example CC0, CC-BY etc

## How Will It Look?
Section in the documentation noting the license of the documentation or it can be autogenerated at the bottom of every page

## Why Do We Need It?
1. Having a license specifically for the documentation encourages people to contribute to the documentation which results in better docs and increased longevity of the project as a whole
2. Would allow for other people to build on the documentation such as teachers creating educational material for a course 

## Provide Examples
Example of good similar open source project which does this well
- [Farmbot](https://farm.bot/pages/open-source) an open source machine for seeding, watering and monitoring farming machine

vonnieda, 2021-04-18:
My inclination would be to use https://creativecommons.org/licenses/by-sa/4.0/. This allows modification with attribution, and commercial use.

@markmaker As the person with the greatest number of contributions to the documentation aside from me, how would you feel about this?

markmaker, 2021-04-19:
Oh, that sounds fine to me as far it concerns my stuff, including images/gifs.

I posted a few images/gif animations from source material created by others, Mike Menci(?), [Peter Betz](https://github.com/openpnp/openpnp/wiki/Transformed-Axes#negate-an-axis). I always asked them, but I have no idea if that covers formally licensing it now. There is at least one [Wikimedia image that I recall](https://github.com/openpnp/openpnp/wiki/Backlash-Compensation) and this seems to be the same license (older version number). There might be other material, so I'm not sure if there should be a clause like "unless otherwise attributed" or similar. I'm no expert in these matters. 

_Mark

darigovresearch, 2021-04-24:
May be worth getting the main contributors in this issue to discuss if they are happy to do the same. 

Was considering opening another pull request to discuss porting the documentation to another format which can be version controlled and easier to track contributions, issues, pull requests etc such as [Docusaurus](https://docusaurus.io/), [Docsy](https://www.docsy.dev/) or [read the docs](https://readthedocs.org/) depending on your coding preference.

The only reason mentioning it here is that GitHub Wiki can be a bit limiting after a while in terms of translation, tracking etc and this may be a good time to port all existing docs under whatever license you chose (for those who agree). Then you can deprecate the current one and any new contributions to the docs can have a confirmation that their contributions must be under the new license for consistency. Also no expert in these matters.

@markmaker & @vonnieda what do you think?