openpnp/openpnp #1168: Feature #1153 breaks #1068

JulienDevillers, 2021-04-09:
I did a post for it at the end of the merge request of #1153 but I don't know if somebody did see it.

Basically "Changed table row visuals to alternate colors" was applied without distinction and overrides
the behaviour of "a feeder in feeder's list is grayed out if no active placement uses it".

doppelgrau, 2021-04-09:
The checkboxes should show the differences used/not used if I understood it correctly, example see
https://discord.com/channels/693329785769689098/694915062388883567

JulienDevillers, 2021-04-12:
Sorry, link does not work.

JulienDevillers, 2021-04-12:
Hum, ok. I see. 
Depending on the appearance used, it's not very easy to determine whether the line is enabled or not.

closed 2021-04-12 by JulienDevillers