openpnp/openpnp #1175: Machine "Freezes" if Job-movement outside soft-limits

RRudi92, 2021-04-14:
# Machine "Freezes" on Job-movement outside soft-limits
## Describe the Issue
If the Z-Position of a board is set outside of the softlimits (e.g. positve instead of negative, or below "the table" (-300mm)), a placing process ist running up to the point the nozzle should lower to place the part (X & Y at Part destination). Than the machine stops without an usefull error, if only sending a "timeout on M114".

## Steps to Reproduce
Setting Board Z Position outside of Softlimits
Placing any Part

## Expected Result
Best: 
Check all Placement positions on Job start if positions are in allowed movement area of the machine (inside Softlimits). If not -> warning massage.

At least: 
Throw Error message like "Can't move to -300 , lower than soft limit -30.5mm". The same like the one on Jog-Movements

## Actual Result
Error "Connection timout" is thrown.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue

FIRMWARE_NAME:TinyG, FIRMWARE_URL:https%3A//github.com/synthetos/TinyG, FIRMWARE_VERSION:0.97, FIRMWARE_BUILD:440.21, HARDWARE_PLATFORM:1.00, HARDWARE_VERSION:8.00

OpenPnP Ver: 2021-04-08_04-19-29.d3a4c39

doppelgrau, 2021-04-18:
Just to clarify: Are you talking about (enabled) soft limit on the axis (Machine Setup -> ReferenceMachine -> Axes), or about the soft limit configured inside tinyG?
The later I'd call expected behavior, since OpenPNP has no method to know how the controller is configured.

RRudi92, 2021-04-19:
Actually I'm talking about the axis soft limits. But I have to check if both are consistent (the ones inside OpenPNP should be within the ones configured in tinyG).

RRudi92, 2021-04-21:
OK, if the soft limits inside OpenPNP Software are set up correct, everything is fine:
[image]

closed 2021-04-21 by RRudi92