openpnp/openpnp #1176: Import Board on KiCAD .pos  does not assign every part

RRudi92, 2021-04-14:
# Problem Report
## Describe the Issue
I have encountered several problems with importing placement files. Using .pos (KiCAD), some parts are not assigned, using Eagles .mnt files, none part is assigned.

**Using Eagle .mnt files:**
None of the parts are assigned. Even if the part-ID are exact the same as already existing parts.
=> See issue <<<<will be created in a few minutes>>>>>

**Using KiCad .pos files:**
Some of the parts are not assigned.

## Steps to Reproduce
Import file with contents (both parts exist in parts library/tab):
```
# Ref Val Package PosX PosY Rot Side
C7 1µ rcl-C0805 -22.25 -14.25 270.00 top
C8 10n rcl-C0603 -3.75 -17.00 90.00 top
```

## Expected Result
Correct assignment of both parts

## Actual Result
only C8, the 100n-part is assigned. The assignment of 1µ is missing.
[image]
[image]

If using "Use only Value as Part ID", no part is assigned.

If using "Create missing parts", a new part is created for the 1µ:
```
<part id="rcl-C0805-1Âµ" height-units="Millimeters" height="0.0" package-id="rcl-C0805" speed="1.0" pick-retry-count="0"/>
```

Image:
First two parts: Assignment
Parts 3&4: Assignemt using only value
Parts 5&6: Assignem using "Create missing parts"
Last two parts: Assignment after new parts were created.
[image]

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
The Char 'µ' seems to be wrongly assigned to two Chars.

doppelgrau, 2021-04-18:
I'm quite sure at least for the kicad importer: It's an encoding problem.
Just to be sure: only non-ASCII-characters are affected in your setup?

Additional information that might help: Which encoding is default on your system and in which encoding the .pos is saved?

RRudi92, 2021-04-18:
The file is in UTF-8.
As far as I'm aware only the µ is affected. I will chek on wednsday if sth. else could be affected. 

So for now simply don't use µ, but instead u.

doppelgrau, 2021-04-18:
Can you try this fix?
https://github.com/doppelgrau/openpnp/tree/fix/kicadImporterCharset
Attached a compiled jar-file with that fix for easier testing. (rename to .jar, needed to change to .zip to make github happy)
[openpnp-gui-0.0.1-alpha-SNAPSHOT.zip](https://github.com/openpnp/openpnp/files/6332378/openpnp-gui-0.0.1-alpha-SNAPSHOT.zip)

RRudi92, 2021-04-21:
@doppelgrau 
Yes, now the import works (at least for the "µ")
A second Issue is that if a part does not exist, the field stays empty. 

e.g.: 
`C12 9p rcl-C0603 -15.00 9.00 180.00 top`
will result in:
[image]

It would be nice if the name would be imported, and "missing Part" beeing assigned. I might open a new Issue / suggestion.

doppelgrau, 2021-04-21:
Just to clarify, the "create missing parts" option was checked, but did not work?

RRudi92, 2021-04-21:
No, without checking the "create missing parts".

doppelgrau, 2021-04-21:
Missing parts are only created, if that option is checked. To avoid getting dozens of unwanted parts, if someone used a different naming schema.

RRudi92, 2021-04-21:
The background is: Sometimes there are alternative parts on the machine. Like with the 9pF, I place a 10pF. But if the Field stays empty, I can't see what part "should have been used".

vonnieda, 2021-04-21:
@RRudi92 Yes, this is a major limitation in OpenPnP right now, and I have plans to fix it. Right now a Part dropdown can only contain a valid part reference or null. I will be changing this so it can contain an "invalid" part reference so you can see what was contained in the import.