openpnp/openpnp #1184: The latest version resets the Z axis position

bacrua, 2021-04-26:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
Please describe the issue in as much detail as possible. Without a good description it's impossible for developers to understand the problem.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Jog the machine 10mm to the right.
4. The machine fails to jog and shows an error.

## Expected Result
What did you expect to happen? Be descriptive.

## Actual Result
What actually happened? Be descriptive.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
The more information you can provide about the issue the more likely it is that a developer will be able to respond.

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
What do you want it to do? Be as descriptive as possible.

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.

## Why Do We Need It?
1. What problem does it solve?
2. Is it useful for everyone, or does it only solve a problem for your machine?

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

doppelgrau, 2021-04-28:
Not sure if that's intended behavior.
If not a fix would be quite simple, in the Wizzard lines 600ff just keep the old height instead setting it to zero.

`                                 final Location referenceHole1 = referenceHoles.get(0)
                                                                               .derive(null, null,
                                                                                       null, 0d);
                                 final Location referenceHole2 = referenceHoles.get(1)
                                                                               .derive(null, null,
                                                                                       null, 0d);
`

markmaker, 2021-04-29:
Yep.

```
diff --git a/src/main/java/org/openpnp/machine/reference/feeder/wizards/ReferenceStripFeederConfigurationWizard.java b/src/main/java/org/openpnp/machine/reference/feeder/wizards/ReferenceStripFeederConfigurationWizard.java
index df574b7e84..5ea7996df0 100644
--- a/src/main/java/org/openpnp/machine/reference/feeder/wizards/ReferenceStripFeederConfigurationWizard.java
+++ b/src/main/java/org/openpnp/machine/reference/feeder/wizards/ReferenceStripFeederConfigurationWizard.java
@@ -599,11 +599,15 @@ public class ReferenceStripFeederConfigurationWizard extends AbstractConfigurati
                                          firstPartLocation, secondPartLocation,
                                          part1HoleLocations, part2HoleLocations);
                                  final Location referenceHole1 = referenceHoles.get(0)
-                                                                               .derive(null, null,
-                                                                                       null, 0d);
+                                         .derive(null, null,
+                                                 null, 0d)
+                                         .derive(feeder.getReferenceHoleLocation(), 
+                                                 false,  false,  true, false);
                                  final Location referenceHole2 = referenceHoles.get(1)
-                                                                               .derive(null, null,
-                                                                                       null, 0d);
+                                         .derive(null, null,
+                                                 null, 0d)
+                                         .derive(feeder.getLastHoleLocation(), 
+                                                 false,  false,  true, false);
 
                                  feeder.setReferenceHoleLocation(referenceHole1);
                                  feeder.setLastHoleLocation(referenceHole2);

```

I traced this back to 2015 i.e. the very beginning. Are you sure, @bacrua this ever worked? 

_Mark

markmaker, 2021-04-29:
Found the real reason and fixed it. See #1186.

closed 2021-05-03 by markmaker