openpnp/openpnp #1199: Log Tab is replaced by Issues & Solutions

fxframes, 2021-05-20:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
When accessing the Issues & Solutions tab, the Log tab right next to it also changes name to "Issues & Solutions".
Please see image below.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Go to the Issues & Solutions tab.

## Expected Result
The Log tab should have retained its name.

## Actual Result
The Log tab's name has been replaced.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue

[image]

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
What do you want it to do? Be as descriptive as possible.

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.

## Why Do We Need It?
1. What problem does it solve?
2. Is it useful for everyone, or does it only solve a problem for your machine?

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

markmaker, 2021-05-21:
Hi @fxframes,

Thanks for reporting this. 

I hope I fixed it, see #1201. 

**Note:** As I cannot reproduce the issue, this may or may not work.

Can you please upgrade and confirm?

_Mark

fxframes, 2021-05-21:
Hi @markmaker,

Thank you this has been fixed.

fxframes, 2021-06-22:
This issue is back after the latest update.

markmaker, 2021-06-22:
Could you give me the version (from Help/About)?

markmaker, 2021-06-22:
I just checked the source and the fix is in. 

Just to be sure: you did not download the Testing version, right? 

The `test` branch is behind.

_Mark

fxframes, 2021-06-22:
I updated it from within the program, I got the (weekly, in my case) update notice.

[image]

fxframes, 2021-06-22:
I just noticed that you actually have to click on one of the I&S list items for the tab to "change".
If you just click on the I&S tab it's working normally.

[image]

markmaker, 2021-06-22:
I'm at a loss. I reread the Java spec and some stackoverflow solution etc. according to these it should be right. 

https://github.com/openpnp/openpnp/blob/4ed8f8a6976122f144e128df991358751f894072/src/main/java/org/openpnp/gui/IssuesAndSolutionsPanel.java#L310-L331

@vonnieda, any ideas?

_Mark

vonnieda, 2021-06-22:
@markmaker Sorry - no idea. If it were me I would add some logging of the index and ask @fxframes to report back. You could also, instead of using indexOfComponent, just loop through the tabs and find the one with the right title.

markmaker, 2021-06-25:
@fxframes, hopefully fixed and new version ready. Please try again.

markmaker, 2021-07-02:
@fxframes, can we close this again?

fxframes, 2021-07-04:
Hi @markmaker unfortunately the problem persists in the latest version.
Please see the video attached.

https://user-images.githubusercontent.com/68516637/124388694-de45d880-dcdb-11eb-9ea1-b29ccf2a882d.mp4

markmaker, 2021-07-04:
Thanks, @fxframes 

What could possibly be wrong here? As @vonnieda suggested, I'm going through each tab by _index_, ask whether the word "Issue" is in the title, then immediately set the new title using the same _index_. 

https://github.com/markmaker/openpnp/blob/bf831b38f842a916fbfb5bf8c13789900475219f/src/main/java/org/openpnp/gui/IssuesAndSolutionsPanel.java#L318-L336

I'm starting to think this is not my bug. Short of just removing the functionality, I don't know what else to do. 

_Mark

fxframes, 2021-07-04:
Honestly I don’t think this is that much of a problem.
I reported it but it doesn’t bug me nearly as much as having the pipeline windows open behind the current window. :)

ian-arkver, 2021-07-04:
If you comment out the two setTitleAt calls does the problem go away (along with the colouring)? As Jason said, it might be worth adding logging around these calls to show the index values, title of that index, etc. But the code (both old indexOfComponent and new version) looks correct to me. Very odd.

Is it just the log tab's title that's being set? i.e. does it still open the log tab contents if you click on it?

ian-arkver, 2021-07-04:
Works for me. :-)

I added these...
```
                    Logger.debug("Set colour title tab={}", index); //$NON-NLS-1$
                    tabs.setTitleAt(index,
                            "<html>Issues &amp; Solutions <span style=\"color:#"
                                    + String.format("%02x%02x%02x", color.getRed(), color.getGreen(), color.getBlue())
                                    + ";\">&#" + (indicatorUnicode) + ";</span></html>");
```
and
```
                    Logger.debug("Set plain title tab={}", index); //$NON-NLS-1$
                    tabs.setTitleAt(index, "Issues & Solutions");
```

and I see it in the log...
`2021-07-04 19:12:14.289 IssuesAndSolutionsPanel DEBUG: Set colour title tab=5`

and the correct tab gets decorated. Clicking around on the suggestions it continues to be correct.

I'm running openjdk 11.0.11 2021-04-20 on Manjaro Linux.

fxframes, 2021-07-08:
This is somehow theme-related.
As @markmaker pointed out here
https://github.com/openpnp/openpnp/issues/1222
When I change the theme to any other than the default - MacOSX theme, it works as expected.
As soon as I switch back to the default theme, though, the problem is back.

markmaker, 2021-09-13:
In the next testing version:

- Start and Exit OpenPnP (so the new `machine.xml` format is written).
- Open your `machine.xml`
- Search for `show-indicator`
- Set to `"false"`
- Save
- Restart OpenPnP

```xml
    ...
    <home-after-enabled>true</home-after-enabled>
      <solutions show-indicator="false" target-milestone="Calibration">
         <dismissed-solutions class="java.util.HashSet">
    ...
```

_Mark

markmaker, 2021-09-13:
Hi @fxframes,

please upgrade and report back, if this switch works for you.

_Mark

fxframes, 2021-09-14:
Hi Mark,

Yes that fixed it. 👍🏻

A side note: after installation of the test version, the installer seems to be “holding on” to something, by that I mean that I can’t eject it like in the main version. The only way to do this was to relaunch Finder and “Force Eject” it after that. OpenPnP seems to be working normally, unaffected by that.

Thanks!

markmaker, 2022-05-19:
According to discussion [here](https://github.com/openpnp/openpnp/issues/1415#issuecomment-1130536223), I'm reopening this.

markmaker, 2022-05-19:
@Shaische 

can you confirm that the [workaround here](https://github.com/openpnp/openpnp/issues/1199#issuecomment-918392848) works?

_Mark