openpnp/openpnp #1200: After the latest OpenPnP upgrade a timeout is generated every time a nozzle is changed (LitePlacer)

fxframes, 2021-05-20:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
After the latest OpenPnP upgrade, a timeout is generated every time a nozzle is changed (LitePlacer/TinyG).
It was working perfectly before.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Go to a part which is associated with a different nozzle than the currently loaded one
3. OpenPnP will change the nozzle and move it to the bottom camera
4. OpenPnP will stop and throw a timeout error.

## Expected Result
The nozzle should be calibrated as usual. No errors.

## Actual Result
OpenPnP stops and throws a timeout error.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue

[image]

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
What do you want it to do? Be as descriptive as possible.

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.

## Why Do We Need It?
1. What problem does it solve?
2. Is it useful for everyone, or does it only solve a problem for your machine?

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

markmaker, 2021-05-21:
Hi @fxframes,

This is a confirmation that my bug-fix has **helped**. In previous versions of OpenPnP the error was swallowed, and it could lead to follow-up problems.

See #1193 

It is **good** that you now see the error, you must fix the underlying problem, which I guess is a **command timeout** that is too short. 

https://github.com/openpnp/openpnp/wiki/GcodeAsyncDriver#gcodedriver-new-settings

Increasing the timeout is probably necessary because you have let Issues & Solutions configure continuous motion, which means that multiple move commands are queued and the executed as one sequence, which obviously is more efficient for the controller/machine, but takes longer as a whole command sequence, hence the need for a longer timeout. 

See here for background info:
https://github.com/openpnp/openpnp/wiki/Motion-Planner#motion-path-planning

_Mark

fxframes, 2021-05-21:
Hi @markmaker,

Thank you.
Increasing the timeout did the trick.

closed 2021-05-21 by fxframes