openpnp/openpnp #1215: Tool Changer "Change on Manual Pick" doesn't move to Manual Change Location

bilsef, 2021-06-14:
# Problem Report
## Describe the Issue
With a nozzle tool changer set to "Change On Manual Pick", it does not move to the Manual Change Location when a manual tool change is required.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Home machine
4. On the Feeders tab, pick a part that requires a tool change.

## Expected Result
Nozzle should move to manual change location.

## Actual Result
Move was not issued to controller.

2021-06-14 13:39:58.531 ReferenceNozzle DEBUG: Left.unloadNozzleTip(): Start
2021-06-14 13:39:58.532 ReferenceNozzle DEBUG: Left.unloadNozzleTip(503A): moveTo manual Location
2021-06-14 13:39:58.532 ReferenceHead DEBUG: H1.moveToSafeZ(1.0)
2021-06-14 13:39:58.532 AbstractHeadMountable DEBUG: Left.moveToSafeZ(1.0)
2021-06-14 13:39:58.532 ReferenceNozzle TRACE: Left.transformToHeadLocation((317.100000, 392.850000, 0.000000, 0.000000 mm), ...) runout compensation (-0.106298, 0.009784, 0.000000, 0.000000 mm)
2021-06-14 13:39:58.532 ReferenceNozzle TRACE: Left.transformToHeadLocation((317.100000, 392.850000, 0.000000, 0.000000 mm), ...) runout compensation (-0.106298, 0.009784, 0.000000, 0.000000 mm)
2021-06-14 13:39:58.533 AbstractHeadMountable DEBUG: Right.moveToSafeZ(1.0)
2021-06-14 13:39:58.533 AbstractHeadMountable DEBUG: Top.moveToSafeZ(1.0)
2021-06-14 13:39:58.534 AbstractHeadMountable DEBUG: Left.moveTo((140.740000, 10.000000, 0.000000, 0.000000 mm), 1.0)
2021-06-14 13:39:58.534 ReferenceNozzle TRACE: Left.transformToHeadLocation((140.846298, 9.990216, 0.000000, 0.000000 mm), ...) runout compensation (-0.106298, 0.009784, 0.000000, 0.000000 mm)
2021-06-14 13:39:58.535 AbstractHeadMountable DEBUG: Left.moveTo((140.740000, 10.000000, 0.000000, 0.000000 mm), 1.0)
2021-06-14 13:39:58.536 ReferenceNozzle TRACE: Left.transformToHeadLocation((140.846298, 9.990216, 0.000000, 0.000000 mm), ...) runout compensation (-0.106298, 0.009784, 0.000000, 0.000000 mm)
2021-06-14 13:39:58.537 MessageBoxes DEBUG: Error: java.lang.Exception: Manual NozzleTip 503A unload from Nozzle Left required!

If you then jog any amount in any direction, it will move to the Manual Change Location.

markmaker, 2021-06-15:
Hi @bilsef 

Thanks for reporting this. 

The problem is clear: the process uses an Exception to provoke a message box to the user informing them to change the nozzle tip. Before throwing the Exception, it should at least issue a `waitForCompletion()`, otherwise pending planned motion will not be executed. 

https://github.com/openpnp/openpnp/blob/ab113fb9ef31e05878069bc10099f6bac57abe9d/src/main/java/org/openpnp/machine/reference/ReferenceNozzle.java#L612

https://github.com/openpnp/openpnp/blob/ab113fb9ef31e05878069bc10099f6bac57abe9d/src/main/java/org/openpnp/machine/reference/ReferenceNozzle.java#L700

@vonnieda, as far as I understand, the use of an Exception as a user prompt is questionable, there are other issues not solved like run-out calibration after manual nozzle tip load (see below the throw in the linked code). Also, I wonder what this does to a running job and retry logic. Would it be acceptable to instead suspend the machine task thread, defer to the GUI thread, use a modal Message Box/Dialog asking the user to load the tip and confirm, and then resume the machine task? I'm also asking because I would like to implement something similar for interactively teachable bottom vision/alignment. 

There is another bug (that I already fixed earlier in my ongoing branch, but have not yet broken out) that the planner will retain queued motion commands through an Exception, and then unexpectedly execute later. That explains the last sentence in @bilsef's description.

> If you then jog any amount in any direction, it will move to the Manual Change Location.

Once clear how, I'll fix those ASAP.

_Mark

ian-arkver, 2021-06-15:
I agree. Exceptions should be for unrecoverable errors and ideally would unwind any side effects of the code that's throwing. eg. unfinished queued movements.

If you need a user action/confirm dialog then as you say there should be a mechanism for a modal popup followed by return to the calling point in normal flow (or maybe throwing an Exception for an "abort" button).

No idea how much work would be involved in that though.

markmaker, 2021-06-15:
> No idea how much work would be involved in that though.

I don't think it is difficult. I'll just call ...

```
SwingUtilities.invokeAndWait(() -> {
    // Modal dialog comes here...
});
```

... from the places I linked above. Maybe refactor that into a proper interface on the Machine (where the other task stuff is defined), to hide the GUI interaction from the implementation there and make if fail properly when running without GUI (from unit tests etc.). 

The "Abort" option which in turn throws is a good point, btw.

_Mark

vonnieda, 2021-06-15:
It throws an exception because we cannot call UI code from the machine.
Since we intrinsically handle exceptions as “pause the job” this all works
out the same as any other issue that the operator should address.

It seems like the easy and correct fix is just to add the wait for move
completion call.

Jason

markmaker, 2021-06-15:
> It seems like the easy and correct fix is just to add the wait for move completion call.

Yes, that will certainly be added. 

I just confirmed that run-out calibration _is_ performed after a manual tip change and job pausing. It goes through that step again. 

> It throws an exception because we cannot call UI code from the machine.

Just to understand (for the future): Why can't I call `SwingUtilities.invokeAndWait()` from a machine task? 

Trying to answer it myself: Is it because this could create a deadlock, as the GUI thread _in turn_ might be waiting for the machine task? 

_Mark

vonnieda, 2021-06-15:
Separation of concerns. If we put UI code in non-UI classes then it breaks
tests and use of the machine code outside of the Swing UI.

- Jason

markmaker, 2021-06-15:
Thanks.

I've created #1216 

_Mark

markmaker, 2021-06-15:
Hi @bilsef 

I just merged the #1216, and it is deployed.

Please test again, thanks!

_Mark

bilsef, 2021-06-20:
Hi Mark,

It seems to be working.

Thank you!

closed 2021-06-20 by bilsef