openpnp/openpnp #1222: On MacOS pipeline popups open in the background

fxframes, 2021-06-22:
# Problem Report
## Describe the Issue

While editing any pipeline all popup windows are opened in the background.

## Steps to Reproduce
Very easy to replicate:

Edit any pipeline, then try to close (save it) and the following screen will open behind OpenPnP:

[image]

Also when adding a new stage the same thing happens, the following screen will open behind OpenPnP:

[image]

## Expected Result
The popup windows should come up on the foreground.

markmaker, 2021-06-25:
@fxframes,

Just so you know: I looked at the code. The parent frame is properly assigned, as far as I can tell. However, it seems Java can only specify a `Frame` as the stacking parent, not a mere `JDialog`, so it is possible that the dialogs you mention are pushed behind the Pipeline Editor `JDialog`, so they are sandwiched between the OpenPnP Main Frame and the Editor. 

I don't know enough about Java Swing to know what can be done here. 

Refs:

https://github.com/openpnp/openpnp/blob/b040a09c6239a55ed5e781093000a2236f19b2f7/src/main/java/org/openpnp/vision/pipeline/ui/CvPipelineEditorDialog.java#L17-L30

https://github.com/openpnp/openpnp/blob/fc5efe612db302582dbaf887fde9bc46049f77f2/src/main/java/org/openpnp/vision/pipeline/ui/PipelinePanel.java#L253-L255

_Mark

fxframes, 2021-07-04:
Hi @markmaker thanks for looking into this. I'm a C/C++ guy myself so my Java is limited.
It doesn't affect functionality so I guess it's just annoying during setup, no big deal. :)
Thanks again.

markmaker, 2021-07-05:
AFAIK there are many MacOS OpenPnP users out there _for years_, even including @vonnieda himself, I think. So I'm a bit surprised you get these strange effects (also speaking about #1199). 

Could this be the theming that was recently introduced?

#1151 

_Mark

fxframes, 2021-07-08:
Hi @markmaker that is correct, when I change to any other theme the problem goes away.
As soon as I change back to the MacOSX default theme, though, the issue is back.

fxframes, 2021-07-08:
This one doesn't seem to be theme-related. The issue persists on all themes.

markmaker, 2021-08-19:
It could still be an artifact of the theming support. Even the default is a theme of sorts. 

Like I said [here,](https://github.com/openpnp/openpnp/issues/1222#issuecomment-868719757) I tried what I could. I have no more ideas. Someone more Java/Swing experienced, please step in.

_Mark

fxframes, 2021-08-20:
Thank you Mark.