openpnp/openpnp #1228: Maven Checkstyle not working

markmaker, 2021-07-02:
# Problem Report
## Description of the Issue
The `mvn test` should (among many other things) also ensure the OpenPnP [Coding Style](https://github.com/openpnp/openpnp/wiki/Developers-Guide#coding-style) is followed. Using Maven Checkstyle. This does not work anymore. 

Unfortunately, I have not even the beginning of an idea how all that maven stuff works, so I'll have to leave that to somebody else to fix. 

This is just a reminder for it to be fixed. The sooner we can shore this up, the fewer the violations that pop up once it is enforced again. 

## Steps to Reproduce

1. Use an identifier with snake_case instead of the mandated camelCase.
2. Run mvn test.

## Expected Result
It should fail.

## Actual Result
It does not fail.

## Bonus

@vonnieda, you once [said that line-endings could also be checked by Maven Checkstyle](https://groups.google.com/g/openpnp/c/9iA1PK1RaO0/m/1e3dRYgwBAAJ). Maybe whoever is looking into that, can just add that feature too, while they're at it.  ;-)

tonyluken, 2021-09-02:
Maven Checkstyle seems to be working now although I don't know why - I didn't do anything to make it work - I just noticed that Maven tests where failing on things that the Checkstyle was supposed to catch.  In any case, I added the line-ending check, see PR #1275

markmaker, 2021-10-11:
Thanks, @tonyluken.

The fact that CheckStyle _has never actually_ reported illegal [snake_case](https://en.wikipedia.org/wiki/Snake_case) is discussed [here](https://github.com/openpnp/openpnp/pull/1275#issuecomment-912935929). I was mistaken to assume it was broken. Snake case cannot be excluded due to WindowBuilder generating it. 

So we can close this.

_Mark