openpnp/openpnp #1235: ReferenceDriverCommunications can lose lines if a character read timeout occurs

Jnesselr, 2021-08-04:
# Problem Report
## Describe the Issue
In the `org.openpnp.machine.reference.driver.ReferenceDriverCommunications` class, there's a `readUntil` method. The way it's written now, if the read method gets a timeout exception, there might still be data in the StringBuffer that just gets thrown away. Now supposedly, this read timeout only happens after 500 milliseconds for the serial port, and in my testing, it was MUCH shorter than that. I'm running on Mac OS X 11.3.1.

In either case, I think the bug here is still a thing even at the 500 ms timeout point.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
I noticed it at first when my M115 command wasn't returning the firmware information correctly. When I started to debug it, there'd be points where it'd get part of the message, throw it away, then get the rest of the message and it wouldn't be able to parse it. It's even more noticeable on some of my code that has a timeout, but my timeout is only like... idk 100 ms? So I don't know why it's so trigger happy.

## Expected Result
I expected the entire response line to be read in.

## Actual Result
It read part of the line, discarded it with a timeout, then read the rest of the line which didn't match some regex and that made OpenPnP mad.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
I can't do a video atm, and I don't have the logs from before I tested this code, but here's the code I've been using to avoid this issue entirely. It basically will NOT return if it has started reading some data UNTIL it hits one of the characters it's looking for. I wanted to open an issue first because I'm sure this approach has other consequences that should be thought about.

```java
public String readUntil(String characters) throws TimeoutException, IOException {
        StringBuilder line = new StringBuilder();
        while (true) {
            try {
                int ch = read();
                if (ch == -1) {
                    return null;
                }
                else if (characters.indexOf((char)ch) >= 0) {
                    if (line.length() > 0) {
                        return line.toString();
                    }
                }
                else {
                    line.append((char) ch);
                }
            } catch (TimeoutException ex) {
                if (line.length() == 0) {
                    throw ex;
                }
            }
        }
    }
```

markmaker, 2021-10-07:
@Jnesselr,

Frankly, I have never investigated communications, so I'm a bit out of my depth here. 

I'm astonished, why there is a timeout configured at all. 

https://github.com/openpnp/openpnp/blob/b040a09c6239a55ed5e781093000a2236f19b2f7/src/main/java/org/openpnp/machine/reference/driver/SerialPortCommunications.java#L112-L113

The reader is threaded, so as far as I understand it, the right thing would be configuring fully blocking operation. But again, that's just based on a quick look. 

I guess your fix could equally solve it, but amounts to the same behavior in a more wasteful way.

One thing to look out for is disconnection. Somehow the whole thing must be torn down. 

_Mark

markmaker, 2022-02-03:
Hi @ian-arkver 

I tried to understand and find a way but failed. Could you please make a PR?

_Mark

ian-arkver, 2022-02-03:
Sure, I can take a look at it, but can't test it easily as I use TCP comms.

closed 2022-02-03 by markmaker