openpnp/openpnp #1240: Negative nozzle rotation with Marlin is not working

mariusmotea, 2021-08-05:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
Hi! 

I build a PNP machine based on last Marlin firmware. I have fixed most of the issues (mostly with vision) but and i have one 
problem that i cannot find a solution so far. In OpenPNP i cannot rotate the nozzle to negative values. It works fine if i'm on a positive range. Because of this issue most components are placed horizontally no matter if they have a vertical alignment on the PCB. 
## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Rotate the nozzle few mm to positive values. It must work
4. Rotate the nozzle back to 0. It must work
5. Rotate the nozzle few mm to a negative value. It will not work.

## Expected Result
To be able to rotate in both directions +/- 180 degree

## Actual Result
It rotate only it the final rotation is a positive value

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
My Machine.xml file:

https://pastebin.com/Jr7Lzdm5

markmaker, 2021-08-05:
Hi @mariusmotea,

Please post a log.

First guesses: 

1. Does Marlin have soft limits? Remove them.
2. Does Marlin have a special rotational axis mode (angles/degrees) that is limited to 0° ... 360°? 

OpenPnP needs an open range axis mode, it will optimize rotations, wrapping-around the shorter way etc.
https://github.com/openpnp/openpnp/wiki/Machine-Axes#controller-settings-rotational-axis

_Mark

mariusmotea, 2021-08-05:
I think there is no Marlin issue since with a 3d printer tool i can "extrude" to negative values. What i think that mai be  the issue is the use of E1 and E2 letters instead of A,B

mariusmotea, 2021-08-05:
Logs:

All axex are 0 after home:
1. Rotate +10 degree
2021-08-05 18:14:24.863 AbstractHeadMountable DEBUG: Left.moveTo((-5.767000, -55.389000, 24.500000, 10.000000 mm), 1.0)
2021-08-05 18:14:24.867 GcodeAsyncDriver DEBUG: serial://COM4 commandQueue.offer(M204 S6666.67, 15000)...
2021-08-05 18:14:24.867 GcodeAsyncDriver DEBUG: serial://COM4 commandQueue.offer(G1    E010.0000  F15491.93 ; move to target, 15000)...
Rotate back to 0 with - 10 degree
2021-08-05 18:14:28.293 AbstractHeadMountable DEBUG: Left.moveTo((-5.767000, -55.389000, 24.500000, 0.000000 mm), 1.0)
2021-08-05 18:14:28.297 GcodeAsyncDriver DEBUG: serial://COM4 commandQueue.offer(M204 S6666.67, 15000)...
2021-08-05 18:14:28.297 GcodeAsyncDriver DEBUG: serial://COM4 commandQueue.offer(G1    E00.0000  F15491.93 ; move to target, 15000)...
Rotate another - 10degree (noting happens)
2021-08-05 18:14:29.775 AbstractHeadMountable DEBUG: Left.moveTo((-5.767000, -55.389000, 24.500000, -10.000000 mm), 1.0)
2021-08-05 18:14:29.779 GcodeAsyncDriver DEBUG: serial://COM4 commandQueue.offer(M204 S6666.67, 15000)...
2021-08-05 18:14:29.779 GcodeAsyncDriver DEBUG: serial://COM4 commandQueue.offer(G1    E0-10.0000  F15491.93 ; move to target, 15000)...

markmaker, 2021-08-05:
Yes, but as you see, OpenPnP correctly sends

`G1 E0-10.0000`

so the problem is on the controller side.

My guess would be that you should configure axes as normal axes, _not extruder_ axes,

Unfortunately, I have no clue _how_. I know almost all the other commonly used controllers in OpenPnP, but **not Marlin**. 

Maybe @bilsef could help, I believe he knows it in and out. 

_Mark

KarampasEfthimios, 2021-08-05:
I have the same issue despite the fact that i use C axis and not an extruder.

KarampasEfthimios, 2021-08-05:
The new marlin supports X Y Z A B C axis. So Extruder is not the problem for sure

mariusmotea, 2021-08-05:
I work entire day on this issue. Found folowing changes on the internet that works, but i cannot adapt in my machine.xml and i think the reason is with 

on C1 and C2 axis:
`<pre-move-command><![CDATA[ T0S1 G92E{Coordinate:%.4f} ]]></pre-move-command>`
and 
`<pre-move-command><![CDATA[ T1S1 G92E{Coordinate:%.4f} ]]></pre-move-command>`

on driver section:
```
<command type="SET_GLOBAL_OFFSETS_COMMAND">
	<text><![CDATA[G0 {X:X%.4f} {Y:Y%.4f} {Z:Z%.4f} {Rotation:E%.4f} F{FeedRate:%.0f} ; Send standard Gcode move]]></text>`
 </command>
```

the bit problem in my case is the use of GcodeAsyncDriver, while in that example is used the GcodeDriver.

markmaker, 2021-08-05:
Oh, now I see the problem:

`G1 E0-10.0000`

You probably set `E0` as the axis letter. I doubt this will work, it is certainly not G-code compliant. 

You need to configure it to use `A` `B` `C` instead.

@KarampasEfthimios  is right, AFAIK Marlin supports `A` `B` `C`. Perhaps you need to upgrade Marlin?

_Mark

markmaker, 2021-08-05:
> ```
> <command type="SET_GLOBAL_OFFSETS_COMMAND">
> 	<text><![CDATA[G0 {X:X%.4f} {Y:Y%.4f} {Z:Z%.4f} {Rotation:E%.4f} F{FeedRate:%.0f} ; Send standard Gcode move]]></text>`
>  </command>
> ```

This looks completely wrong.

markmaker, 2021-08-05:
Perhaps you need this branch.

https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

See this about the 6axis support:
https://github.com/DerAndere1/Marlin/wiki

mariusmotea, 2021-08-05:
Thanks,

I will try to compile Marlin with support for 5 axis. Will return with the solution maybe it will help others in similar situation.

markmaker, 2021-08-05:
Configure axes here:
https://github.com/MarlinFirmware/Marlin/blob/924e4f95c8676aea02b5c33cb230b8ea9d84546a/Marlin/Configuration.h#L168-L189

Limits seem to be defined here, I assume `I` `J` `K` stand for whatever name you defined above like `A` `B` `C`:
https://github.com/MarlinFirmware/Marlin/blob/924e4f95c8676aea02b5c33cb230b8ea9d84546a/Marlin/Configuration.h#L1365-L1377

markmaker, 2021-08-05:
Ah yes, here it says that  `I` `J` `K` are the internal names, regardless of the names you give externally:

https://github.com/DerAndere1/Marlin/wiki#axis4_name

markmaker, 2021-08-05:
> 
> 
> Thanks,
> 
> I will try to compile Marlin with support for 5 axis. Will return with the solution maybe it will help others in similar situation.

Please note that I edited the comment above: 
https://github.com/openpnp/openpnp/issues/1240#issuecomment-893774134

It seems the official 2.0.x branch now has 6axis, no need to go with the bugfix branch. 

_Mark

KarampasEfthimios, 2021-08-06:
Also You must define a pin for endstop even if you dont use it because you will get an error. This pin must be triggered or else M114 wont return anything when you home your axis. And also in the place that you define the max travel distances for your axis for the new axis I put a negative value in order openpnp to turn your C axis in negative degrees

mariusmotea, 2021-08-06:
@KarampasEfthimios thanks for your feedback.
 So currently is mandatory to have a manual switch to be able to home the nozzles (A and B letters).

KarampasEfthimios, 2021-08-06:
> @KarampasEfthimios thanks for your feedback.
> So currently is mandatory to have a manual switch to be able to home the nozzles (A and B letters).

Yes its mandatory in software BUT

#define I_MIN_ENDSTOP_INVERTING false OR true depending your configuration in order to always be triggered and cheat the machine if you dont want to use limit switch for C axis
if you send M119 it should be always trigered

mariusmotea, 2021-08-09:
@KarampasEfthimios i think i made it to work without triggering M119 by setting a small value for I_MIN_POS and J_MIN_POS. Now when i start OpenPNP job on first component i need to wait few seconds more until it rotate to -720 then it operate normally until all components are placed.

Thanks again for your help.

```
#define I_MIN_POS -720
#define I_MAX_POS 3600
#define J_MIN_POS -720
#define J_MAX_POS 3600
```

markmaker, 2021-08-16:
> when i start OpenPNP job on first component i need to wait few seconds more until it rotate to -720

You mean it rotates from -720° (after homing position) to 0°, right?  (If in doubt, post a log.)

I guess you could remedy that by defining the manual home position here:

https://github.com/MarlinFirmware/Marlin/blob/924e4f95c8676aea02b5c33cb230b8ea9d84546a/Marlin/Configuration.h#L1719-L1721

_Mark

markmaker, 2021-08-16:
> Also You must define a pin for endstop even if you dont use it because you will get an error. 

Again, I think this can be avoided, if you set these:

#define MANUAL_I_HOME_POS 0
#define MANUAL_J_HOME_POS 0
//#define MANUAL_K_HOME_POS 0

https://github.com/MarlinFirmware/Marlin/blob/924e4f95c8676aea02b5c33cb230b8ea9d84546a/Marlin/Configuration.h#L1719-L1721

But that's just a guess.

_Mark

mariusmotea, 2021-08-16:
> Again, I think this can be avoided, if you set these:
>
> ```
> #define MANUAL_I_HOME_POS 0
> #define MANUAL_J_HOME_POS 0
> //#define MANUAL_K_HOME_POS 0
> ``` [...]

Thanks, will try this today and return with the feedback.

mariusmotea, 2021-08-16:
@markmaker you've made my day. That change was exactly what i need. Now i set `#define I_MIN_POS -99999` and `#define J_MIN_POS -99999` to be sure i will never hit negative rotation limit.

closed 2021-08-16 by mariusmotea

crashmatt, 2022-09-02:
A bit late to the party.  Just got a [working openpnp branch of Marlin 2](https://github.com/crashmatt/Marlin/tree/openpnp) running.