openpnp/openpnp #1250: Getting `Stage "match_template" not enabled` error

yconst, 2021-08-16:
# Problem Report
## Describe the Issue
In some operations I am getting a `Stage "match_template" not enabled` error. 

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Home the machine.
4. Sometimes the machine fails to home, stops early and displays this error.
-- or --

1. Start OpenPnP
2. Click the Power On button.
3. Home the machine.
4. Select Head.
5. Select "Visual Home", error appears. 

## Expected Result
- Homing should succeed every time
- Visual homing should not display an error

## Actual Result
I got the above errors in a message box.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
Liteplacer with TinyG board, Windows 10, Openpnp 2021-08-09

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

Unfortunately I got `error posting on Pastebin`.

Log here: https://pastebin.com/vgNt5A4a

markmaker, 2021-08-16:
pastebin is empty.

yconst, 2021-08-16:
You're right, thanks for pointing out. Updated the link: https://paste.ee/p/2qlAS

markmaker, 2021-08-16:
Unfortunately, the log is not at TRACE level. Set it to TRACE level on the log tab and retry/resend.

As a wild guess: Have you configured the `COMMAND_CONFIRM_REGEX`, or rather let [Issues & Solutions](https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions) do it for you?

_Mark

yconst, 2021-08-17:
Re-did the homing and pasted new log with trace level: https://paste.ee/p/O05SO

During the first home, it incorrectly ended the homing process without the x endstop being hit. So I had to use the e-stop to stop the process. The second homing went well. It seems like there is an implicit timeout during homing. But this might be unrelated to the present issue of the error.

The command confirm regex is ` ^tinyg .* ok.* `. Shoud I change it? It seems to correctly ack the commands.

yconst, 2021-08-18:
So for some reason it seems that the "match_template" step in some vision pipelines has been disabled. I enabled it and now visual homing wirks. This was probably why OpenPNP complained.

There is still the issue of timing out during regular homing. I will investigate if this is a TinyG issue.

Thanks for your help, closing this issue

closed 2021-08-18 by yconst

markmaker, 2021-08-18:
That log looks suspicious. I see no M114 issued and still it reports 

`Reported location changes current location from (x:0.000000, y:0.000000, z:0.000000, rotation:0.000000) to (x:15.000000, y:10.000000, z:0.000000, rotation:0.000000)`

Like if you set your own `POSITION_REPORT_REGEX` based on TinyG status reports (that you should switch off in the first place).

Have you accepted the recommendations of Issues & Solutions?

https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions

_Mark

yconst, 2021-08-19:
There was indeed a `POSITION_REPORT_REGEX` in the GCODE panel. I removed it and tried homing. I still get the same result, namely the machine stops after a while (not at the endstop), thinks it has homed, and then tries to move to an unrelated position to perform visual homing. I then have to push the e-stop to prevent the machine from crashing.

I experimented a bit more with this and it really seems like a timeout, either on the TinyG side or OpenPNP. If I placce the head manually close to the endstops, the timeout does not occur and homing is completed successfully (also visual homing).

I am fairly new to this. I am using the Issues and Solutions system to a large extent. 

I'm pasting the latest log with homing here: https://paste.ee/p/f9dTS

markmaker, 2021-08-19:
Please send the `machine.xml`.
_Mark

yconst, 2021-08-19:
Pasted here: https://paste.ee/p/dYHKo

markmaker, 2021-08-19:
`machine.xml` seems mostly ok, just follow Issues & Solutions to the end.

I reexamined the log, and it could be something similar as was discussed here:

https://groups.google.com/g/openpnp/c/veyVAwqS0do/m/VExzBskQAgAJ

It seems that TinyG initializes unhomed axes' (typically the rotation axes A B C) coordinates to garbage positions and then the first move will take very long. 

Perhaps adding `G92 A0 B0 C0` to your `HOME_COMMAND` will help. It initializes axes to zero. 

If this is the case, I will add it to Issues & Solutions for future users.

_Mark

yconst, 2021-08-20:
I found the culprit to the homing issue. TinyG stores default values regarding min and max positions per machine axis, and computes a max homing time based on these and the set homing velocity. In my case those were much smaller than the actual machine dimensions, and the driver though homing had failed. The whole shebang is explained here: https://github.com/synthetos/TinyG/wiki/Homing-and-Limits-Description-and-Operation

So I modified those in my connect code and now homing completes normally:

    $ex=1
    $sv=0
    G21 ; Set millimeters mode 
    $XTN=0
    $XTM=540 ; Machine width in mm, i.e. X axis
    $YTN=0
    $YTM=320 ; Machine length in mm, i.e. Y axis
    G90 ; Set absolute positioning mode