openpnp/openpnp #1254: MaskCircle not working for Nozzle Tip Calibration on Windows, but works on Mac

GeoSaffer, 2021-08-19:
I am using version 2021-08-09_16-22-36.dee37b9 on Windows and after a couple of days struggling to get it to calibrate the nozzle, gave up and tried on the mac. Version 2021-08-09_16-22-36.dee37b9 work on Mac.

The Problem

Trying to calibrate the Nozzle Tip, I went into the Pipeline to see why its not calibrating. When selecting the mask circle stage I noticed that the entire image was black and since the diameter for the mask was 100 , I was atleast expecting some form of image showing in a circle. 

I have tried values from 0 all the way up to 1000 and they all give a solid black image.

Then I changed it to -1 all the way to -1000 and all that did was giving me the full camera image but also no mask.

I have eventually found that using MaskRectangle works.

[image]

ian-arkver, 2021-08-19:
According to "Madison" on discord this can be fixed by reverting this change...

-                .getPipeline(camera, calibration.getCalibrationLocation(camera));
+               .getPipeline(camera, getUiCalibrationNozzle(nozzleTip).getLocation());

to  src/main/java/org/openpnp/machine/reference/wizards/ReferenceNozzleTipCalibrationWizard.java
introduced in commit 5b968a21e1fb1b4f5a00bf18831e83122c16a9d1

I've not verified this fix myself, and don't know how reverting this change would affect the rest of that commit.

markmaker, 2021-08-19:
This is new: 

> The nozzle tip calibration Wizard pipeline edit button has been changed to take the nozzle location as the nominal target location. This can be used to test nozzle tip detection at an off-center position.

See #1242

You need to actually move the nozzle (the one with that tip loaded) to the camera to test the pipeline.  Use the button: 
[image]

- **The Pro**: you can test the pipeline for non-centric nozzle positions. After using the button, jog away a bit.

- **The Con**: (that I failed to consider, sorry) if your nozzle is not at all at the camera, it will give you the result you observed.

I will add code to check the nozzle location and throw an error if it is AWOL (IMHO, it does not make sense to edit the pipeline without the nozzle tip in view). 

@ian-arkver & everybody at discord: please report to the group or to Issues (here) if you notice these things. I'm not on discord (I don't like it). I unsubscribed mail mention notifications once, and I cannot bring it back, [that stupid *** does not allow it](https://support.discord.com/hc/en-us/community/posts/360067778631-Re-enable-email-notifications). They want you locked-in all the way, which only adds to my dislike.

_Mark

ian-arkver, 2021-08-19:
@markmaker  Yeah, I know you don't hang out on Discord, which is why I summarised the conversation above.

I don't understand why @GeoSaffer would be seeing different behaviours on windows and mac platforms. That makes no sense.

Regards,
Ian

markmaker, 2021-08-19:
I think that is just a fluke. He had the nozzle over the camera on the Mac, but not on Windows. He attributed the wrong cause to the symptoms, happens all the time 🤓 

_Mark

markmaker, 2021-08-19:
I added the error message. Will be part of the next testing version.

[image]

_Mark

GeoSaffer, 2021-08-19:
Well that's great....a popup for the person that didn't move the nozzle to the camera.

If you pay a little closer attention to the images provided above you will notice that the nozzle is dead centre in the view of the camera.

The only difference between the 1st and second image is the diameter value set between a positive value and a negative value.

In both cases the nozzle is in position over the camera.

Clicking on the 
[image]

Has absolutely no effect whatsoever as it is in position already.

As I explained before, simply swapping the MaskCircle for a MaskRectangle makes it all work.

I am happy to do a conference call screen share and try all the things you would like if you are still not convinced.

markmaker, 2021-08-20:
@GeoSaffer 

I was mostly responding to @ian-arkver's diagnosis, assuming (perhaps wrongly) he and you talked on discord. 

I'm not sure, I understand the "tone" of your message right. 

If you want to sound offensive, then I suggest you solve the problem yourself. This is an open source project, it's all there in the code. And we all work on it for free and do our best.
 
If not, these are some facts that might help:

1. The nozzle location is taken as the target position (center of mask circle) right _before_ you open the pipeline editor. Moving the nozzle _after_ you have opened the editor has no effect. That's how parametrizing a pipeline (setting its properties) internally works, you have to do it _before_ the editor takes over, same as in production, i.e. _before_ the actual computer vision takes over. 
2. I re-checked the source code: the nozzle location i.e. parametrized circle center off the view is the only thing that can reasonably explain your phenomena.
3. Setting a negative radius does not change the circle itself, it just inverts the mask, i.e. the circle is blackened, everything else is clear. So again, this fits your description, the circle is just off your view.
4. It takes the location of the nozzle with the calibration tip loaded on it. If you have a two nozzle machine, be careful to move the right nozzle into view, especially if you are switching between nozzle tips in the Machine Setup tree  (use the 
[image] button every time, _before_ opening the editor).
5. MaskRectangle is not parametrized by any location, that's why it works. 
6. Check your bottom camera Units per Pixel. If they are completely wrong, this could perhaps happen too.

_Mark

ian-arkver, 2021-08-20:
It was Madison's diagnosis, all I did was repost it. As I said in the comment on this issue I haven't tested it myself.

I guess undoing that Wizard change just removes the need to position the nozzle before opening the pipeline editor.

With regard to your point 6 above - I have seen that before, when my pix/mm got reset to zero and MaskCircle fell over yet MaskRectangle worked. But that was a long time ago and restoring the right pix/mm fixed it.

All the best,
Ian

markmaker, 2021-08-20:
Hi,

If the nozzle is off view, I made it optional to move:

[image]

_Mark

GeoSaffer, 2021-08-20:
@markmaker 

Thanks for the feedback and apologies for offending you in my last post, I appreciate the work you guys do and fully understand that you are all doing it for free in your spare time.

Once again thanks for the detailed feedback, I will go through what you have suggested and let you know how it goes.

Question:
Why does it matter if I manually jog it to the camera or I press the 
[image]
 button, is it not based on the current head/nozzle coordinates ?

GeoSaffer, 2021-08-20:
ok I think I have found the issue as  both of you have suggested, Lastnight I calibrated the units per pixel and today it makes again no difference if I click the suggested button  or move the nozzle tip manually into position..................but..............The MaskCircle is working now.

I also notice there is a new oval on the bottom camera that was not there yesterday.
[image]

If I could make a suggestion, its great that you added the popup to notify that you are not in position and so in, but how about a similar popup for not being calibrated in unit per pixel.

markmaker, 2021-08-21:
Issues & Solutions should remind you, that you don't have Units per Pixels set. If it does not, please report!

Rationale: It would lead to complex and bloated code to check for the sanity of each and every machine setting before each and every machine action that uses it. With Issues & Solutions, this has been successfully decoupled. 

The new testing version will actually propose to calibrate Units per Pixels automatically for you, **you could hop on the testing train!** 🤣 But I propose you wait a day or so until I have the next improved version ready:

https://makr.zone/openpnp-automatic-machine-calibration-with-issues-solutions/676/

_Mark

GeoSaffer, 2021-08-21:
That is fantastic geez that would make life soooooo much easier

closed 2022-02-03 by markmaker