openpnp/openpnp #1258: Movements are merged, that should be separated

doppelgrau, 2021-08-22:
# Problem Report
## Describe the Issue

Downloaded the latest version something in the motion planer has changed. The waypoints in the movement from a heap feeder to the drop box are ignored and merged in one motion causing a crash.

## Steps to Reproduce
1. Set up a heap feeder
2. click on "get samples"
3. watch how the head moves to Safe Z after finding a part, then straight to the drop location, ignoring the waypoints.

## Expected Result
The patch follows the waypoints

## Actual Result
straight to the drop location => crash

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue

Thought it was a missing synchronization, but adding a wait for completion didn't help.
No open issues (expect 3d camera calibration) in issues.

    public void moveFromHeap(Nozzle nozzle) throws Exception {
        nozzle.moveToSafeZ();
        nozzle.waitForCompletion(CompletionType.WaitForStillstand);
        nozzle.moveTo(way1.derive(null, null, Double.NaN, Double.NaN), Motion.MotionOption.SpeedOverPrecision);
        nozzle.waitForCompletion(CompletionType.WaitForStillstand);
        nozzle.moveTo(way2.derive(null, null, Double.NaN, Double.NaN), Motion.MotionOption.SpeedOverPrecision);
        nozzle.waitForCompletion(CompletionType.WaitForStillstand);
        nozzle.moveTo(way3.derive(null, null, Double.NaN, Double.NaN), Motion.MotionOption.SpeedOverPrecision);
        nozzle.waitForCompletion(CompletionType.WaitForStillstand);
}

doppelgrau, 2021-08-23:
Even happens with the null motion planer, something is really strange there.

markmaker, 2021-08-23:
>  latest version

Do you mean the testing version?

If yes, there were some changes in the backlash compensation, where the `Motion.MotionOption.SpeedOverPrecision` is actually involved. Could you post a log?

_Mark

markmaker, 2021-08-23:
Or it could be something different, that just occurred to me... investigating.

doppelgrau, 2021-08-23:
Was the "develop"-branch.
Logfile will have to wait until tomorrow, maybe this evening (try to avoid crashing again after I cleaned up yesterday=> remove stuff from the table)
If it helps: the changes must have occurred in the last three month since my last PR.

markmaker, 2021-08-23:
... investigating ... Nope, it can't be [this](https://github.com/openpnp/openpnp/pull/1246/files#diff-5b8a73fd9b434c700d32dfe314f2d030a0d2d85878287a1eb114d0b24d6f7739), especially if you have the same symptoms using the `NullMotionPlanner`.

Again, please state the branch you're on (`test` ?) and send a log. 

_Mark

doppelgrau, 2021-08-23:
So, to make sure I do not have some problems in my build environment I downloaded the 2.0 version from the website. (Build date today 13:39). The attached configuration is used.
[OpenPNP-Konfiguration.zip](https://github.com/openpnp/openpnp/files/7032594/OpenPNP-Konfiguration.zip)

Movement after finding a part should look like this (only faster, used the setting to move the nozzle manually to the points, also the reason for necessary nozzle Z movement):
https://user-images.githubusercontent.com/6315907/130465889-f49976e0-c4ae-4293-a6d6-c34663f1f329.mp4

But result is (selecting "get samples"):
https://user-images.githubusercontent.com/6315907/130466140-44fc9666-0527-4e67-8258-1b9a55bb8732.mp4

Logfile in trace mode:
[OpenPnP.log](https://github.com/openpnp/openpnp/files/7032744/OpenPnP.log)

Hope that helps

markmaker, 2021-08-23:
Thanks @doppelgrau, 

I found the culprit. It is actually the `ContactProbeNozzle` that introduced the problem. 

Passing `Double.NaN` as coordinates is deprecated. The code that for the moment still handles them was bypassed by an override that the `ContactProbeNozzle` made.

So I fixed this. See #1261 . 

But the better (future-proof) thing would be to abandon the `Double,NaN` hack and use `derive()` instead. So your code would look like this:

```java
public void moveFromHeap(Nozzle nozzle) throws Exception {
    nozzle.moveToSafeZ();
    nozzle.waitForCompletion(CompletionType.WaitForStillstand);
    Location location = nozzle.getLocation();
    nozzle.moveTo(location.derive(way1, true, true, false, false), Motion.MotionOption.SpeedOverPrecision);
    nozzle.moveTo(location.derive(way2, true, true, false, false), Motion.MotionOption.SpeedOverPrecision);
    nozzle.moveTo(location.derive(way3, true, true, false, false), Motion.MotionOption.SpeedOverPrecision);
```

Also, much easier to understand, IMHO. 

_Mark

doppelgrau, 2021-08-23:
Works, will remove legacy just to be future proof.

closed 2021-08-23 by doppelgrau