openpnp/openpnp #1267: BlindsFeeder: Using wrong nozzle tip if not allowed to push

doppelgrau, 2021-08-31:
# Problem Report
## Describe the Issue

If a nozzle designated to "grab" a part in a BlindsFeeder is not allowed to push, the nozzle gets changed to a nozzle where pushing is allowed to move the cover.
But after moving the cover, the same nozzle tip is used to pick the part, not one compatible with the package.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Set up a BlindsFeeder with cover
3. allow only one nozzle tip to push the cover
4. add a part that has a footprint with only other nozzle tips compatible
5. feed part

## Expected Result
Cover is pushed with the compatible nozzle tip, part is picked with the compatible nozzle tip

## Actual Result
Part is being picked with the "push"-nozzle tip.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue

If nobody is faster I'll take a look in a few days.

doppelgrau, 2021-08-31:
A short question: is there a reason the planer checks the nozzleTip before feed, but not before pick?
And should that problem addressed there (move the check before pick) or in the feeder (can cause tripple nozzle change: Switch to nozzle for the part, switch to push nozzle, switch to part nozzle).

markmaker, 2021-09-01:
Just to be sure before I search: is this `develop` or `test` branch?

markmaker, 2021-09-01:
You say "feed", not "pick". What button exactly are you pressing?
[image]

I see it does **not** respect the compatible nozzle in the "feed" button handler, it just takes the nozzle that happens to be currently selected in the machine controls. 

But if you take the "pick" button i.e. when the part is actually picked by the nozzle tip, then it should work. Can you confirm?

The same should be true inside a Job. Again, can you confirm?

I'll fix the "feed" button. 

_Mark

doppelgrau, 2021-09-01:
Develop and observed it during a job. try to reate a video later to be 100% sure and also show the mechanics.

markmaker, 2021-09-01:
> 
> 
> Develop and observed it during a job. try to reate a video later to be 100% sure and also show the mechanics.

Hmmm... that's unexpected (and not what it says in your **Steps to Reproduce** 🤓).

I will still fix the feed button first. 

_Mark

markmaker, 2021-09-01:
> 
> 
> On job start does not work with push cover, it always pushes (logically, since previous feed operations could have moved the current position to a still covered one).
> As I said, try later to get a video of the behavior.

Ahh, it is a push cover, not a blinds cover. That explains it. Will fix that. 

https://github.com/openpnp/openpnp/blob/d8dde15e8408c23a414f711d928b06f03ef2fa95/src/main/java/org/openpnp/machine/reference/feeder/BlindsFeeder.java#L347

It should use the extended method signature with the three `boolean` and set the last one `true`.

**But**: this will clearly be very inefficient. Push-cover only works efficiently, if the picking nozzle tip is also allowed to push. I assumed that push covers are mostly used for rather large/clunky parts, where the nozzle tip is equally large and therefore always sturdy enough for pushing. 

What are your parts?

_Mark

doppelgrau, 2021-09-01:
because I had some trouble with pushing (probably related to the large height differences observed in #1269 ) I only allowed the N08 and N14 nozzle to push, for a stepper motor driver I configured N24 (HTSSP-28) and during job tried to pick with the N08, was surprised it worked, but not great.
Reproduced it with the pick button (perform feed and pick operation). should I upload the video or do you have found the issue?

markmaker, 2021-09-01:
Yes, I think I found it, see [above](https://github.com/openpnp/openpnp/issues/1267#issuecomment-910023203).

markmaker, 2021-09-01:
Please test with new version.

doppelgrau, 2021-09-01:
Seems to work, with an insane amount of nozzle tips changes, but works:

(first pick: N40 -> N14 -> )N40 (incompatible to all) -> N24(pick) -> runout calibration -> N14 (push) -> N24 -> pick up part

But not sure if related: each time before the pick (nozzle already placed at the part) I get an error: "GcodeDriver timeout waiting for response to M114; get position", timeout in machine setting is 30.000 ms, can that be to low? A quick test resulted with 45s no exception?!

closed 2021-09-09 by doppelgrau