openpnp/openpnp #1269: Feature Request: Ignore nozzle height calibration on nozzle tip change

doppelgrau, 2021-08-31:
# Feature Request
## Describe the Feature
Add an option for the nozzle tip changer to ignore the calibration.

## How Will It Look?
Some checkbox would be my suggestion

## Why Do We Need It?
With nozzle height calibration "on nozzle tip change" you can adjust for uneven nozzle tip lengths.
But if the difference is large (> 0.3mm in my case) the movements to load/unload the nozzle tips do not work anymore/crashes in holder.
Not sure if that problem happens often, didn't noticed it until now, since the spring in the CP40 holder has hidden it. so not sure if useful or I should simply go and shop some other nozzle tips.

## Provide Examples
Not that I am aware.

markmaker, 2021-09-01:
**Just to exclude this:** Are you using vision calibration? If yes, do you use the **newest** OpenPnP version (`develop` or `test`)?

There was a bug in tool changer vision calibration, see #1264. It causes collisions that may look like the Z was wrong, when in fact it was Y that was wrong. Sorry about that.

_Mark

doppelgrau, 2021-09-01:
No, vision was not used (although I like the idea).
Was develop. And Z was right, but the different nozzle length (calibration was nearly one mm difference)resulted in a different "real Z" between nozzle tip loading -> calibrating (changes) and unloading (does not fit anymore).
For the rest of the machine it was better

markmaker, 2021-09-01:
Just so I understand:

You have a nozzle tips, that have varying "Z for loading" vs. "Z calibration" offsets?

[image]

_Mark

doppelgrau, 2021-09-01:
Not sure if we understand us correctly.
The load/unload without Z calibration works fine => loading unloading with the "machine Z" works.
But for not totally clear reasons* when I calibrate the nozzles the differences add up to nearly one mm => the "calibrated Z" when switching nozzles causes problem. since the unloading move can result in the nozzle tip holder being one millimeter higher/lower than during load => tip crashes into holder.

*: not the best measuring equipment for that kind of stuff, but seems to add up from small machining differences, on resulting that the holder does not slide 100% in some nozzles, and one or two tenth somewhere else ...

markmaker, 2021-09-01:
Yes, I believe that _in the end result_, we're talking about the same thing. 

If the geometry of the nozzle tip and changer slot are otherwise the same, then it should manifest in [these Z coordinate](https://github.com/openpnp/openpnp/wiki/Nozzle-Tip-Changer#four-positions) **offsets** being different between different slots:

(From the [Wiki](https://github.com/openpnp/openpnp/wiki/Nozzle-Tip-Changer#four-positions))
[image]

It _does_ make sense to switch off Z calibration right before unloading. The question is: does this need another checkbox? Shouldn't it be the default?

Both loading and unloading should always happen at "naked nozzle" Z calibration. It is only after the loading/before the unloading, that the nozzle tip length (or how far back on the nozzle, the nozzle tip was loaded) actually changes the _effective_ Z of the nozzle. Therefore, there is no reason why there should be a checkbox. 

**Note**: you can even create the "unloaded" nozzle tip (see screen-shot above) as a stand-in for the naked nozzle. So the naked nozzle can be Z calibrated too (and runout-calibrated, btw. which is crucial on my banged-up holder). 

So what I need to do, is "uncalibrate" it right before the unloading. In case there is an "unloaded" nozzle tip present, I should instead load _its_ Z calibration, if present. Theoretically, that could be a loop-hole, as the machine might have been powered up with a nozzle tip loaded and the "unloaded" Z calibration not yet done. Users of machines that have huge naked nozzle Z offsets, would have to make sure that nozzle tips are unloaded before powering off.

_Mark

doppelgrau, 2021-09-01:
I see one use case (not sure if relevant) where you might want to run nozzle change with calibration: 
Your machine has really bad homing for Z, so you calibrate either the current nozzle or the raw holder before you are accurate enough for the nozzle changer.
But that can be catched with your "unloaded" tip. So that sounds good.

markmaker, 2021-09-01:
For the Wiki:
https://github.com/openpnp/openpnp/wiki/Contact-Probing-Nozzle#z-calibration-inside-the-tool-changer
___

[image]

markmaker, 2021-09-01:
Hi @doppelgrau,
If I create and merge a PR with the fix, could you test it quickly? I don't see how I can meaningfully test this on my machine.
_Mark

doppelgrau, 2021-09-01:
In around two hours I can test it.

markmaker, 2021-09-01:
Cool! Do you need it merged, i.e. as download version, or can you checkout and build the PR?

https://github.com/openpnp/openpnp/pull/1273

_Mark

doppelgrau, 2021-09-01:
Building should be no problem

doppelgrau, 2021-09-01:
Building not, but testing :-)
My configuration has already the "poke"-Setting from the HeapFeeder maintenance. Is there a chance to put the change in some "official" branch, so I can simply pull the change into my branch? (didn't found how to pull from other repositories than upstream)

doppelgrau, 2021-09-04:
Did a "redneck"-merge (just copied the three changed files) and tested it, as far as I can tell it forks perfect now. thanks.

markmaker, 2021-09-09:
Thanks for reporting and testing, @doppelgrau!

markmaker, 2021-09-09:
#1273 is merged into `test`.

closed 2021-09-09 by markmaker

markmaker, 2021-09-09:
Actually, it was first merged into `develop`, then `test`.