openpnp/openpnp #1274: use script to start a job

ghostyu, 2021-09-01:
# Problem Report
## Describe the Issue
Maybe it is not a issues, I've wrote a scripting that control a job starting, but it can not start the job

## Steps to Reproduce
1. dir C:\Users\Liu\.openpnp2\scripts\starttest.js
2. code the js
`
var imports = new JavaImporter(org.openpnp.model, org.openpnp.util);
with (imports) {
	print("## start a job")
	gui.jobTab.jobStart()
}
`

3. run the script

## Expected Result
I expect run the current job, but it do nothing

tonyluken, 2021-09-03:
Can you try gui.**getJobTab()**.jobStart()

ghostyu, 2021-09-10:
The problem has been solved.

`
        machine.setEnabled(true);
        machine.execute(function() { 
            var job = gui.jobTab.getJob();
            var jobProcessor = machine.getPnpJobProcessor();
            jobProcessor.initialize(job);
            while (jobProcessor.next());
            return true;
        });
`

closed 2021-09-10 by ghostyu