openpnp/openpnp #1281: BlindsFeeder: More Z clearance needed before pick, vacuum check can fail

doppelgrau, 2021-09-06:
# Problem Report
## Describe the Issue
When picking large but lightweight parts, the vacuum check before pick can cause the parts to be sucked to the nozzle tip => nozzle free/empty check fails

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. set up a part that is large but light (in my case a ptc fuse)
3. enable pre pick nozzle free check
4. pick a part in a BlindsFeeder with cover
5. after pushing the cover, the nozzle hovers directly over the part with a minimal distance
6. during vacuum check 8free) part can be sucked to the nozzle tip

## Expected Result
Part stays in the feeder until pick

## Actual Result
sucked before pick operation to the nozzle tip, => exception => job halts

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
Don't know it the distance needs to be configurable, or just 3mm extra should be added. (or even safeZ)

markmaker, 2021-09-06:
For most feeders, it should do it at Safe Z. 

It is called here:
https://github.com/openpnp/openpnp/blob/9f1d11e73dcbb86cda85d7dab9bb33d1493b5e97/src/main/java/org/openpnp/machine/reference/ReferencePnpJobProcessor.java#L621-L625

Before actually moving to the pick location via this
https://github.com/openpnp/openpnp/blob/9f1d11e73dcbb86cda85d7dab9bb33d1493b5e97/src/main/java/org/openpnp/machine/reference/ReferencePnpJobProcessor.java#L679-L683

via this
https://github.com/openpnp/openpnp/blob/9f1d11e73dcbb86cda85d7dab9bb33d1493b5e97/src/main/java/org/openpnp/machine/reference/ReferencePnpJobProcessor.java#L695-L704

But: there are feeder types that already move down to the feeder in the `feed()` call. One example is the BlindsFeeder, when it has a PushCover and needs pushing with the nozzle. This is probably not compatible. 

But you can switch the **Before Pick** option off, and just keep the **After Place** option:

[image]

https://github.com/openpnp/openpnp/wiki/Setup-and-Calibration%3A-Vacuum-Sensing#measurement-method

IMHO, the **Before Pick** option is dubious anyway, because it faults the _next_ placement (the one to be picked), not the one that was obviously not placed correctly, before.

_Mark

doppelgrau, 2021-09-06:
before pick makes sense, if the nozzle is used during the feed operation, makes sure no "stuck parts" remain.
So I see some applications.
Question is, is this a bug in the BlindsFeeder, that it moved there during feed and can/should that be changed or is it "works as designed". (I'd prefer changing that, but would mean some additional travel for all users)

markmaker, 2021-09-07:
@doppelgrau,

For now, we can just add a `moveToSafeZ()` in the test, for those who want it. It may involve extra Z moves, but as a deliberate/optional choice that's ok and for most feeders it will do nothing, as it is at Safe Z already.

> Question is, is this a bug in the BlindsFeeder, that it moved there during feed and can/should that be changed or is it "works as designed". 

First some background:

This is only the case for the push cover, not for the blinds cover, where (at least in a job context, and with the default cover opening method set) all the covers are opened before the job starts. For the push cover, it really only makes sense to push the cover using the nozzle that also picks, immediately before that pick.

Around the time I made the BlindsFeeder I also made the ContactProbeNozzle, where I wanted to introduce the feeder as a parameter to `pick()`, so I could ask the feeder to perform pre-pick ops, i.e. open the cover inside the `pick()`. But for good reasons, Jason did not like that:

https://github.com/openpnp/openpnp/pull/859#issuecomment-505209480

So I had to implement it in `feed()` i.e. I postponed a more elaborate solution (see the following discussion in the link above).

I later introduced the `Nozzle.moveToPickLocation()` in #1174 which is _independent_ and _before_ the `pick()` and therefore not changing the semantics of `pick()`. 

I guess we could call a `feeder.approachPickLocation()` method (or similar) in `Nozzle.moveToPickLocation()`, which passes the pre-pick location to the feeder. The pre-pick location may just be the pick location, or in the case of the ContactProbeNozzle, a small Z offset above the pick location for Z probing tolerance. Inside this new `feeder.approachPickLocation()`, the cover opening might happen.

But I'm unsure. Where is the red line between `feed()` and `feeder.approachPickLocation()`? Should lever feeders, drag feeders and push-pull feeders also be changed over, as they are also near pick location actions? What about those feeders that do pre-pick vision? Does it make sense to move a large part of the `feed()` methods over? Should we make two `feed()` methods instead, like `feedOffLocation()` for electronic feeders, and `feedOnLocation()` for all those feeders that involve moving the head to (near) the pick location? 

_Mark

doppelgrau, 2021-09-07:
The "test at safe Z" option sounds like a quick solution.

the general question is a good one, there I do not feel very competent to discus. Too many trade-offs.

markmaker, 2021-09-09:
Added Safe Z move in #1284, any chance you could test this?

It is both in `test` and `develop`.

_Mark

doppelgrau, 2021-09-09:
Not soon, in one and a half week at earliest.