openpnp/openpnp #1295: Request: have bottom vision `max offsets` settable per part (not globally)

maholli, 2021-09-21:
# Feature Request
## Describe the Feature
Rotational and positional precision should be settable for parts (not just global). They're currently called `Max linear offset` and `Max angular offset` as configured in Bottom Vision -> Machine Setup -> Vision -> Bottom Vision.

## Why Do We Need It?
1. This solves the general problem of overly-precise global positioning requirements necessary for fine-pitched parts but potentially (nearly) necessary for passives/larger parts. 
2. This is useful for everyone

## Provide Examples
Example scenario:
I have a board with 200 placements. Passives are 0603, but some ICs are 0.5mm pitch or less. I set a global alignment precision suitable for the passives, and have to life with the fact that sometimes my ICs are slight rotated or off-center and must be fixed by hand afterwards.

tonyluken, 2021-09-21:
Why not just turn off pre-rotate on the Bottom Vision -> Machine Setup -> Vision -> Bottom Vision tab and set pre-rotate to "always on" on the individual fine pitch part's ReferenceBottomVision tabs?  Just set all your other parts to "always off".

maholli, 2021-09-21:
> 
> 
> Why not just turn off pre-rotate on the Bottom Vision -> Machine Setup -> Vision -> Bottom Vision tab and set pre-rotate to "always on" on the individual fine pitch part's ReferenceBottomVision tabs? Just set all your other parts to "always off".

That doesn't address the issue. If `Max Linear Offset` is set to sub-0.5mm globally to accurately align a 0.5mm pitch part, the vision system is going to spend copious amounts of time fidgeting with bigger passives (for example) throughout the job.

markmaker, 2021-09-21:
Hi @maholli,

Two remarks:

1. If less precision is needed _"with bigger passives (for example)"_, why align these at all? 

2. On my machine, I noticed that alignment now almost always nails the alignment on the _first_ iteration, after two requisites are given: 

    a) automatic calibration of the bottom camera [through the new I&S solution (testing version)](https://makr.zone/openpnp-automatic-machine-calibration-with-issues-solutions/676/) which improves the prediction of correctional moves greatly, and 

    b) setting part heights by [Auto-Focus](https://github.com/openpnp/openpnp/wiki/Up-looking-Camera-Auto-Focus).

    There is no copious amounts of time spent fidgeting 😉

Please try these and report back. 

I'm not altogether opposed to adding some kind of setting, however, it should be well justified, as it complicates the UI. 

_Mark

tonyluken, 2021-09-21:
> That doesn't address the issue. If Max Linear Offset is set to sub-0.5mm globally to accurately align a 0.5mm pitch part, the vision system is going to spend copious amounts of time fidgeting with bigger passives (for example) throughout the job.

If you set pre-rotate to "always off" on the parts you don't want to align with bottom vision, the vision system won't spend any time fidgeting with them.   Max linear offset and Max angular offset only come into play if you use pre-rotate and they don't really set the precision at which parts get placed - they only control how many attempts (up to the maximum) the vision system will make at doing the pre-rotate.  Say you have Max angular offset set to 10 degrees.  If after the first pass, the part angle is less than that, say 9 degrees, the vision system is done (assuming the Max linear offset is also met).  However, that 9 degrees still gets applied after the vision is completed and before the part is placed.  So the placement error ends up being way less than 9 degrees.

closed 2021-09-21 by maholli

markmaker, 2021-09-22:
Thanks.

FYI, the release notes will be written, once the testing was concluded and the "new goodies" are merged into branch `develop` a.k.a. OpenPnP 2.0.

_Mark