openpnp/openpnp #1306: Issues and solution tab appears twice on MacOSX

Shaische, 2021-10-07:
# Problem Report
## Describe the Issue
Mac OS Big Sur 11.2.2 with a fresh install of OpenPNP and fresh install of Machine.XML has tab issue where issues/solution tab appears twice. But clicking the last tab is still pointing to the log window as it should. Just the tab is not right. Sometimes when starting up OpenPNP this doesn't happen, but sometimes it does. It's random. This happened on a previous install and still happens with a fresh install.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP on a Mac.

## Expected Result
Should have Log tab.

## Actual Result
Shows Issues/Solution tab twice.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
[image]

Shaische, 2021-10-07:
Just noticed this is happening when you also click the button "Find Issues/Solutions

[image]
"

markmaker, 2021-10-07:
> screenshot_404

see here:

https://github.com/openpnp/openpnp/pull/1151#issuecomment-937507609 
<https://github.com/openpnp/openpnp/pull/1151#issuecomment-937507609>

Shaische, 2021-10-08:
> For a workaround see here:
> [#1199 (comment)](https://github.com/openpnp/openpnp/issues/1199#issuecomment-918392848)
> _Mark

I searched for "show-indicator" and there is no text in my machine.xml even with just indicator. Seems this workaround is outdated?

markmaker, 2022-02-03:
`test` was merged in #1322.

closed 2022-02-03 by markmaker