openpnp/openpnp #1325: Panel options window not showing OK and Cancel buttons in Linux

neliooliveira, 2021-11-10:
# Problem Report
## Describe the Issue

Panel options window not showing OK and Cancel buttons in Linux.
The buttons are there because I can still select them using TAB.
It's probably just the size of the window that is not high enough to fit all the elements in the Linux theme.

## Steps to Reproduce

Open the panel options in a Linux machine.

## Expected Result
Buttons to be visible

## Actual Result

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
[image]

kaklik, 2022-10-04:
The same happens to me: 

[image]

I should use 3x Tabulator and enter to hit the "hidden" OK button. 

I am running that version of OpenPnP
[image]

on Ubuntu:

Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy

kaklik, 2023-03-02:
The recent version of OpenPnP has changed the Panelization Settings layout which resolved this issue. 

[image]

Therefore the issue should be closed. 

FIY @neliooliveira

markmaker, 2023-03-03:
Thanks, @kaklik!

closed 2023-03-03 by markmaker