openpnp/openpnp #1326: Fiducials in panel

neliooliveira, 2021-11-10:
# Feature Request
## Describe the Feature
The panel options window should have the option to select different fiducials and not just one for both locations.

## How Will It Look?
The panel option window will show a second dropdown to select the part for the second fiducial.

## Why Do We Need It?
I find it required because sometimes I can't adjust the vision to detect both fiducials due to differences in the machine locations and the reflection of light.
If I had to different parts I could adjust one for each corner individually.

## Provide Examples

noeldum, 2022-08-02:
Are you thinking of using a board fiducial as a panel fiducial here? 

I think it's a good idea. We would select the board number in the panel and one of its fiducial to define the panel fiducial. Can you confirm this is how you visioned it?

As a work around what if you just enter the coordinate of the desired board fiducial in the panel option window. That should do what you need? Say your board size is 100x100 and one fiducial is at position (10, 10). Say the panel is a 2x2. You could define the two panel fiducials as (10, 10) and (110, 110). 

Yes the jog to panel fiducial is also a missing feature. The trick I found is to add the panel fiducial (the one on the rail and not the board) as a component with positions relative to the first board. Then I jog to that new part to check its position.

tonyluken, 2023-08-02:
@neliooliveira,

Can we close this issue as I believe it was address by [PR1507](https://github.com/openpnp/openpnp/pull/1578)?

neliooliveira, 2023-08-03:
Yes, I think so although I haven't tested yet.

closed 2023-08-03 by neliooliveira