openpnp/openpnp #1342: 'Step by Step' button must behave for operation (not debugging)

noeldum, 2021-12-03:
# Problem Report
## Describe the Issue
'Step by Step' button seems to be more a debugging function than an actual operational action.
Other proprietary pnp that I used have a step by step operation that trigger a move after each click on the button. In the current state Openpnp seems to trigger not only physical steps but also some internal functionality. Leaving some blank clicks between each physical moves. 
This is certainly really useful when troubleshooting Openpnp itself but from the view of the operator it makes it really difficult to properly use the step by step function.

## Steps to Reproduce
1. Open a job ready for assembly.
2. Press the 'One Step' button with the expectation that each click trigger a movement of the machine.
3. Notice the blank clicks between each physical movement.

## Expected Result
Every single click should be responded with a physical move of the machine. Even though some movements are combined (e.g. feed/pick/retract) these are fine to be triggered in a single click. What should really be avoided is one click does nothing in physical domain.
There should probably be two step by step button. A debugging step by step and an operation step by step.

## Actual Result
Some clicks only trigger internal function giving the feeling the button does not work. 

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
None for now. Happy to make a video of the step by step on the Charmhigh. Although I think the issue is fairly easy to understand without this.

markmaker, 2022-05-15:
Due to the architecture (machine task threading model) it is not possible to add this as an afterthought. Machine tasks cannot communicate with the GUI and be blocked by it. At least that is my state of knowledge.

Furthermore, I disagree, that only machine motion is worthy of a step. There are actuator actions, vision operations etc.

_Mark

noeldum, 2022-05-15:
If you agree or disagree makes no difference to one fact. Manufacturers do offer proper step-by-step motion. I invite you to try a Charmhigh machine with stock firmware.
Probably some steps includes packs of motion together but i am only presenting things from the perspective of the operator. So whatever technical explanation goes for or against this feature. Openpnp does not deliver a basic function here. 

It really sounds like a car manufacturer refusing to implement reverse gear just because an engine only goes forward.

When you don't want you find excuses. When you want you find solutions.

markmaker, 2022-05-15:
Hey @noeldum, 

this is Open Source software provided free of charge. You are welcome to add this feature. I will gladly accept a PR if you pull it off while respecting the architecture. 

And no, the comparison with a missing reverse gear is not adequate. The stepping function in OpenPnP is just modeled after the semantic steps rather than after arbitrary machine motion segments. Hundreds of users find it adequate.

Your tone is rather misplaced. 

_Mark

noeldum, 2022-05-15:
The suggestion is not to remove of modify anything but rather add an extra feature existing on commercial machines.
I just found this video that shows exactly the functionality that hopefully Openpnp will match one day. Start from 14:13.

https://youtu.be/Y-LPYxpTh9U?t=853

(obviously yes a missing reverse gear is different to this problem. This was an allegory to give an equivalent on how I perceive the issue. Not something asking for validation for being correct or incorrect).

noeldum, 2022-05-17:
The issue seems to have been closed on the decision of a single individual. Also the prompt closing after the last message would indicate a decision made on gut feeling. I highly regret this for the community. 

My contribution to this project is for providing operator feedback. Also being a profesional in production I do consider the suggested feature to be something valuable for OpenPnp. Clearly this issue should remain opened to allow further contribution. I am not sure we should close issues just based on a misunderstanding. 

I apologise if my words sounded rude at some point.

markmaker, 2022-05-18:
We can reopen it now that the real intent is sorted out. Maybe you can edit the title to better reflect that intent.

_Mark

markmaker, 2022-07-13:
This was now implemented, with the known limitation that it cannot break up steps into individual machine moves, i.e. steps can have multiple motions in them. Example: the fiducial check will run as a whole, not individually per fiducial.

See #1441.

Available in the testing version:

https://openpnp.org/test-downloads/

markmaker, 2022-07-13:
If breaking up steps, i.e. single machine move stepping was ever attempted, it is likely better implemented at a more general level, i.e. _outside_ the JobProcessor, so other actions could be motion-stepped too. If somebody really needs that, a new Issue should be filed.

I'm closing this now.

noeldum, 2022-07-23:
Works well. Thanks for the fix.