openpnp/openpnp #1354: Dead lock during part pickup

sovcharenko, 2022-01-03:
# Problem Report
## Describe the Issue

Was playing with openpnp and got a deadlock when it was picking up a part.

[image]

Neither valve nor pump triggered.

Luckily i was running openpnp from IDE and managed to capture thread dump:

[threads_report.txt](https://github.com/openpnp/openpnp/files/7802235/threads_report.txt)

[logs.txt](https://github.com/openpnp/openpnp/files/7802245/logs.txt)

markmaker, 2022-01-03:
Are you sure? 

I'm no expert in Java concurrency, with that in mind, it this does not (yet) look like a dead-lock to me.

## Consumer  

```
pool-4-thread-1@5106" prio=5 tid=0x25 nid=NA waiting
  java.lang.Thread.State: WAITING
	  at sun.misc.Unsafe.park(Unsafe.java:-1)
	  at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
	  at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.awaitNanos(AbstractQueuedSynchronizer.java:2078)
	  at java.util.concurrent.LinkedBlockingQueue.poll(LinkedBlockingQueue.java:467)
	  at org.openpnp.machine.reference.driver.GcodeDriver.getReportedLocation(GcodeDriver.java:621)
	  at org.openpnp.machine.reference.driver.GcodeAsyncDriver.waitForCompletion(GcodeAsyncDriver.java:338)
	  at org.openpnp.machine.reference.driver.AbstractMotionPlanner.waitForDriverCompletion(AbstractMotionPlanner.java:795)
	  at org.openpnp.machine.reference.driver.AbstractMotionPlanner.waitForCompletion(AbstractMotionPlanner.java:718)
	  at org.openpnp.spi.base.AbstractActuator.coordinateWithMachine(AbstractActuator.java:359)
	  at org.openpnp.machine.reference.ReferenceActuator.read(ReferenceActuator.java:326)
```

## Producer

"Thread-12@5090" daemon prio=6 tid=0x23 nid=NA runnable
  java.lang.Thread.State: RUNNABLE
	  at java.net.SocketInputStream.socketRead0(SocketInputStream.java:-1)
	  at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
	  at java.net.SocketInputStream.read(SocketInputStream.java:171)
	  at java.net.SocketInputStream.read(SocketInputStream.java:141)
	  at sun.nio.cs.StreamDecoder.readBytes(StreamDecoder.java:284)
	  at sun.nio.cs.StreamDecoder.implRead(StreamDecoder.java:326)
	  at sun.nio.cs.StreamDecoder.read(StreamDecoder.java:178)
	  - locked <0x17b0> (a java.io.InputStreamReader)
	  at java.io.InputStreamReader.read(InputStreamReader.java:184)
	  at java.io.BufferedReader.fill(BufferedReader.java:161)
	  at java.io.BufferedReader.read(BufferedReader.java:182)
	  at org.openpnp.machine.reference.driver.TcpCommunications.read(TcpCommunications.java:70)
	  at org.openpnp.machine.reference.driver.ReferenceDriverCommunications.readUntil(ReferenceDriverCommunications.java:91)
	  at org.openpnp.machine.reference.driver.ReferenceDriverCommunications.readLine(ReferenceDriverCommunications.java:71)
	  at org.openpnp.machine.reference.driver.GcodeDriver$ReaderThread.run(GcodeDriver.java:1270)

It seems your Smoothieboard does not respond to the `M440` and `M114` code, the driver is simply still waiting for that response: 

[image]

You have a very high timeout of 360000ms or 360s or 6 minutes configured, which easily looks like "dead-locked", but from what I see, I expect it to timeout after that time. 

We need to figure out, why the Smoothie does not respond to the `M440` and `M114` or why the serial connection dies, if that is the reason. Another user [has a similar issue](https://groups.google.com/g/openpnp/c/XLuQGxR_r4Q/m/uwCJdJbJDwAJ), so I'm not excluding a bug inside OpenPnP  here. I'm assuming you are using an original Smoothieboard, right?

Like I said at the beginning, I'm no expert in Java concurrency, so if you see something in there that points to a deadlock, that could be an excellent explanation. Please educate me, then 😅

_Mark

markmaker, 2022-02-03:
Any feed-back?

As far as I can tell, documentation in this Issue not support the claim (to be a deadlock), so I will close this, if no further info is provided.

_Mark

closed 2022-02-03 by markmaker