openpnp/openpnp #1363: Java17: Simple XML serializer ignores converter annotations under Java 17

ian-arkver, 2022-01-21:
I commented out the test serialization before the machine.xml file write so we get the partially written file before the exception is thrown.

Comparing [the Java 17 version](https://github.com/openpnp/openpnp/files/7909150/machine.xml.txt) with [the Java 8 version](https://github.com/openpnp/openpnp/files/7909196/machine.xml.txt) we can see that the next element to be written would have been:

```
                  <cv-stage class="org.openpnp.vision.pipeline.stages.DrawContours" name="7" enabled="true" contours-stage-name="9" thickness="2" index="-1">
                     <color r="255" g="255" b="255" a="255"/> 
                  </cv-stage>
```
which is an element with the custom serializer. So I guess this isn't working with 17, at least when run by the testbench.

_Originally posted by @ian-arkver in https://github.com/openpnp/openpnp/issues/1361#issuecomment-1018029545_

ian-arkver, 2022-01-21:
Running BasicJobTest under VSCode debugger with a breakpoint in the ColorConverter code. The breakpoint is hit under JDK 1.8, but not hit under JDK 17 resulting in an illegal reflective access to the Color object.

[image]

The version of simpleframework.xml used is 2.7.1 which is the latest but dates from 2013. I guess this package is abandonware and perhaps a different xml serialiser should be used.

markmaker, 2023-06-07:
I understand you found a workaround, and this can be closed, right?

ian-arkver, 2023-06-07:
Well, I didn't, @lags did, but yeah I think this can be closed.

closed 2023-06-07 by ian-arkver