openpnp/openpnp #1369: Homing Fails

PNPls, 2022-01-26:
# Problem Report
I have a PNP machine that was working for a while but is now not wanting finish homing for some reason. Essentially it will home itself to 0 on all Axes and then it times out before it makes the move to the homing fiducial. The error it gives me when it times out is the "GcodeAsyncDriver timeout waiting for response to M114; get position". It is saying in the issues and solutions tab that axes are not being reported in the expected manner, but I haven't made any changes to the names of the axes or anything, and even if this is the issue I don't know how I would go about changing the way the axes get reported. 

I have tried to include everything that may be of use to illustrate my problem but let me know if there is anything else I can include to chad some light on this. Thank you for your time, and hopefully I am in the right place for this type of report.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Home Machine
4. The machine homes to min limit switches, then times out and doesn't finish homing to homing fiducial

## Expected Result
1. Start OpenPnP
2. Click the Power On button.
3. Home Machine
4. The machine homes to min limit switches, then to homing fiducial and home button turns black to show completion of automatic adjustments and successful homing.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
[LOG REPORT 1-26-22.txt](https://github.com/openpnp/openpnp/files/7944305/LOG.REPORT.1-26-22.txt)
[FIRMWARE.txt](https://github.com/openpnp/openpnp/files/7944318/FIRMWARE.txt)
[config.txt](https://github.com/openpnp/openpnp/files/7944323/config.txt)
[machine.txt](https://github.com/openpnp/openpnp/files/7944356/machine.txt)
[parts.txt](https://github.com/openpnp/openpnp/files/7944368/parts.txt)
[packages.txt](https://github.com/openpnp/openpnp/files/7944370/packages.txt)

closed 2022-01-26 by markmaker

PNPls, 2022-01-26:
Thank you,
The board I was trying to use was using a different firmware than I needed after all. Glad it was a clear case. I am using a Liteplacer machine with Openpnpversion2; Firmware version is now 5-axis 'feature/best-for-pnp-23a1f0db' and I can now get the machine to home. This newer board seems to have issues with getting the firmware flashed even though I followed the beautiful guide you gave me. Oh well back to using the old one for now. Should I refer that kind of issue to the google message boards? Thank you for a timely response have a nice day!

markmaker, 2022-01-27:
> This newer board seems to have issues with getting the firmware flashed

What newer board?

@arthurwolf , any changes in the v1 Smoothieboards?

@PNPls, are you using a China clone? I guess you've seen the **WARNING**, right?

https://github.com/openpnp/openpnp/wiki/Motion-Controller-Firmwares#smoothieware

_Mark

arthurwolf, 2022-01-27:
No changes.

Issues with flashing are often users having downloaded a html file pointing
to the .bin file instead of the actual .bin file. Check the file's contents.

Also, users should make sure they use the very latest example config file
with the very latest bin file. No mix-and-matching.