openpnp/openpnp #1388: Importing example CSV even gives error

jellewie, 2022-03-13:
# Problem Report
## Describe the Issue
So I want to [import EASYEDA file](https://pastebin.com/PpmSMP1A), but the format was wrong so I was writing a small script to convert them. But I run into the issue that my new clean install of opening doesn't allow even [the example SCV file](https://pastebin.com/hXqMqqQA). 

When importing the example SCV it complaints it can't find the relevant headers
[image]

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Add a new board job (green add icon, and save it somewhere)
3. Copy [the example SCV file](https://pastebin.com/hXqMqqQA) somewhere
4. File - import board - Named SVG, and select that sample SCV file
5. Click on "Import"
6. Now an "Import Error" shows up "Unable to find relevant headers' names.<be> See https://github.com/openpnp/openpnp/wiki/Importing-Centroid-Data"

## Expected Result
That the example SCV at least worked (but would be cool if EASEDA file was supported out of the box)

## Actual Result
Well it thorn that error

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
Clean 64bit install on windows 10, literally the first thing I tried to do

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.
[image]
No not possible, throws this error

jellewie, 2022-03-13:
Sitenote, this error message also has bugs,
1. The `<BR>` should be a line enter
2. and there should be an "info." at the end. how it states "see here for more" which is weird.

Another sidenote, if openPNP uses windows settings somewhere, here is my local format
[image]

jellewie, 2022-03-13:
So after a search, I've found what is the problem, the log TRACE is
[image]

the `ÿþ` (0xfffe in UTF-8) has been added as a prefix to the file to display the byte order mark in UTF-16.

I guess the best option would be that OpenPNP can accommodate for this, but I leave that to you.

GeoffState, 2022-10-09:
Unable to reproduce this error

GeoffState, 2022-10-09:
Refer https://groups.google.com/g/openpnp/c/oLnIIVpTDMc