openpnp/openpnp #1407: Program fails to start

FLYBYME, 2022-04-29:
# Problem Report
## Describe the Issue
I keep getting this error. I delete the folder and it makes no difference.

## Steps to Reproduce
Fresh install of windows 10 with version "2022-02-03_09-01-37.370b98b" also tried "2020-12-31_20-40-49.5524445"

## Provide Examples

[image]

markmaker, 2022-04-30:
Can this help?
https://groups.google.com/g/openpnp/c/sW4Btj0zXU0/m/Le-_8TLeAwAJ
_Mark

FLYBYME, 2022-05-05:
I tried install the requested file (Windows6.1-KB2670838-x64.msu) but its for windows 7 not 10

markmaker, 2022-05-05:
Maybe ask the same question on the user group:

http://groups.google.com/group/openpnp

_Mark

closed 2022-05-15 by markmaker