openpnp/openpnp #1415: Issues and Solutions tab repeats twice

Shaische, 2022-05-17:
# Problem Report
## Describe the Issue
Pretty sure this issue was submitted before but cannot find it. This is a UI bug. The issues and solutions tab is repeated twice where it should say "Log" in the tab instead. This is on a fresh install of latest stable OpenPNP with a clean machine.xml file. See attached image.

[image]

markmaker, 2022-05-17:
Duplicate of #1199.

Seems to be a bug in the MacOS theming. 

A workaround is here:
https://github.com/openpnp/openpnp/issues/1199#issuecomment-918392848

_Mark

closed 2022-05-17 by markmaker

Shaische, 2022-05-18:
@markmaker Why is the other issue closed then if it still exists?

markmaker, 2022-05-18:
I thought I close it, because there is a workaround solution to it. 

AFAIK it is not our problem, but that of the MacOS theming. It does not seem to support dynamically changing tab titles, on some but not all Mac machines. Because I don't use a Mac, I also can't explore this further. 

@vonnieda, should we keep Issues open in such cases?

_Mark

vonnieda, 2022-05-18:
It's fine to leave it open for a while. Maybe someone wants to investigate it further.

Shaische, 2022-05-19:
It's not just the name being copied over. All highlighted tabs have white text, making it hard to see what the tab says. See attached image. I think these two issues are related?

[image]
.